dresden-elektronik/deconz-rest-plugin #5519: deCONZ initial setup doesn't show coordinator node

manup, 2021-11-26:
## Describe the bug

In some installations during the initial setup the creation of the coordinator node fails.
When this occurs an empty node window is shown and no devices can be joined.

## Steps to reproduce the behavior

This can't always be reproduced, but one way to try trigger the bug is to delete config.ini, session.default and zll.db to fake a initial setup.

## Expected behavior

The blue coordinator node shows up and new devices can be joined.

## Screenshots

## Environment
 - Host system: cases are confirmed on all platforms
 - Running method: independend
 - Firmware version: not relevant as the issue is in deCONZ core
 - deCONZ version: at least 2.12.6 – 2.13.2
 - Device: not relevant here
 - Do you use an USB extension cable:  yes
 - Is there any other USB or serial devices connected to the host system? not relevant here

## deCONZ Logs

There are no helpful messages visible in the logs for this bug.

## Additional context

One workaround that helps often, but not always, is to restart deCONZ.

Se7enair, 2021-12-07:
Is the stick connected to a USB3 port? For me it worked to change it to a USB2 port.

manup, 2021-12-09:
The issue was in deCONZ core :)
This is fixed now in v2.13.3-beta.

closed 2021-12-09 by manup