dresden-elektronik/deconz-rest-plugin #5520: CentraLite 3400 Keypad

cobirnm, 2021-11-26:
## Device

- Product name: Centralite 3400 Keypad
- Manufacturer: CentraLite
- Model identifier: 3400
- Device type : Please remove all unrelated device types. 
  - Other: Keypad

## Screenshots

### Basic
[image]

### Power Configuration
[image]

### Identify
[image]

### Pool Control
[image]

### Temperature measurement
[image]

### IAS Zone
[image]
[image]

### Diagnostics
[image]

cobirnm, 2021-12-18:
Keeping this open

cobirnm, 2022-01-12:
In the mean time I have had partial success on creating a DDF file that might help everyone. Right now the keypad is working with this but does not always get updated. Maybe some of the developers could have a look to see if something is wrong in the file. Never the less this might be useful 
[3400.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7854067/3400.zip)

Smanar, 2022-01-12:
Ha good question, I realy have no clue how is working DDF for alarm system stuff ...
But having the device in the api is just a starting point, you have take a look on the alarm system stuff ?

https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/alarmsystems/#create-alarm-system

Have you set a pin0 ?

cobirnm, 2022-01-12:
> Ha good question, I realy have no clue how is working DDF for alarm system stuff ... But having the device in the api is just a starting point, you have take a look on the alarm system stuff ? [...]

yes and it works. The only thing that I notice right now is that the status in the keypad only gets updated when I insert the pin in the keypad itself. 
In other words when I arm or disarm the alarm on the keypad it shows the status in both the keypad and the alarm system. when I arm or disarm via api I don’t get the status in the keypad.
One more thing in order to get the status in the keypad even when arming on it I had to remove the arming delay (I set it to 0)

Smanar, 2022-01-13:
>when I arm or disarm via api I don’t get the status in the keypad

It's not normal ? This device is sleeping and just be awake when it detect someone front of it no ? So it's no possible to send request to it (I can be wrong, if someone with the device know it better than me ...)

Smanar, 2022-01-18:
@cobirnm can you  share your DDF ?
We are trying to do same but for a different device.

cobirnm, 2022-02-04:
> > when I arm or disarm via api I don’t get the status in the keypad
> 
> It's not normal ? This device is sleeping and just be awake when it detect someone front of it no ? So it's no possible to send request to it (I can be wrong, if someone with the device know it better than me ...)

The device has some proximity sensor that is not exposed (at least i was not able to expose it). Would it be possible to have this sensor check for the status of the alarm when triggered?

Smanar, 2022-02-04:
Someone else have asked that, it was removed on last version https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4845#issuecomment-991187987

I will try to re add it on the future keypad code, but ATM new code is blocked, so will be later.

Smanar, 2022-02-25:
OK so for information, if you make a DDF with the presence sensor, it seem it can work again https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5352#issuecomment-1050620761

cobirnm, 2022-03-20:
> Someone else have asked that, it was removed on last version [#4845 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4845#issuecomment-991187987)
> 
> I will try to re add it on the future keypad code, but ATM new code is blocked, so will be later.

Do you think you can implement this? Regarding the DDF and the presence I added it and did a hot reload but it is not exposed.

Smanar, 2022-03-20:
Can you show the DDF you are using ?
Set the status to "gold" to be sure it will be used.

cobirnm, 2022-04-01:
@Smanar when I set the status to gold the I can't use the keypad. It changes the name to presenceXX but it does not work. 
Never the less I cant get the presence sensor working or the keypad updating the status by itself.

P.S sorry for the late answer
[3400.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8398065/3400.zip)
.

Smanar, 2022-04-01:
Strange your json don't look so bad, can you try with this one

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "CentraLite",
  "modelid": "3400",
  "product": "CentraLite 3400 Keypad",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0402"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "cppsrc": "power_configuration.cpp:44",
            "ep": 255,
            "eval": "const vmin = 20.0; const vmax = 30.0; let bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;"
          }
        },
        {
          "name": "config/offset",
          "default": 0
[... 200 lines cut ...]
      ]
    }
  ]
}
```

There is not the cluster list capture, you are sure for endpoint and input/output cluster ?
```
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ],
        "out": [
          "0x0501"
        ]
      },
```

Can you show the json you have for this device on the ZHAAncillaryControl sensor ?
This device don't work too without the DDF ? (perhaps it s a  misng setting on the alarm system)

cobirnm, 2022-04-02:
@Smanar I tried your DDF and it did not work (Same behavior in home assistant showing as Unknown state). The difference here is that I could not go back to the old DDF and I had to go to an entire restore of Deconz to make it work again.
Regarding the Json do you mean this:

`"126": {
        "config": {
            "battery": 70,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "89bf790091196764512bce843628177a",
        "lastannounced": null,
        "lastseen": "2022-04-02T08:06Z",
        "manufacturername": "CentraLite",
        "modelid": "3400",
        "name": "AncillaryControl 126",
        "state": {
            "lastupdated": "2022-04-02T07:57:23.311",
            "lowbattery": false,
            "tampered": false
        },
        "type": "ZHAAncillaryControl",
        "uniqueid": "00:0d:6f:00:05:93:2a:42-01-0501"`

Smanar, 2022-04-02:
Ok, so I realy have no idea.
Your DDF is working, but stop working if you add the presence sensor ?
Can try with removing the fingerptinat part ?

Edit:

```
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
```

If I m right your device don't have 0x0406 cluster ? it use the cluster 0x0500 instead no ?

closed 2022-05-01 by github-actions[bot]