dresden-elektronik/deconz-rest-plugin #5527: AM43 Blinds Drive- Tuya Zigbee. Device: AM43-0.45/40-ES-EZ(TY)

sanderlv, 2021-11-27:
## Device
- Product name: box: AM43 Blinds Drive- Tuya Zigbee. Device: AM43-0.45/40-ES-EZ(TY)
- Manufacturer: TZE200_zah67ekd
- Model identifier: TS0601.
- Device type : - cover

## Screenshots

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-11-27:
Seems to be a duplicate of  #5049  please use that one. 

Closing.

Mimiix, 2021-11-27:
Reopened on behalf of @Smanar

Smanar, 2021-11-28:
Solved, after so so so much tries.
Classic problem with tuya, random inclusion, will be solved by DDF.

closed 2021-11-28 by Smanar

Mimiix, 2021-11-28:
Can you share the ddf?

Smanar, 2021-11-28:
Nope, the issue will be solved in the future using DDF, tuya are boring to include, they need special trick.

DDF are not working yet for covering, need a Swoops PR (already merged), I think it will be possible on next deconz version.

sanderlv, 2021-11-28:
It was a HASSLE to get it in...

But the trick was to "pair" in phoscon and equally "pair" the pyscial device (3 sec button down and menu) and then keep right klicking like hell on the following:
[image]

jscheurwater, 2021-12-01:
@sanderlv Thanks, that did the trick :)