dresden-elektronik/deconz-rest-plugin #5530: Danfoss Ally Enable local temperature calibration

simoneras, 2021-11-29:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi 3 plus
 - Running method: Raspbian Pi OS
 - Firmware version: (26680700)
 - deCONZ version: (2.13.02 beta)
 - Device: ConBee II 
 - Do you use an USB extension cable:  no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

simoneras, 2021-11-29:
hello everyone I would like to know if you can change the att value. 0x0010 local temperature calibration

Tk

Mimiix, 2021-11-29:
Hi,

As this isn't a bug report as mentioned in #5113 i am closing this issue.

Please use the forum or the discord for this question/request.

Kind regards,

closed 2021-11-29 by Mimiix