dresden-elektronik/deconz-rest-plugin #5532: Firmware update issue on GUI

Jenni32222, 2021-11-29:
## Describe the bug
 I will get a screen that I need to update my firmware, this can be really annoying because my Conbee 2 gets disconnected and my sensors won't work. Manually updating the firmware works, but the GUI still wants me to update, even though the firmware files have been deleted as mentioned in the Discord server.

## Steps to reproduce the behavior
 Unknown, randomly happens at least once a month, sometimes even twice a week.

## Expected behavior
 To not *kick* me out of the GUI because of a firmware update.

## Screenshots
 GUI firmware update notice: https://i.ibb.co/GkkBhHv/unknown.png

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26700700
 - deCONZ version: 2.12.6
 - Device: Conbee II
 - Do you use an USB extension cable: Yes, USB 2.0
 - Is there any other USB or serial devices connected to the host system? None

## deCONZ Logs
 Not sure if this is relevant since the issue isn't occurring all the time.

## Additional context

Mimiix, 2021-11-29:
@Jenni32222 It should be fixed in 2.13.0.

Let's keep this open until you verified its fixed :)

Jenni32222, 2021-11-29:
Thank you for the tip, I updated. I'll keep you posted.

Mimiix, 2021-11-30:
Sure thing :)

manup, 2025-05-02:
Closing since the GUI currently doesn't do any firmware updates and these are done via GCFFlasher.