dresden-elektronik/deconz-rest-plugin #5536: Single Channel LED Controller is reporting to support HS color which raises exceptions in hassio deCONZ component

MartinStellmacher, 2021-11-30:
## Describe the bug

The single channel LED light controller Philips 511.012 was working fine until this summer. Than after an update hassio was only able to switch the light on. But afterwards showed still the light in the off state and wasn't able to switch it off.
I filed an issue for hassio under https://github.com/home-assistant/core/issues/60659.

The problem is, that deconz reports 'colormode': 'hs' for the light info (the controller does not support color)

> '21': {'etag': 'c6e877bb0e614123636ed357ea0e6b57',
>                'hascolor': True,
>                'lastannounced': '2021-10-13T13:11:00Z',
>                'lastseen': '2021-11-30T19:42Z',
>                'manufacturername': 'Philips',
>                'modelid': '511.012',
>                'name': 'TerraceStripe1',
>                'state': {'alert': 'none',
>                          'bri': 110,
>                          'colormode': 'hs',
>                          'on': False,
>                          'reachable': True},
>                'swversion': '2.5.3_r47',
>                'type': 'Dimmable light',
>                'uniqueid': '68:0a:e2:ff:fe:3d:92:39-01'},
> 

but does not provide a hs_color value in the state messages:

> 2021-11-30 20:07:27 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.9/site-packages/pydeconz/gateway.py", line 183, in session_handler
>     self.event_handler(self.websocket.data)  # type: ignore
>   File "/usr/local/lib/python3.9/site-packages/pydeconz/gateway.py", line 207, in event_handler
>     self.update_group_color([device_id])
>   File "/usr/local/lib/python3.9/site-packages/pydeconz/gateway.py", line 240, in update_group_color
>     group.update_color_state(light, update_all_attributes=first)
>   File "/usr/local/lib/python3.9/site-packages/pydeconz/group.py", line 309, in update_color_state
>     self.update({"action": data})
>   File "/usr/local/lib/python3.9/site-packages/pydeconz/api.py", line 128, in update
>     signal_update_callback()
>   File "/usr/src/homeassistant/homeassistant/components/deconz/deconz_device.py", line 93, in async_update_callback
>     self.async_write_ha_state()
>   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 486, in async_write_ha_state
>     self._async_write_ha_state()
>   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 521, in _async_write_ha_state
>     attr.update(self.state_attributes or {})
>   File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 908, in state_attributes
>     data.update(self._light_internal_convert_color(color_mode))
>   File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 835, in _light_internal_convert_color
>     if color_mode == COLOR_MODE_HS and self.hs_color:
>   File "/usr/src/homeassistant/homeassistant/components/deconz/light.py", line 173, in hs_color
>     return (self._device.hue / 65535 * 360, self._device.saturation / 255 * 100)
> TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'
> 

## Steps to reproduce the behavior

Setup a hassio instance with Conbee and Philips 511.012 LED controller and try to switch the light on/off.

## Expected behavior

Report the light as dimmable without color.

## Screenshots

## Environment

 - Host system: Pi4
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: (2.12.06)
 - Device:  ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context
https://github.com/home-assistant/core/issues/60659

Mimiix, 2021-11-30:
what happens if you read the color cluster of this device?

MartinStellmacher, 2021-11-30:
Thank you for the quick reply!

How I can read the color cluster? I'm just using Hassio with the deconz and phoscon apps.

Mimiix, 2021-11-30:
Check the wiki :)

MartinStellmacher, 2021-12-01:
All the Attributes (including Current hue, saturation, x, y) are grayed out.

[image]

Mimiix, 2021-12-01:
Toobad, then we have to wait for a dev to check in

Smanar, 2021-12-01:
As we have this kind of issue since so long time, I think they will want to solve this issue using DDF, make a DDF for this device and remove the "colormode" in the json.

MaStahl84, 2021-12-21:
Hi,
with Version 2.13.4 i could solve this problem with a new DDF file.

I set colormode to static "xy". The lights are now controllable with Home Assistant.
All other only dimmable lights had the same settings.

I didn't try to remove the attribute colormode, maybe that works too.

Smanar, 2021-12-21:
>I didn't try to remove the attribute colormode, maybe that works too.

Depend of issue, if you have a device that is a dimmable bulb, better to remove it. This attribute is only present on color device.

odinr, 2021-12-25:
> Hi, with Version 2.13.4 i could solve this problem with a new DDF file.
> 
> I set colormode to static "xy". The lights are now controllable with Home Assistant.
> 
> I didn't try to remove the attribute colormode, maybe that works too.

have some rebranded dimmers (Namron 24v), after hours of debugging, I found this thread. Modified the DDF to `0,0` in deCONZ and symlinked the profile to all my dimmers. My issue was that it always reported "off" and never brightness.

Smanar, 2021-12-25:
>have some rebranded dimmers (Namron 24v), after hours of debugging, I found this thread. Modified the DDF to 0,0 in deCONZ and symlinked the profile to all my dimmers. My issue was that it always reported "off" and never brightness.

Hello, can I have more imformation ? wich one field have you modified ?
Was the state/on that was always "off" ?

dbaxiell, 2022-01-22:
Still an issue, the report is valid.

bohmandan, 2022-01-22:
Still an issue, the report is valid.

Mimiix, 2022-01-22:
@bohmandan please read smanars last message.

bohmandan, 2022-01-22:
> @bohmandan please read smanars last message.

Thank you, setting `colormode` to `xy` (static) worked like a charm!

Smanar, 2022-01-23:
And BTW, someone can share a working DDF for others ?

bohmandan, 2022-01-23:
I can share the DDF, but isnt this a workaround?
The device itself does not have colormode `xy`.
Removing the colormode would be the proper fix, right?
(just that it does not work at the moment, considered a bug I am guessing?)

Mimiix, 2022-01-23:
This is where ddfs are for. The issue here is the device not working as per zigbee spec. It's not a bug in deconz but in the device firmware.

bohmandan, 2022-01-23:
But I mean setting it to `xy` could potentially cause other issues, since the light only support brightness really.
If I take a look here: https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/lights/
It is mentioned under the property `hascolor`:
_Indicates if the light can change color. Deprecated - use state instead: **if light has no color colormode, hue and xy will not be shown.**_

So in my mind, that means that `colormode` should not be used. `hue` and `xy` should not be used either.
That would be the correct solution, according to specification, no?

ebaauw, 2022-01-23:
Indeed, the lights resource shouldn't contain `colormode` and `hascolor` should be false.

The "funny" thing is that the light (correctly) advertises to be a _Dimmable Light_, but (incorrectly) carries a _Color Control_ cluster.  Unfortunately, there are also devices that correctly carry the cluster, but advertise the wrong type.  I'm not even sure if it would be possible theoretically to handle all of these (somewhat) non-compliant devices correctly, so whitelisting seems to be the only practical workaround.  This is rather done in DDF files, than in the C++ code.

 I don't like the inconsistency in the API that it correctly doesn't expose `hue`, `sat`, `xy`, `ct`, `effect`, but still exposes `colormode` and sets `hascolor`.  My guess is that it set `hascolor` because of the _Color Control_ cluster.  Because it cannot read the _Color Mode_ attribute, it probably defaults `state.colormode` to 0, which is the numeric value for `hs`.  And because it cannot read _Color Capabilities_, it defaults that to 0 as well, causing the `state` attributes not to be created.

bohmandan, 2022-01-24:
@ebaauw thank you for the information.
So is it possible through DDF (with or without JavaScript) to change the capabilities that is being reported by the device?
For example then, remove the reported "Color Control" cluster, `0x0300` if I got it right.
So we fix it as early as possible in the process...

Smanar, 2022-01-24:
Sure, you can remove field or make some of them fixed (if it work)
It's something to try, I know some users have failed to remove a "power" field to a json.

bohmandan, 2022-02-15:
I finally did get it to work with just removing the property `colormode`.
It seems there was something wrong when I loaded/saved the DDF first, might missed setting the status.

Should I upload the DDF in the community forum, or open a pull request for the specific device?

closed 2022-03-18 by github-actions[bot]