dresden-elektronik/deconz-rest-plugin #5541: Device name

DubbyMcDubs, 2021-12-01:
## Device

- Product name: 2 Channel ZigBee Switch DC5-32v
- Manufacturer: eWeLink
- Model identifier: ZG-2001
- Device type : 
  - Other: Two channel relay module

- Links:
   - https://zigbee.blakadder.com/eWeLink_ZG-2001.html

   - https://www.aliexpress.com/item/1005002347566295.html?aff_fcid=838cd87b9730419a949bb351a9225daf-1635037251054-04238-_9gTCE1&tt=CPS_NORMAL&aff_fsk=_9gTCE1&aff_platform=shareComponent-detail&sk=_9gTCE1&aff_trace_key=838cd87b9730419a949bb351a9225daf-1635037251054-04238-_9gTCE1&terminal_id=c60aa1c2bd3d4f80907b0cc2716fb935

## Screenshots
Required screenshots:
[image]
[image]

### Scenes
 - Deconz crashes when I hit the read button for Scenes

Mimiix, 2021-12-01:
Can you please add all screens?

DubbyMcDubs, 2021-12-01:
> Can you please add all screens?

Which additional screens would you like?

Smanar, 2021-12-02:
Hello, wich one deconz version are you using ? Can you use the last one (that support DDF) ?

DubbyMcDubs, 2021-12-03:
v2.12.6

Qt: 5.11.3
GCC: 8.3.0 (C++ 201402)

Commit: d39d01
Firmware: 0x26660700

Smanar, 2021-12-03:
So you are on HA too ? Can't use a recent one ?

DubbyMcDubs, 2021-12-03:
I’m not sure what HA is, but if you have a link to download the version you
want me to run I will install.

Smanar, 2021-12-04:
HA = home automation, they use docker so they always have older versions.
What is your OS and your installation method ?
You need to switch to beta branch and just update package, it will use the lastest version

Mimiix, 2021-12-04:
Home assistant *

DubbyMcDubs, 2021-12-04:
No I am not using that. I use homebridge. Rasbian.

Smanar, 2021-12-04:
> Home assistant *

Oupssss, sorry.

So on pure raspian, no problem to use the beta version. You have the procedure here https://phoscon.de/en/conbee2/install#raspbian

A short description about the DDF here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Try to start a DDF  and show the preview or say where you are blocked.

DubbyMcDubs, 2021-12-08:
Thank you. 

I have updated it to the beta version and can see the DDF edidtor successfully.

I have made the changes but they dont appear to be taking effect in deCONZ. Are there any more steps?

Here is the preview:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "eWeLink",
  "modelid": "ZB-SW02",
  "product": "ZB-SW02",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/eWeLink/zb-sw02.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

DubbyMcDubs, 2021-12-08:
UPDATE:

It appears to only take effect when Control panel is set to "Strict". After that it seems to take effect even after restarting deConz and that setting reverting back to "Disabled". Is that expected?

Smanar, 2021-12-08:
>It appears to only take effect when Control panel is set to "Strict". After that it seems to take effect even after restarting deConz and that setting reverting back to "Disabled". Is that expected?

IDK for the strict, it need to work with the "Normal", but yes it s disabled at every restart, the feature is still beta, it s a security to don't broke a network.

>Also had a number of crashes when removing sub-devices/units and then hot loading. Hot loading in DDF disabled mode also only seems to work when adding devices but crashes when removing. This again is once you have tricked it into use the DDF by running at least once in strict mode.

Crash ? deconz is closing with segmentation fault error ?

DubbyMcDubs, 2021-12-10:
> IDK for the strict, it need to work with the "Normal", but yes it s disabled at every restart, the feature is still beta, it s a security to don't broke a network.
>
>
> Crash ? deconz is closing with [...]

I will give it another go using normal to make sure but I did try a few times with no result until I switched to strict.

Yes it is crashing without  presenting any errors.

Smanar, 2021-12-10:
And BTW there is a command to force DDF be enabled "dev-test-managed" https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-command-line-parameters

closed 2022-01-09 by github-actions[bot]

cobch, 2022-01-16:
Hi @DubbyMcDubs I was wondering if you managed to make working the relay with deconz? I will need one like that to make a project for my projector. Great to hear if I can order one soon. Thanks!

DubbyMcDubs, 2022-01-16:
> Hi @DubbyMcDubs I was wondering if you managed to make working the relay with deconz? I will need one like that to make a project for my projector. Great to hear if I can order one soon. Thanks!

Hi @cobch, yes I did get it working using the recommendations above so that it appears as I wanted it to inside Homebridge. I ended up writing a custom plugin for Homebridge to make it appear as garage doors. I have not made it public yet, but may in the future.

While the modules are very good and work well, one thing to look out for is the double firing of the output when it is using its one shot mode. Sometimes it stops my doors when opening and closing. My fix is to control it myself in the code.

Are you going to use this to trigger your projector to turn on? Home theater is my main hobby so may be able to help you acheive that without using the device, as there are many options.

cobch, 2022-01-16:
Great to hear. My plan is to use the interlock mode for running an linear actuator dc motor in order to drawn my UST projector from the wall to the desire distance. I will create a custom drawer so it does not need to be far from the wall when I am not looking a movie... I will order the same model and keep in touch with you in the futur if I have problem with deconz (-;
Thanks a lot for your answer!

carryonrewardless, 2023-12-09:
Sorry. I don't understand how I can use the DDF editor when it has not established a connection by the pairing process. Are the some steps I need to know to do this when not paired? Thanks.