dresden-elektronik/deconz-rest-plugin #5550: Firmware not connected - Raspberry Pi 4, Rasbee II, Home Assistant, deCONZ

eversvagen, 2021-12-03:
Hello,
I have searched in forums, search engines, youtube and more but can not find an answer to my problem.

I use a Raspberry Pi 4 together with a Rasbee II and Home Assistant + deCONZ. I follow guides and add rows to config.txt. Selects / dev / ttyAMA0 in the deCONZ plugin but still it says in firmware not connected. I have tried to format the SD card again and load everything again but the same error.

[image]

closed 2021-12-03 by close-issue-app[bot]

Mimiix, 2021-12-03:
This is something you need to ask on the forums :)

eversvagen, 2021-12-03:
> This is something you need to ask on the forums :)

I can't even find the forums.. 😂 Throwing the damn Raspbee II in the bin soon..

Mimiix, 2021-12-03:
https://forum.phoscon.de