dresden-elektronik/deconz-rest-plugin #5552: Wireless Scene Switch 3-gang Zigbee TS0043 _TZ3000_w8jwkczz

kvnnap, 2021-12-04:
## Device

- Product name: Scene Switch (Powered by Tuya)
- Manufacturer: _TZ3000_w8jwkczz
- Model identifier: TS0043
- Device type :
  - Switch
  Front:
  [image]
  Back:
  [image]

## Screenshots

[image]

[image]

## Switch 01

### Basic
[image]

### Power Configuration
[image]

### On/Off
[image]

### OTAU (Grey)
[image]

### Time (Grey)
Shows nothing.

### On/Off (Grey)
[image]

## Switch 02

### Power Configuration
[image]

### On/Off
[image]

### On/Off (Grey)
[image]

## Switch 03

### Power Configuration
[image]

### On/Off
[image]

### On/Off (Grey)
[image]

## Attempt

I have attempted to imitate other code changes, such as this PR [here](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4756/files) and compiled, tested and prepared a commit [here](https://github.com/kvnnap/deconz-rest-plugin/commit/9e4d97cbf4cab1f5e0ca8b9244179f7254d3f657). The device buttons work. I see the correct codes when a call to the REST API is made. However, I still see 3 On/Off entries under Lights in Phoscon. The resulting json and screenshots are below:

Correct switch addition:

[image]

```json
    "3": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "7b783e20cf59c61f25ff3e2841af53a2",
        "lastannounced": "2021-12-04T10:01:50Z",
        "lastseen": "2021-12-04T10:01Z",
        "manufacturername": "_TZ3000_w8jwkczz",
        "mode": 1,
        "modelid": "TS0043",
        "name": "Switch",
        "state": {
            "buttonevent": 3002,
            "lastupdated": "2021-12-04T09:56:12.904"
        },
        "type": "ZHASwitch",
        "uniqueid": "80:4b:50:ff:fe:06:9a:e6-01-0006"
    }
```

Incorrect light additions:

[image]

```json
    "8": {
        "etag": "0f4db69999cf0585054066ba8287affb",
        "hascolor": false,
        "lastannounced": "2021-12-04T10:01:50Z",
        "lastseen": "2021-12-04T10:01Z",
        "manufacturername": "_TZ3000_w8jwkczz",
        "modelid": "TS0043",
        "name": "On/Off light 8",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "80:4b:50:ff:fe:06:9a:e6-01"
    },
    "9": {
        "etag": "e3443691a63e2147e780d80e469dc850",
        "hascolor": false,
        "lastannounced": "2021-12-04T10:01:50Z",
        "lastseen": "2021-12-04T10:01Z",
        "manufacturername": "_TZ3000_w8jwkczz",
        "modelid": "TS0043",
        "name": "On/Off light 9",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "80:4b:50:ff:fe:06:9a:e6-02"
    },
    "10": {
        "etag": "0555947fec87e0ea2599a6ad5d17e405",
        "hascolor": false,
        "lastannounced": "2021-12-04T10:01:50Z",
        "lastseen": "2021-12-04T10:01Z",
        "manufacturername": "_TZ3000_w8jwkczz",
        "modelid": "TS0043",
        "name": "On/Off light 10",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "On/Off light",
        "uniqueid": "80:4b:50:ff:fe:06:9a:e6-03"
    }
```

## Environment and Other

The software runs on a Rasperry Pi 4b inside a docker container. To compile the change, the following Dockerfile was produced:

```dockerfile
FROM debian:10.11-slim

ARG MYUSER=deconz

RUN apt-get update && export DEBIAN_FRONTEND=noninteractive \
    && apt-get -y dist-upgrade \
    && apt-get -y install wget gnupg2 g++ make qtdeclarative5-dev \
    && wget -O - http://phoscon.de/apt/deconz.pub.key | apt-key add - \
    && echo "deb http://phoscon.de/apt/deconz $(cat /etc/os-release | grep -oP \
      '(?<=VERSION_CODENAME\=).*')-beta main" > /etc/apt/sources.list.d/deconz.list \
    && apt-get update \
    && apt-get -y dist-upgrade \
    && apt-get -y install deconz-dev \
    && useradd -ms /bin/bash $MYUSER \
    && apt-get -y install --no-install-recommends sudo \
    && echo $MYUSER ALL=\(root\) NOPASSWD:ALL > /etc/sudoers.d/$MYUSER \
    && chmod 0440 /etc/sudoers.d/$MYUSER

USER $MYUSER
WORKDIR /home/$MYUSER
```

and this Dockerfile was used to test the change after libde_rest_plugin.so was compiled:

```dockerfile
FROM deconzcommunity/deconz:latest
COPY libde_rest_plugin.so /usr/share/deCONZ/plugins
COPY button_maps.json /usr/share/deCONZ/devices
```

Mimiix, 2021-12-04:
@Smanar

Smanar, 2021-12-04:
It's normal.
Just ignore them for the moment, need to cxode something to prevent their creation.

kvnnap, 2021-12-04:
ok, will test the code as soon as there is an update.

Smanar, 2021-12-04:
I think the next update will be a DDF file and need more time for battery switch (the code is not ready yet)

kvnnap, 2021-12-28:
Keeping issue alive. @Smanar Any updates on this?

Smanar, 2021-12-28:
Nope, for the moment all PR are blocked.
The official method is using the DDF file, but the deconz core don't support battery switches yet.
So for the moment all is waiting list.

kvnnap, 2022-01-20:
Keeping issue alive.

kvnnap, 2022-02-02:
Hi @Smanar 

Is there any timeline about this and is there anywhere I can follow updates about support for battery switches?

I am asking so I can get an idea about when this issue will be tackled.

Mimiix, 2022-02-02:
@kvnnap please check #5733

kvnnap, 2022-02-26:
Keeping this alive..

kvnnap, 2022-03-25:
keeping this alive

kvnnap, 2022-04-19:
keeping this alive

Smanar, 2022-04-19:
Hello.
It's old, but If I right the code is working, you justt need to want to remove the useless entries ?

I think it's possible using DDF.
We will use them during the creation and let the legacy code make his work after.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Just select "edit DDF", and remove useless light entry, or just add a ZHASwitch one. If it don't work can you copy paste your DDF ?

kvnnap, 2022-04-25:
Hi @Smanar,

Been busy, will try later today and let you know.

Thanks!

kvnnap, 2022-05-07:
Hi @Smanar,

I tried editing the DDF and removing the useless light entries, however they still show in Phoscon Web. Then I tried removing the lights from Phoscon Web but it removed the switch too.

Not sure how to proceed.

Smanar, 2022-05-07:
Are you using the GUI or text editor ? The easier way is reincluding the device with a working DDF, don't forget to set the status to "Gold"
Can you show your final DDF ?

And yes it s something normal, you can't remove a "part of a device", it s all or nothing.

kvnnap, 2022-05-07:
I was using the GUI and I set it to Silver (but allowed silver DDFs).  I just tested with status Gold and I have the same result. I also tried removing the device and adding it back after saving this DDF. Here's the final DDF:

```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_w8jwkczz",
  "modelid": "TS0043",
  "product": "TS0043",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/ts0043.json",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0000",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001"
        ],
        "out": [
          "0x0006"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "description": "The current device battery level in 0&ndash;100&thinsp;%.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "description": "The last received button event."
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 2700,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Smanar, 2022-05-07:
So all seem fine for me ...
And if you try to include the device you have 1 sensor and still 3 lights entries.
Will ask to others devs.

kvnnap, 2022-05-07:
Correct, the web interface still shows 1 switch and 3 light entries. 

Current deCONZ version: 2.15.03

I am using this docker image:  deconzcommunity/deconz:latest

[image]

Smanar, 2022-05-07:
I m asking to other devs, Not sure at 100% the legacy code is bypassed by DDF, but the result is still strange.
You have a way to be sure the DDF is used ? because it s hard to use DDF on docker OS, you can see it on the GUI (on the editor window title when making "edit DDF") or on logs ?

kvnnap, 2022-05-07:
I restarted the container and noticed that the logs show that it is loaded:

[image]

Also, when I click Edit DDF on the switch I get this:

[image]

Phoscon Web still shows:

[image]

and

[image]

I also tried reverting the docker container to a version where I have no switch setup and restarting this whole setup, but it still gives this result.

Mimiix, 2022-05-07:
Phoscon will never show the correct result as they have to add support on their own.

Please use the rest api and websocket to determine if the  behavior is correct.

Smanar, 2022-05-08:
He have give the API extract on the first post, with all json.

For me it's realy an issue on DDF core

kvnnap, 2022-05-08:
I checked the API to make sure and the extra lights are still listed:

[image]

kvnnap, 2022-05-08:
I have also noticed that I am unable to get the button events (buttonevent is null). Prior to DDF I had changed the button_maps.json file (see [here](https://github.com/kvnnap/deconz-rest-plugin/commit/9e4d97cbf4cab1f5e0ca8b9244179f7254d3f657)) in order to make it work. How does it work now with DDF?

```json
{
    "config": {
        "battery": 0,
        "on": true,
        "reachable": true
    },
    "ep": 2,
    "etag": "72e466e434e9860936a90f40f4328e37",
    "lastannounced": "2022-05-08T08:52:49Z",
    "lastseen": "2022-05-08T08:52Z",
    "manufacturername": "_TZ3000_w8jwkczz",
    "mode": 1,
    "modelid": "TS0043",
    "name": "Switch 7",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "type": "ZHASwitch",
    "uniqueid": "80:4b:50:ff:fe:06:9a:e6-02-0006"
}
```

Smanar, 2022-05-08:
I m seing this PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6021

I readly don't see what do the code but seem usefull according to description

> Since DDF devices and their light and sensor resources are created async, the rule triggers can't be indexed in the plugin constructor.
> 
> The PR delays/debounces the indexing so that all devices are handled properly. This is also a bit faster since prior for each device useless container calls where made during startup.

kvnnap, 2022-05-09:
Is this PR going to be included in the upcoming release 2.15.04? If so I will wait and test when this is out.

Smanar, 2022-05-09:
Yep, it's already merged.

closed 2022-06-08 by github-actions[bot]