dresden-elektronik/deconz-rest-plugin #5556: Tuya LY-TAD-K616S-ZB || Zigbee ID: "TS0601"; "_TZE200_auin8mzr"

kiekerjan18, 2021-12-05:
## Device
- Product name: Ceiling Mounted Motion & Illuminance Sensor
- Manufacturer: Tuya
- Model identifier: TS0601
- Device type : 
  - Sensor
 
https://zigbee.blakadder.com/Tuya_LY-TAD-K616S-ZB.html

Screenshots:
[image]
[image]
[image]

Mimiix, 2021-12-05:
Missing some screens

kiekerjan18, 2021-12-05:
> Missing some screens

added screens, still missing some? I think I made screens of everything I could.

Mimiix, 2021-12-05:
Looks good

closed 2022-01-03 by github-actions[bot]