dresden-elektronik/deconz-rest-plugin #5560: Ikea Tradfri ON/OFF Model E1743 stopped connecting and now can't add

rdrr, 2021-12-06:
## Describe the bug

Using the HA add on, deconz has been working great!  Have one Ikea on/off switch that suddenly stopped functioning.  It was 'greyed' out in the Phoscon UI.  In the deconz application, the device was there, but had no connections to any others.  I have another Ikea on/off switch which is still working.

I restarted multiple times and finally 'deleted' the switch to try an pair it, but now I can't add it at all.  I hit the "pair" button on the switch 4 times and the red LED blinks 4 times, but deconz fails to make a connection.

## Steps to reproduce the behavior

Don't know how one could reproduce.  It has been solid for about a year, then was not functional about 7 days ago and has remained that way.

## Expected behavior

I expected to be able to re-pair the device after removing it.
## Screenshots

[Link to logs screenshot](https://imgur.com/VQeJ2I5)
## Environment

 - Host system: Raspberry Pi 4
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: 26660700
 - deCONZ version: 2.12.06 / 8/19/2021
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: Which? One Nortek HUSBZB-1 only using Z-wave and one samsung NVME 256GB drive (which hassos runs off of).

## deCONZ Logs

The 'copy' button on the debug logs doesn't seem to be working (multiple browsers on windows 10).  If needed I can try to get them some other way?

## Additional context

Thank you very much for any help.  I have an aquara switch that has never really worked correctly, but the Ikea ones were solid.  Would be great to get this working again as I plan to buy several more.  Thanks again!

Mimiix, 2021-12-06:
Hi,

As this is not a bug as described by #5113, I am closing this issue. Please open a question on the forums.

In addition to that, the on off model broke with the latest ikea firmware. Not sure if your issue is related but it caused the button to stop sending events.

Either way, I'm closing this issue. Feel free to ask on the forums https://forum.phoscon.de

Kind regards,
Dennis

closed 2021-12-06 by Mimiix

rdrr, 2021-12-12:
Sorry about that Mimiix.  Will do.