dresden-elektronik/deconz-rest-plugin #5563: Elko Smart Socket Single outlet 16

espen4001, 2021-12-07:
https://proff.elko.no/elko-smart/ELKO%20Smart%20System%20User%20Guide/English/ELKO%20Smart%20System%20User%20Guide.html

## Device

- Product name: ELKO Smart Socket Single 16A.
- Manufacturer: Schneider Electric.
- Model identifier: SOCKET/OUTLET/1.
- Device type :
  - Switch

## Screenshots

### Node Info
[image]

### Basic
[image]

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]

### On/Off
[image]
[image]

### Simple Metering
[image]

### Electrical Measurement
[image]

### Diagnostics
[image]

### OTAU
[image]

oyvindhauge, 2021-12-30:
@espen4001: did you solve it? I have exactly the same issue with a "double" stikk from Elko.

Jangomnz, 2022-01-06:
Looks like ELKO (Schneider) support is still lacking.  I have the ELKO 16A smart plug with metering and have the same issue.  Metering data is not provided in the API.  This unit is also based on Schneider components (model SMARTPLUG/1).

Mimiix, 2022-01-06:
Should be doable with DDF :)

Jangomnz, 2022-01-06:
Never done this before, but it looks like I in theory can just add a ZHAConsumption and a ZHAPower items..?  These do point to the correct clusters..

Mimiix, 2022-01-06:
Perhaps this helps?
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

Jangomnz, 2022-01-06:
It did :-)
I just added the ZHAConsuption and ZHAPower items (no changes in config).  
Set the device status to Gold. 
Saved the new DDF in the .local/share/dresden-elektronik/deCONZ/devices (created an elko subfolder).  
Restarted the HomeAssistant NUC (not sure if necessary). 
Ran the DECONZ.DEVICE_REFRESH service, and voila my consumption is displayed.

[image]

I'll do some testing to see if this behaves as expected.

Thanks

Mimiix, 2022-01-06:
@Jangomnz Great! I suggest putting it on Bronze as it's untested , but then you canshare it here perhaps?

Jangomnz, 2022-01-06:
I've changed to Bronze as advised.  Everything still works, and both power and consumption is updating.

Let me know if you want me to share more (or anything specific)

Mimiix, 2022-01-06:
You can share the DDF itself (the Json) so others can use it aswell :)

Jangomnz, 2022-01-06:
I'm running deCONZ though Home Assistant so I can't copy the JSON preview.  I also seem to be unable to find a location to save the file that I can access though the exposed samba shares from HA..  I'm a linux noob unfortunately..

Mimiix, 2022-01-06:
If you can follow that > https://community.home-assistant.io/t/deconz-addon-how-to-enable-vnc-set-port-in-network-configuration/137597

Then you can setup VNC and you can copy-paste :)
This really would help @espen4001  and @oyvindhauge

Jangomnz, 2022-01-06:
Here's the resulting .json:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "SMARTPLUG/1",
  "product": "SMARTPLUG/1",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/ELKO/elko_smartplug_16a.json",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300
        },
        {
          "name": "state/voltage",
          "description": "The measured voltage (in V).",
          "refresh.interval": 300
        }
      ]
    }
  ]
}

Jangomnz, 2022-01-06:
You probably have to change the modelid and product to SOCKET/OUTLET/1

cobch, 2022-01-13:
Hi @Jangomnz I have the schneider version and it looks really similar to the Elko. I have followed exactly your instructions (thanks anyway) but I do not manage to find the folder ".local/share/dresden-elektronik/deCONZ/devices " when I navigate through HA (Hassio on raspberry Pi). Any idea where I should save the config file?

cobch, 2022-01-13:
Hi again, I did find that folder (hidden in data folder) but unfortunately after a reboot I get the unhappy "unknown" beside power and consumption. Any ideas why? Thanks in advance for your help.
[image]

Smanar, 2022-01-13:
You need to check in deconz the same attribute on same cluster to be sure it s working.

But this DDF is realy "light", miss some detail and the binding, you have a better one for plug here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5577/files

But you can too select the attribute in the item/description when you select for exemple state/voltage https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet#attributes-on-top-of-the-left-scrollable-area-in-purple

To make a try, for state voltage > attribute 0x0505 cluster 0x0B04 endpoint 0x01 (not sure) manufacture code empty.

cobch, 2022-01-13:
I am not sure I understand everything but I tried to check and I can get Voltage and current working. But power and consumption are not responding (grey in deconz)
[image]
[image]
[image]

Jangomnz, 2022-01-13:
@cobch hopefully you've been able to progress some with the tip from Smanar above.  I have been able to get this working using the predefined templates for the metering clusters, so I think I got lucky.  Could be that you have to make sure the clusters and attribute addresses are mapped accordingly.  At a glance it looks like your RMS Voltage attribute should read the correct id, but it looks like your device is not providing an Active Power measurement (id 0x050b).

Jangomnz, 2022-01-13:
My device have a Electrical Measurement cluster that provides these attributes:
[image]
The 0x050B attribute is current power draw in W.
This is matched in the DDF entity:
[image]

Jangomnz, 2022-01-13:
In my device consumption comes from the Simpel Metring Cluster:
[image]
This is then pulled by the consumption state in the ZHAConsumption item:
[image]

cobch, 2022-01-13:
Ok, I have check the attributs and I have changed the attribut according to your advices. I get the following information from the simple metering cluster:
[image]

I have changed the power state and the consumption state:
[image]
[image]

After a reboot of Hassio I still get "unknown" when I look at my sensor in deconz integration. Something I am missing?

cobch, 2022-01-13:
I have saved my json file into that folder:
[image]

Jangomnz, 2022-01-13:
The fact that the deConz integration picks up the additional DDF parameters should be an indication that your DDF is found, so I don't think that's the issue. 
 In my device the power item is configured to pull from attribute id 0x050B on clusterid 0x0b04 (Electrical Measurement), not from the id 0x0400.  This id does not exist in my Electrical Measurement cluster (id 0x0B04) .  If you were trying to pull power from attribute id 0x0400 from Simple Metering you need to correct the cluster id's for this item (should be same cluster id as consumption). 

I really don't have much more to suggest as my knowledge is severely lacking in this regard.  Hopefully someone with a bit more insight to the inner working of deConz can chime in..

Jangomnz, 2022-01-13:
Just in case.  Cluster id's are displayed before the name in the dropdown.\:
[image]

cobch, 2022-01-13:
I will change with the correct cluster id because it was wrong )-: Thanks a lot.
Do you have any ideas why the bindings are unknown? Maybe another problem..
[image]
.

Jangomnz, 2022-01-14:
Unfortunately no.  I did not have to mess with bindings to get anything to work, and according to the DDF cheat sheet (
[image]
) these should be drag and drop (from the cluster dropdown on each device in the main window) which is difficult through the VNC interface..   My DDF has no bindings, so maybe try to delete these..?

Jangomnz, 2022-01-14:
Sorry, was supposed to copy a link not the screenshot.  https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet.

cobch, 2022-01-14:
No luck. Same behaviour after removing the bindings. I was wondering how was looking the bindings in deconz. Mine has no green lines, see below:
[image]

The last thing I was wondering if I did correct the config of control of DDF:
[image]

Mimiix, 2022-01-14:
What status did you put on the DDF?

Jangomnz, 2022-01-14:
Bronze for now..

Mimiix, 2022-01-14:
Does @cobch have that set too?

Jangomnz, 2022-01-14:
DDF mode, same as you Bronze and Gold.  The DDF file is set to Bronze..

cobch, 2022-01-14:
Yes, it is set to Bronze as well in my DDF file @Mimiix

cobch, 2022-01-14:
At least I have now my plug connected with green bindings but still the same "unknown" in front of power and consumption )-:

Mimiix, 2022-01-14:
Where does it say that?

What does the RestAPI itself show?

Mimiix, 2022-01-14:
@cobch Check the link i just send :)

cobch, 2022-01-14:
I have checked )-: But where should I write the following command? "
GET /api/<apikey>/sensors/<id>"

cobch, 2022-01-14:
[image]
I get this when I connect to deconz

cobch, 2022-01-14:
(-: I will do. Thanks!

Mimiix, 2022-01-14:
There also should be sensors.

cobch, 2022-01-14:
Like that (-:

"236": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "a93a672a38ea4c344e5566b98397cf30",
    "lastannounced": null,
    "lastseen": "2022-01-14T13:09Z",
    "manufacturername": "Schneider Electric",
    "modelid": "SOCKET/OUTLET/2",
    "name": "Power 236",
    "state": {
      "current": null,
      "lastupdated": "none",
      "power": null,
      "voltage": null
    },
    "swversion": "002.006.014 R",
    "type": "ZHAPower",
    "uniqueid": "50:32:5f:ff:fe:4d:30:5f-01-0b04"
  },
  "237": {
    "config": {
      "on": true,
      "reachable": true
    },
    "etag": "a93a672a38ea4c344e5566b98397cf30",
    "lastannounced": null,
    "lastseen": "2022-01-14T13:09Z",
    "manufacturername": "Schneider Electric",
    "modelid": "SOCKET/OUTLET/2",
    "name": "Consumption 237",
    "state": {
      "consumption": null,
      "lastupdated": "none"
    },
    "swversion": "002.006.014 R",
    "type": "ZHAConsumption",
    "uniqueid": "50:32:5f:ff:fe:4d:30:5f-01-0702"

cobch, 2022-01-14:
(lots of null, sounds not good...)

Mimiix, 2022-01-14:
Your is different it seems 🤔 
The one that @Jangomnz has seems to be "product": "SMARTPLUG/1", 

Perhaps that's the reason and it maybe works differently.

@Smanar can you take a look along?

cobch, 2022-01-14:
Thanks. I see that the "product"  is remplaced with "modelid". Great if @Smanar could help me (and it will be the second time (-;

Smanar, 2022-01-14:
>@Smanar can you take a look along?

Impressive that you have done, seriously, lot of search (and too much for me, have probably skipped some informations ^^)

[image]

Yeah I think the problem can be here.
IDK what happen, if you can check youtr DDF, need to have something like 

```
        "bind": "unicast",
        "src.ep": 1,
        "cl": "0x0B04",
        "report": [
          {
            "at": "0x0505",
            "dt": "0x21",
            "min": 1,
            "max": 500,
            "change": "0x00000003"
          },
```
the endpoint 0x01
the cluster 0x0B04
the attribute 0x0505

This part is important, but you can update the value at least one time. If you read the value in deconz, it will update the value in the API (so in HA too) but only 1 time (the binding/reporting can't work ATM on your DDF)
IDK if you have configured the bind yourself or use an old DDF ? (if you can show what you have on this part ?)

The endpoint is 0x06 not 0x01 on your device. you are sure you have corrected this value in all field ? Because you still have 0x01 in the json.

>"uniqueid": "50:32:5f:ff:fe:4d:30:5f-01-0702"

Don't use "auto" for endpoint, set direclty 0x06, there is some issue with "auto"

cobch, 2022-01-14:
Hi @Smanar 
I will try to understand what you describe above tomorrow (-:

I got my json file from the DDF and I copy it below if it can help:

  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "SOCKET/OUTLET/2",
  "product": "SOCKET/OUTLET/2",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/schneider/exxact.json",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x06"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "description": "The measured current (in mA).",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 300,
          "read": {
            "at": "0x0400",
            "cl": "0x0b04",
            "ep": 0,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0400",
            "cl": "0x0b04",
            "ep": 0,
            "eval": "if (Attr.val != -32768) { Item.val = Attr.val; }"
          }
        },
        {
          "name": "state/voltage",
          "refresh.interval": 300
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 300
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}

cobch, 2022-01-14:
Here some screenshots to help to understand, maybe?
[image]
[image]

[image]

Mimiix, 2022-01-14:
@cobch I am going to intervene here. As you have a different device opposed to the one in this issue, I am asking you to open a seperate device request.

cobch, 2022-01-14:
Of course. I will fix that asap!
Thanks anyway for your support.

cobch, 2022-01-14:
@Smanar  and @Mimiix I have created the issue #5672 with all the informations required. It was my first issue on Github so it was a big long to learn...