dresden-elektronik/deconz-rest-plugin #5568: TS011F model missing in 2 files

barts2108, 2021-12-09:
## Describe the bug

I received new tuya power plugs metered. I have some of them that have model TS0121. Now I received model TS011F and although it is metered, the metering does not show up in my Domoticz, where it does for the TS0121 model.

In file bindings.cpp line 2121 I can find TS0121 but not the TS011F
In file bindings.cpp line 3073 I can find TS011F but not the TS0121

In file database.cpp line 3637 I can find TS0121 but not the TS011F

While in electrical_measurement.cpp I can find both TS0121 and TS011F on lines 203 and 209
and in simple_metering.cpp I can find them on lines 100 and 101

## Steps to reproduce the behavior

## Expected behavior

I seems to me that those devices are more or less equal, then the software handling should be eqaual too ?

## Screenshots

[image]

On the left the TS011F basic/electrical measurement/simple metering
On the right the TS0121 basic/electrical measurement/simple metering

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: ConBee II
 - Do you use an USB extension cable: yes (and since I used it I have 0 problems with zigbee)
 - Is there any other USB or serial devices connected to the host system? If so: SSD

## deCONZ Logs

## Additional context

Mimiix, 2021-12-09:
What version of deconz you on?

Mimiix, 2021-12-09:
Hi,

Just did some research on github and I have 2 comments

- Make sure to make the issue as complete as possible. You didn't provide all needed information(environment). 
- please search the github on the model ID/ manufacturer name next time as there could be relevant issues. In your case : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4838

As this is a double post, I'm closing this issue.

Kind regards,

closed 2021-12-09 by Mimiix

barts2108, 2021-12-09:
Updated my version and firmware.

Mimiix, 2021-12-09:
It remains closed either way as it's duplicate.