dresden-elektronik/deconz-rest-plugin #5571: Tuya Zigbee 3.0 Remote Control (2 Gang) || Zigbee ID: "TS0042"; "_TZ3000_i3rjdrwu"

julunde, 2021-12-11:
## Device

- Product name: Tuya ZigBee 3.0 Brazil Brazilian Wireless US Remote Control (2 Gang)
- URL: https://www.aliexpress.com/item/1005003188106807.html?spm=a2g0s.9042311.0.0.62904c4dd51NGi
- Manufacturer: _TZ3000_i3rjdrwu (Tuya)
- Model identifier: TS0042
- Device type :
  - Switch

[image]

  

## Screenshots

**Switch Info**

[image]

**Node Info**

[image]

**Basic Cluster Info**

[image]

**Power Config Cluster Info, Button 1**

[image]

**On/Off Cluster Info, Button 1**

[image]

[image]

[image]

**Power Config Cluster Info, Button 2**

[image]

**On/Off Cluster Info, Button 2**

[image]

[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

julunde, 2021-12-11:
I attempted to get this to work myself by following the changes [here](https://github.com/dresden-elektronik/deconz-rest-plugin/commit/39a33e9a14cfb1017ed753161ea0d49120c14b90#diff-a3adce467fe93b9c612c198c187dad34f5284d6c9adf66aa28127f0f619979b5) and adding the manufacturer ID to the button_maps.json and then de_web_plugin.cpp file and trying to complile the deconz_rest_plugin but unfortunately I just get the following error when I try:

[image]

I'm hoping someone more familiar with the product can do better than I can 😄

Smanar, 2021-12-15:
I think you have an error some lines above.
Your changes are online or only locals ?

You can try to compile the code "official first" (but in other place or need to delete the previous folder)

```
git clone https://github.com/dresden-elektronik/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Can try without the -j2 (I think it will remove the "waiting for unfinished jobs")

julunde, 2021-12-16:
That worked @Smanar it compiled in the end, not sure what was different but the button events are showing up in the API now. Thank you so much!

Maybe the team can add it to a future release so its compatible for anyone buying in future? 😄

ebaauw, 2021-12-17:
```
[16/12/2021, 20:39:01] [Hue] Phoscon-GW: /sensors/4: warning: ignoring unknown ZHASwitch sensor {"config":{"battery":null,"group":null,"on":true,"reachable":true},"ep":1,"etag":"1bfd4b4f5640d78e988d53cf84773807","lastannounced":null,"lastseen":"2021-12-16T20:14Z","manufacturername":"_TZ3000_i3rjdrwu","mode":1,"modelid":"TS0042","name":"Switch (2)","state":{"buttonevent":1002,"lastupdated":"2021-12-16T20:11:40.488"},"type":"ZHASwitch","uniqueid":"8c:f6:81:ff:fe:91:0b:7e-01-0006"}

[16/12/2021, 20:39:01] [Hue] Phoscon-GW: /sensors/5: warning: ignoring unknown ZHASwitch sensor {"config":{"battery":null,"group":null,"on":true,"reachable":true},"ep":2,"etag":"7a73b8f209294b3804ad1dee739f35aa","lastannounced":null,"lastseen":"2021-12-16T20:14Z","manufacturername":"_TZ3000_i3rjdrwu","mode":1,"modelid":"TS0042","name":"Switch (2)","state":{"buttonevent":2002,"lastupdated":"2021-12-16T20:14:11.247"},"type":"ZHASwitch","uniqueid":"8c:f6:81:ff:fe:91:0b:7e-02-0006"}
```

The remote seems to exposed using two ZHASwitch resources, one for each endpoint?  I thought we settled on combining all endpoints into a single resource, cf. e.g. the ubisys?

Smanar, 2021-12-17:
>Maybe the team can add it to a future release so its compatible for anyone buying in future?

The core is changing ATM, we will use soon a way to support device.

>The remote seems to exposed using two ZHASwitch resources, one for each endpoint? I thought we settled on combining all endpoints into a single resource, cf. e.g. the ubisys?

Ha yes right, like for all battery switch.
But If I remember it need to be

```
                             manufacturer == QLatin1String("_TZ3000_a7ouggvs"))
                    {
                        //Making the device only for endpoint 0x01
                        if (i->endpoint() == 0x01)
                        {
                            fpSwitch.inClusters.push_back(ci->id());
                        }
                    }
```

Perhaps the device is not on this check ?

julunde, 2021-12-19:
> Perhaps the device is not on this check ?

I'm not sure if that was my doing when trying to get it to work, but I all I did was [add the device ID](https://github.com/dresden-elektronik/deconz-rest-plugin/compare/master...julunde:julunde-patch-1?expand=1) to the 3 three statements containing `sensorNode->manufacturer`  `lightNode.manufacturer` and `manufacturer` for Tuya devices.

julunde, 2022-01-12:
> The core is changing ATM, we will use soon a way to support device.
>
>
> Ha yes right, like for all battery switch. But If I remember it need to be
>
> ```
>                              manufacturer == [...]

@Smanar - Just for your info, I got 2 more switches and upon further testing it seems that DeConz/Phoscon creates an endpoint per switch  if you add -> delete-> re-add a switches without first deleting it in the device manager on the webapp. If you pair them first time, it will use a 01 endpoint for both buttons and differentiate buttons based on the event IDs (as expected).

Smanar, 2022-01-12:
Hu ?
So new inclusion > 2 uniqueID with 0x01 ?
2 inclusions > 1 uniqueID by endpoint ?

But you need to have only 1 entry in the API, no 2 ?

julunde, 2022-01-12:
Inclusion without removing first results in:

```
Endpoint for button 1: 8c:f6:81:ff:fe:91:0b:7e-**01**-0006 and "buttonevent":1002
Endpoint for button 2: 8c:f6:81:ff:fe:91:0b:7e-**02**-0006 and "buttonevent":2002
```

New inclusion, or inclusion after removing the device first results in:

```
Endpoint for button 1: 8c:f6:81:ff:fe:91:0b:7e-**01**-0006 and "buttonevent":1002
Endpoint for button 2: 8c:f6:81:ff:fe:91:0b:7e-**01**-0006 and "buttonevent":2002
```

closed 2022-02-11 by github-actions[bot]