dresden-elektronik/deconz-rest-plugin #5575: Sercomm XHS2-SE Door/Windows Sensor

CrispyPyro, 2021-12-11:
## Device

- Product name: Sercomm XHS2-SE Door/Window Sensor
- Manufacturer: Sercomm
- Model identifier: XHS2-SE
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots

### Node Info
[image]

### Basic
[image]

### Identify
[image]

### Diagnostics
[image]

### Other clusters that are not mentioned above

### Power Configuration
[image]

### Poll Control
[image]

### Temperature Control
[image]

### Zone Control
[image]
[image]

### OTAU
[image]
[image]

CrispyPyro, 2022-01-02:
Is this still the preferred method for submitting device requests?

Mimiix, 2022-01-25:
@CrispyPyro perhaps you can make a ddf. @Smanar can this be done here?

Smanar, 2022-01-25:
I think yes, the poll control cluster is something new on this kind of sensor but all the rest is same, this device use the IAS cluster 0x0500

Can take this one as sample https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5584
It's a motion sensor, but use same cluster.

closed 2022-02-24 by github-actions[bot]