dresden-elektronik/deconz-rest-plugin #5576: Sandy Beach HY367 Thermostat

moridianmess, 2021-12-12:
## Device

- Product name: Sandy Beach HY367
- Manufacturer: unknown
- Model identifier: unkown
- Device type : 
  - Thermostat

https://www.amazon.co.uk/gp/product/B093B4SCBT/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

I bought this device as the zigbee device compatibility list, https://zigbee.blakadder.com/Tuya_HY367.html, said it was already compatible with deconz, but I can't get it to pair through phoscon. I updated my conbee II firmware and deconz to the latest stable release versions this morning, just to be sure.

## Screenshots

[image]
[image]
[image]

### Basic
[image]

### Groups
[image]

### Scenes
[image]
[image]

### Tuya
[image]

### OTAU
[image]

If you need anything else, please let me know and I'll do my best to supply anything you require. This is the first time that something I've bought hasn't just worked, so please bare in mind I have no experience in debugging deconz.

Thanks,

craigueh, 2021-12-16:
I too have several of these TRVs and can't get them to work with deConz/Homeassistant.
Any help would be appreciated.

Mimiix, 2021-12-16:
@craigueh Their not supported yet.

Adzter, 2022-01-02:
Also bought this looking at the compatibility list as per OP. The Deconz compatibility links to #4675 (TS0601) which this device reports as. Is there some difference with this model? It seems to report as TS0601 in the screenshots?

edgardavim, 2022-01-02:
I have also the Hy367, I managed to link to HA via zigbee2mqtt, but after connecting, just stops working. Showing a huge "1P" at the valve screen.

at Zigbee2mqtt I have this details:
Zigbee Model TS0601
Zigbee Manufacturer _TZE200_2atgpdho
Manufacturer TuYa
Model TS0601_thermostat

i already tried to remove and add, but back to same status error. 

Anyone had identical or know how to move from here?

The message at the TRV
https://live.staticflickr.com/65535/51722394114_d14a1f429a_c.jpg

craigueh, 2022-01-02:
I just went with ZHA and everything was detected properly. Deconz/Z2MQTT
just didn’t have the necessary support.

moridianmess, 2022-01-02:
@craigueh are you able to control the device with ZHA? or just read the sensors?

craigueh, 2022-01-02:
Full control. Works as expected.

craigueh, 2022-01-02:
Forgot to mention, if you install Z2MQTT Edge instead it will also work.
ZHA for me still seems a little “half baked” so either option should be
fine.

Mimiix, 2022-01-02:
@Smanar

Smanar, 2022-01-03:
But this device is not supported ?
As it s a tuya thermostat it's not possible using DDF.

moridianmess, 2022-01-03:
I just tried DDF in basic mode -> bronze, silver and gold ticked and then went to pair window and enabled it. Put the trv into pair mode, it still appeared the same as before.

edgardavim, 2022-01-04:
I finally was able to have it working, but still fighting to understand to have it not working 100% . When I set close valve, until now, keeps letting heat go to radiator.

moridianmess, 2022-01-04:
@edgardavim just out of interest how did you manage to pair it?

With regards to letting heat in, you did let the TRV run through it's set up on that radiator? First boot and after removing batteries, it's suppose to go through a cycle to set itself up with that radiator. Maybe worth re-doing that anyway and trying some different adapters that came with it, as it sounds like the pin isn't being pushed in all the way on the radiator.

Yesterday I did some testing with ZHA, just to see how well the TRV actually works. I am facing a few issues, mainly with not getting sensor values until I physically interact with the device or write an automation to change the temperature every 2 minutes, and even then I don't always get an update. This may just be ZHA being awkward, as it did take four or five attempts to get it to pair successfully. Can anyone confirm whether this is a general issue and would happen in any Zigbee environment?

I will do some more tests over the next few days, I'm going to move the TRV closer to the zigbee gateway and do some more tests with both Deconz and ZHA to see whether these issues are device specific or ZHA specific. If this is a device specific issue, it does make it fairly unusable as I don't get updates about the current temperature and my plan was to never have to physically interact with a radiator valve, apart from changing the batteries.

I was going to try z2mqtt but I can't seem to find any instructions that appear to work when you are running HA in a VM on a linux machine.

edgardavim, 2022-01-05:
hello @moridianmess 

I am using it with zigbee2mqtt, in hassio. I have a zigbee bulb just above it, so reception is OK (it works as a repeater).

Just took out batteries, did this cycle. Still figuring out how to tell it to close the radiator. This is the challenge, how to define the open, as from my side it shows by percentage.

turned it off to see if closes valve.

[image]
[image]
[image]
[image]

OptimusGREEN, 2022-01-27:
I'm glad I found this issue, was just about to order some of these. I think I'll hold off and keep an eye on here...

Adzter, 2022-01-27:
It looks like support may be implemented as part of #5603

closed 2022-02-26 by github-actions[bot]