dresden-elektronik/deconz-rest-plugin #5584: ZM-RT20 Zigbee 3.0 Motion Sensor

rasmusdencker, 2021-12-17:
## Device
https://www.aliexpress.com/item/1005003606082802.html

- Product name: ZM-RT20 Zigbee 3.0 Motion Sensor 
- Manufacturer: _TZ3000_msl6wxk9
- Model identifier: TS0202
- Device type
  - Sensor

## Screenshots

[image]
[image]

### Basic
[image]

### Identify
[image]

###  Power Configuration
[image]

### IAS Zone
[image]

Smanar, 2021-12-17:
This device realy llok classic, will be easy to add it using DDf if you have a recent deconz version.

rasmusdencker, 2021-12-17:
Oh, nice one @Smanar - first time stumble upon that word. Found it in the docs and will take a look, thanks a bunch!

Smanar, 2021-12-17:
You have that to start https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
You can find some sample on the github or the forum, else can start and show your not working DDF. The harder part is having a deconz version that support DDF.

rasmusdencker, 2021-12-18:
I found the whole process to be quite trivial, and I got it (sort of) working. It triggers correctly one time, but after that it never reverts back to "no motion". It doesnt seem to be reflecting it in the 'Zone Status' either. I tried a bunch of different configuration and 'exec' press (just kicking tires), but nothing seems to be doing anything. 

[image]

I have a feeling that the problem resides in one of these areas:

1. Bindings. I know what bindings are as a general term, but I'm having trouble figuring out what it means in a deconz context. For now, I have none.
2. I know the device is configurable to go back to 'no movement' after 30, 60 or 120 seconds. Maybe this needs to be set first? Setting a default for the 'delay' config attribute doesn't seem to be doing. 
3. The device never actually sends a "No presence" state, but only sends a new "motion detected" state, so it's up to the  consumer to revert back if it hasn't heard from the device in a while. This is seems unlikely and feels like it goes against the Zigbee spec, so I don't think that's right.

Current DDF:

```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_msl6wxk9",
  "modelid": "TS0202",
  "product": "TS0202",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/aliexpress/ts0202.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0107",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0406",
          "0x0001"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 84000,
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 255,
            "eval": "Item.val = Attr.val",
            "fn": "zcl"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "awake": true
        }
      ]
    }
  ]
}
```

Smanar, 2021-12-18:
>    Bindings. I know what bindings are as a general term, but I'm having trouble figuring out what it means in a deconz context. For now, I have none.

You probably need one, but not sure it will help you.
Device need "bind" to use "report". You need to say the device to send report to the gateway for exemple, and set the reporting value.
But as the value is not updated when you look at it using deconz, it mean even with a bind you will have the same value

>    I know the device is configurable to go back to 'no movement' after 30, 60 or 120 seconds. Maybe this needs to be set first? Setting a default for the 'delay' config attribute doesn't seem to be doing.

I think it have a defaut value, so even with the worse situation it need to change after 120s

>    The device never actually sends a "No presence" state, but only sends a new "motion detected" state, so it's up to the consumer to revert back if it hasn't heard from the device in a while. This is seems unlikely and feels like it goes against the Zigbee spec, so I don't think that's right.

Yes, but there is other sensor that do that, If I m right deconz will send a new notification at every "motion detected" even if the value is still the same.

Else there is a code part to reset this value

```
        sensor->setZclValue(updateType, sensor->fingerPrint().endpoint, IAS_ZONE_CLUSTER_ID, IAS_ZONE_CLUSTER_ATTR_ZONE_STATUS_ID, num);

        if (alarm && item->descriptor().suffix == RStatePresence)
        {
            if (!IAS_SensorSendsRestoreReports(*sensor, zoneStatus)) {
                IAS_QueueRestorePresence(sensor, *item);
            }
        }
```

But this part use whitelist, so IDK if it s possible using only DDF

Smanar, 2021-12-19:
From ebaauw, you need to use the config/duration, but I m not sure it will work according to the code.

rasmusdencker, 2021-12-20:
Thanks a bunch for your help @Smanar. I was playing around with it yesterday and actually got it working by adding bindings:
[image]

It actually works fully as intended now. I have two of these devices, and I added the other one as well, which also works with that DDF. I have no clue why. I'm not reading the zone status, but rather reacting to the alarm notifications:
[image]

Yet, the zone value is still 0x0001 even when it reverts back to "no movement":
[image]

So, understanding bindings (and reporting?).. When I add a binding, Deconz tells the device to start reporting on changes in that cluster? That would explain why it didn't send alarm notifications before I added the bindings?

Smanar, 2021-12-20:
>So, understanding bindings (and reporting?).. When I add a binding, Deconz tells the device to start reporting on changes in that cluster? That would explain why it didn't send alarm notifications before I added the bindings?

Yep, exactly, so even if the value don't change, you have a notification every time ?

BTW if you can share the DDF for others too ?

rasmusdencker, 2022-01-10:
Sure thing, however I needed to roll back from 1.13.04 to 1.12.x because things started to act weirdly, dropping connections etc. But here's the DDF: 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_msl6wxk9",
  "modelid": "TS0202",
  "product": "TS0202",
  "sleeper": true,
  "status": "Bronze",
  "path": "/devices/aliexpress/ts0202.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0107",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0406",
          "0x0001"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 0,
          "max": 0
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 0,
          "max": 0
        }
      ]
    }
  ]
}
```

GrimDeclaw, 2022-01-12:
Hi,

Thanks for this DDF, I got the same pir sensor and it made it quite working!

The only thing that keeps bother me is : How to I get this to work with HA, as the sensor doesn't appear in phoscon? I'm using the deconZ addon of HA, and if the device doesn't appear in Phoscon, it doesn't appear in HA... I'm quite stuck to this issue right now :(

Mimiix, 2022-01-12:
It should come in HA as HA talks to deconz and not Phoscon.

Check the restapi for that

GrimDeclaw, 2022-01-12:
I'm not exactly sure why... but I checked the api, and it doesn't appear on /api/xxx/sensors endpoint
I then checked the lights endpoint (not usefull but hey... you never know!) nothing as expected.
I then checked the documentation, checked the DDF and while I was toying with it, it suddently change name in deconz interface from 54AC to "Presence 61'.
Since then, it appears in the API, and so in HA!
Thanks again!

Just one last question : the DDF configuration doesn't stand upon restart that's right? I have to wait for this device to be integrated in a future version for that right?

Mimiix, 2022-01-12:
As far as i know it should stand

GrimDeclaw, 2022-01-13:
Thanks for your reply
I think I have found why I was having hard time to get the DDF to work :  Default DDF mode is "Gold", and this DDF is set as "Bronze", so if I understant correctly, it was not picked-up by deconz.
 
I still have trouble adding devices, I need to be very very close to the raspbee II and try many times to pair before I could get anything to work (seems to be with recent HA addon update), but that's a story for another place!
Thanks again

GrimDeclaw, 2022-01-13:
> however I needed to roll back from 1.13.04 to 1.12.x

I ended up doing the same things (my devices pair wonderfully well with 1.12.x, while having hard time to pair with 1.13.04 ... don't know why!)

Do you manage to get the sensor working with 1.12.x or are we stuck (as it doesn't include DDF support) ?

Smanar, 2022-01-14:
Ha ? it s harder on 1.13 ? the DDF need to make the inclusion faster, strange ....

GrimDeclaw, 2022-01-14:
Actually, the pairing process doesn't even start unless I put the device right next to the raspbee2 (against the plastic box of the raspberry) with 1.13
With 1.12 (Same configuration, I just start/stop a docker container using the same configuration) it starts the pairing process at few meters away

I ended up going back to z2mqtt to be able to use the sensor.
Thanks anyway for your time and work!

rasmusdencker, 2022-01-14:
Same thing. I rolled back to 1.12 and just accepted that the device is not working anymore (since the lack of DDF support)

My experience was quite similar, except it wasn't just that I needed to put the devices close to the ConBee II; on 1.13.04, devices just randomly dropped off and became unavailable, devices that have always been working. I was unable to pair them again, no matter how close I put them to the ConBee, reset the device, completely removed from phoscon/deconz. They just wouldn't pair, and the few ones which did became very unstable again shortly after. 

This happened to both battery powered and mains powered, mostly IKEA bulbs, but several other vendors, too. I didn't file a bug because I thought it was just related to my setup, and the symptoms were so scattered, random and no certain way to reproduce. 

After reverting back to 1.12.6, everything became available again (even the devices which I hadn't touched after they became unavailable) and started to run smoothly.

Ceaus, 2022-02-09:
I have the exact same motion sensor and I'd like to try the DDF posted above. How do I "install" this DDF? I’m running the Deconz docker container from the Home Assistant add-on.

Smanar, 2022-02-10:
DDF description https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
Something that install a ddf manually on HA https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459#issuecomment-1019501674

Smanar, 2022-02-11:
Will be same ^^, if it work for him, can work for me too, but if you have issue, you will have same.
If you are not interested, or not have time, Ok, I can undersant, else don't worry for the rabbit hole or your competences, the only problem is your patience.

If I m understanding you have access to the GUI ?

Can you just check if you have the same device, with compating model id and manufacture name.
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters
Thoses values are on the basic cluster 0x0000

Can try to create a fake DDF, save the file, look the path name on the DDF editor title, and "make magic" (sorry I can't help you here for dockers) tu edit the content with the one on this issue.

GrimDeclaw, 2022-02-11:
> 

> I think I have found why I was having hard time to get the DDF to work : Default DDF mode is "Gold", and this DDF is set as "Bronze"

My 2 cents here: do you change the DDF mode to "bronze", or change the "bronze" status of the DDF to golden? This was my mistake when I tried it and it tooks me time to understand why it won't work

Ceaus, 2022-02-12:
> Will be same ^^, if it work for him, can work for me too, but if you have issue, you will have same. If you are not interested, or not have time, Ok, I can undersand, else don't worry for the rabbit hole or your competences, the only problem is your patience.

Alright, you've convinced me to give it another try. I'll be silent for a while, and report back when I have definitive results. Thanks for your kindness.

J3susC, 2022-02-22:
@rasmusdencker

Did you manage to reduce the time, the device kept the "motion detected"-status? Mine takes 2 minutes to revert to "no motion detected" and I was wondering if it is possible to reduce that.

Cheers,

jC

timtatam, 2022-03-23:
I have the same sensor, will the needed configuration be added to phoscon so it can be discovered automatically?

Smanar, 2022-03-23:
You have too the "_TZ3000_msl6wxk9" ?
And no, I don't see it in deconz yet, no one have made the PR.

You can install the file yourself, the DDF is here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5584#issuecomment-1008845513

Take care at path 
`"path": "/devices/aliexpress/ts0202.json",`

or don't forget to edit this line.

If it work for you too, you can make the PR for other or confirm it works for I make the PR myself.

timtatam, 2022-03-24:
Yes, I added the provided DDF.
The sensor appears now after a while in HA, but all entities are 'unknown'. Seems I do something wrong?

Smanar, 2022-03-24:
IDK.
Can you shwo the device json ? You will find it for exemple in phoscon / help / API Information / sensors.

timtatam, 2022-03-25:
Setting the status to 'Gold' made it appear in Phoscon.
The device json
`                    {
    "config": {
        "battery": null,
        "duration": null,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "5e4dd1bcb5fc1d44fff63d5cb8a206f1",
    "lastannounced": null,
    "lastseen": "2022-03-25T07:59Z",
    "manufacturername": "_TZ3000_msl6wxk9",
    "modelid": "TS0202",
    "name": "Presence 30",
    "state": {
        "lastupdated": "none",
        "presence": null
    },
    "type": "ZHAPresence",
    "uniqueid": "a4:c1:38:e2:19:54:a1:58-01-0500"
}`

In Phoscon it still shows the sensor as not reachable, any tips?
[image]

Smanar, 2022-03-25:
Ha yes I m seing missing field in the DDF.
Can you copy / paste the one you are using (exaclty the same) I will update it.

(miss for exemple the enrolement part)

timtatam, 2022-03-25:
I used the one which you directed me to:
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_msl6wxk9",
  "modelid": "TS0202",
  "product": "TS0202",
  "sleeper": true,
  "status": "Bronze",
  "path": "/devices/aliexpress/ts0202.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0107",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0406",
          "0x0001"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 0,
          "max": 0
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 0,
          "max": 0
        }
      ]
    }
  ]
}`

Smanar, 2022-03-26:
Can you try this one

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_msl6wxk9",
  "modelid": "TS0202",
  "product": "_TZ3000_msl6wxk9 Presence Sensor",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/tuya/_TZ3000_msl6wxk9_presence_sensor.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0107",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0406",
          "0x0001"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 35,
            "fn": "zcl",
            "script": "develco_battery.js"
          }
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/tampered",
          "awake": true
        },
        {
          "name": "state/presence",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 0,
          "max": 0
        }
      ]
    }
  ]
}
```

- I have set the Status to gold
- have modified the path and the name , for future emplacement   `"path": "/devices/tuya/_TZ3000_msl6wxk9_presence_sensor.json",`
- have added missing filed
- have removed the attribute reporting

I m not sure for the attribute reporting, I have checked some sampe, and they don't use it, but it's device dependent, if you need it.
- go on DDF editor, tab "binding"
- select the cluster 0x0500
- drag and drop the attribute 0x0002 for the same cluster on the right panel
- Configure min 300, max 3600

To have something like

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 300,
          "max": 3600
        }
      ]
    },
```

Mimiix, 2022-04-10:
Reopening as there's not yet a pr made for this device.

KrOnAsK, 2022-04-11:
Can anyone create a PR for this device because I got the same one and its far beyond my level of competency to create a DDF and  all of that

Mimiix, 2022-04-11:
@KrOnAsK In the mean time: You can perhaps make it work locally. How do you run deCONZ? Do you use the HA Addon or do you have a local deCONZ running?

KrOnAsK, 2022-04-11:
I have HA running on my Pi3 and there I got the Addon running

Mimiix, 2022-04-11:
@Smanar  Do you know how people added a DDF to the addon>?

Smanar, 2022-04-11:
I know some have succed https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459#issuecomment-1018984004

But I m asking too for another issue (ha don't have the same files), and still no answer from the discord.

KrOnAsK, 2022-04-12:
@Smanar can u create a PR?

closed 2022-05-12 by github-actions[bot]

jostie123, 2022-05-19:
> Hu, not realy, I prefer someone test the DDF first.

so i'm trying to test this but the first time i'm trying to recreate a ddf i im pretty far but am not able to find where i can configure the fingerprint part mentioned in [this comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5584#issuecomment-1079665733) is it important?

jostie123, 2022-05-19:
ok got the most important function working it's detecting motion.

only the battery stays at 0%

this is my current DDF Preview
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_msl6wxk9",
  "modelid": "TS0202",
  "product": "_TZ3000_MSL6wxk9 Presence sensor",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/tuya/_TZ3000_msl16wxk9_presence_sensor.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 35,
            "fn": "zcl",
            "script": "develco_battery.js"
          },
          "default": 0
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/enrolled",
          "public": false,
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/presence",
          "awake": true,
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1"
          }
        },
        {
          "name": "state/tampered",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 300,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 0,
          "max": 0
        },
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 0,
          "max": 0
        }
      ]
    }
  ]
}`

Smanar, 2022-05-20:
Hello can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_msl6wxk9",
  "modelid": "TS0202",
  "product": "_TZ3000_MSL6wxk9 Presence sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0107",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0500",
          "0x0001"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery"
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/enrolled",
          "public": false,
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/presence",
          "awake": true,
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1"
          }
        },
        {
          "name": "state/tampered",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 300,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

All values are 0 in the screenshot for the power configuration cluster 0x0001, so I have take the attribute 0x0021 (if you can verify using the GUI)

BTW are you sure you need the report for the cluster 0x0500 ? Some device support badly it

```
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 300,
          "max": 3600
        }
```

jostie123, 2022-05-20:
> All values are 0 in the screenshot for the power configuration cluster 0x0001, so I have take the attribute 0x0021 (if you can  verify using the GUI)

when i read the sensor the power configuration is as below.
[image]

> BTW are you sure you need the report for the cluster 0x0500 ? Some device support badly it
i am not sure of anything :P only thing i can say now is that the motion is working properly for last 24 hours. i removed the report and seems to stil work properly

i am still just messing around in the settings trying to match up the json you provide with mine... is it possible to just copy paste the above somewhere? because in the previewwindow it's not doing it. and because of the home assistant add-on i can not upload any .json files through the VNC

Anyway this is what i got now

```{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_msl6wxk9",
  "modelid": "TS0202",
  "product": "_TZ3000_MSL6wxk9 Presence sensor",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/tuya/_TZ3000_msl16wxk9_presence_sensor.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "default": 0
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/enrolled",
          "public": false,
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/presence",
          "awake": true,
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1"
          }
        },
        {
          "name": "state/tampered",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 0,
          "max": 0
        }
      ]
    }
  ]
}

Smanar, 2022-05-21:
You forget to set the minimum and maximum value for the report (in the binding tab).
The problem si lot of IAS device don't work without the fingerprint (but if it work on your machine)

Else on HA you can traval on file and edit the json file  https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

jostie123, 2022-05-29:
oke so i finally had the time to figure this out 
 
> Else on HA you can travel on file and edit the json file https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

i am not an expert in command stuff so i am probably just missing something here but once i find the .json files that way how can i edit them? i tried commands like "vim" and "nano" but none of them seem to work they al give "command not found".

anyway entered the current min and max now. also removed the tampered state because i don't believe this is implemented in the motion sensor.

the battery say 100 now. is there any way to test is this is a real value?

Current DDF
```{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_msl6wxk9",
  "modelid": "TS0202",
  "product": "_TZ3000_MSL6wxk9 Presence sensor",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/tuya/_TZ3000_msl16wxk9_presence_sensor.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "default": 0
        },
        {
          "name": "config/duration"
        },
        {
          "name": "config/enrolled",
          "public": false,
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/presence",
          "awake": true,
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    }
  ]
}```

Smanar, 2022-05-30:
>i am not an expert in command stuff so i am probably just missing something here but once i find the .json files that way how can i edit them? i tried commands like "vim" and "nano" but none of them seem to work they al give "command not found".

Good question, they said using "Samba" for file management or "docker cp".
An user have made a more detailled procedure but I can't find it again (perhaps with "cat")

BTW if all is working you can make a PR or wana I make it ?

jostie123, 2022-05-31:
You can do it if you want.

modestpharaoh, 2023-03-17:
I recently got one of this motion sensor _TZ3000_msl6wxk9 / TS0202 ... have these problems with the DDF included in Deconz:
 * battery reported as 0.
 * Motion sensor not reported to home assistant

I updated the DDF by checking other similar sensor, and everything work fine:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZ3000_msl6wxk9",
    "_TZ3000_otvn3lne",
    "_TZ3040_6ygjfyll"
  ],
  "modelid": [
    "TS0202",
    "TS0202",
    "TS0202"
  ],
  "product": "TS0202 Presence sensor",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/_TZ3000_msl6wxk9_presence_sensor.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "default": 1
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/presence",
          "awake": true,
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    }
  ]
}

```

Smanar, 2023-03-18:
I m comparing with the actual file https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/ts0202_presence_sensor.json

So you have added some stuff in config/battery.
But I don't  see the change for presence detection ?