dresden-elektronik/deconz-rest-plugin #5585: Pairing Aqara Smart Wireless Switch WXKG11LM

nielsras, 2021-12-17:
Hello,

Unfortunately I'm not able to pair it in the Phoscon App.

Server: Ubuntu20.04
DockerImage:  deconz-community/deconz-docker
Version: 2.13.03 / 9-12-2021
Produkt: ConBee II
Firmware: 26720700
Connecting it as a sensor or switch doesn't make any difference.

Deconz VNC shows it's connected.
[image]

[image]

It's also not possible to view incomming deconz_events in HomeAssistant.
Other Aqara sensors connects without any problem. 

Does anyone have a useful tip for me?

Thanks in advance!

closed 2021-12-17 by close-issue-app[bot]

Mimiix, 2021-12-17:
Looks like it's unsupported.

FuzzyMistborn, 2021-12-20:
FWIW I'm having similar issues.  I have several of these buttons and they've paired fine before in the past.  Perhaps something is different with the new batch.

Mauricevb, 2021-12-20:
I have the same problem unfortunately. I don’t see any event when pressing the Aqara button(s). It worked until today

FuzzyMistborn, 2021-12-21:
My existing buttons still work, only pairing new buttons does not.

Mimiix, 2021-12-21:
@FuzzyMistborn can you share a screenshot of the basic cluster of both devices?

FuzzyMistborn, 2021-12-21:
Sure.  Throwing in the node info too in case that at all helps.

[image]
[image]
[image]

Mimiix, 2021-12-21:
Your sharing the node info. I'd like the cluster info. 

Check the wiki on how to read the basic cluster.,

hille721, 2021-12-22:
Hi,

I had exactly the same issue. It seems to be that it's a problem with the new Phoscon version 2.13.X. 
I just downgraded to 2.12.06 and the switch was pairing immediately!

Mimiix, 2021-12-22:
@hille721 Pairing always works. it might be useful to see _what_ exactly didn't work. How did it show in deCONZ?

hille721, 2021-12-22:
@Mimiix : I've updated deconz to 2.13.04 again. Switch was still working, I deleted it and tried to add it again, same problem:

In Phoscon it looks no devices are discovered if you search for new switches (thus in Phoscon it is looking as pairing is not working). In deconz I see following:

[image]

[image]

hille721, 2021-12-22:
What I can see in the logs (INFO):
```
13:02:05:413 send permit join, duration: 65
13:02:19:315 0xFCFC nwk changed to 0xDEEA
13:02:19:316 device announce 0x00158D0006D4DEF5 (0xDEEA) mac capabilities 0x80
13:02:19:317 set fast probe address to 0x00158D0006D4DEF5 (0xDEEA)
13:02:19:317 FP indication 0x0000 / 0x0013 (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:317                       ...     (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:318 device announce 0x00158D0006D4DEF5 (0xDEEA) mac capabilities 0x80
13:02:19:336 DEV found DDF for 0x00158D0006D4DEF5, path: 
13:02:19:344 FP indication 0x0104 / 0x0000 (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:345                       ...     (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:345 ZCL attribute report 0x00158D0006D4DEF5 for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
13:02:19:534 FP indication 0x0104 / 0x0000 (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:535                       ...     (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:536 Remember Xiaomi special for 0x00158D0006D4DEF5
13:02:19:537 ZCL attribute report 0x00158D0006D4DEF5 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
13:02:19:537 0x00158D0006D4DEF5 extract Xiaomi special attribute 0xFF01
13:02:19:538 	01 battery 3145 (0x0C49)
13:02:19:539 	03 Device temperature 29 °C
13:02:19:539 	04 unknown 424 (0x01A8)
13:02:19:540 	05 RSSI dB (?) 108 (0x006C)
13:02:19:540 	06 LQI (?) 4294967296 (0x0100000000)
13:02:19:541 	0a Parent NWK 60247 (0xEB57)
13:02:20:213 [1] get node descriptor for 0x00158d0006d4def5
13:02:20:214 ZDP get node descriptor for 0xDEEA
13:02:24:414 saved node state in 0 ms
13:02:24:415 sync() in 0 ms
13:02:24:889 FP indication 0x0000 / 0x8002 (0x00158D0006D4DEF5 / 0xDEEA)
13:02:24:889                       ...     (0x00158D0006D4DEF5 / 0xDEEA)
13:02:24:890 ZDP indication search sensors 0x00158D0006D4DEF5 (0xDEEA) cluster 0x8002
13:02:24:890 ZDP indication search sensors 0x00158D0006D4DEF5 (0xDEEA) clear timeout on cluster 0x8002
13:02:24:905 [2] get active endpoints for 0x00158d0006d4def5
13:02:24:906 ZDP get active endpoints for 0xDEEA
13:02:25:654 ZCL attribute report 0x00158D000774335F for cluster: 0x0201, ep: 0x01, frame control: 0x1C, mfcode: 0x1037 
13:02:28:101 FP indication 0x0000 / 0x8005 (0x00158D0006D4DEF5 / 0xDEEA)
13:02:28:102                       ...     (0x00158D0006D4DEF5 / 0xDEEA)
13:02:28:102 ZDP indication search sensors 0x00158D0006D4DEF5 (0xDEEA) cluster 0x8005
13:02:28:103 ZDP indication search sensors 0x00158D0006D4DEF5 (0xDEEA) clear timeout on cluster 0x8005
13:02:28:115 [4] Skipping additional attribute read - Model starts with 'lumi.'
13:02:31:683 Bind response success for 0x00158d000774335f ep: 0x01 cluster: 0x0001
```

Mimiix, 2021-12-22:
forwarded it to Manup. I can't open this issue as it's template is not used.

hille721, 2021-12-22:
where should I open it. Here or because it seems to be more Phoscon related, diretly there: https://forum.phoscon.de?

Mimiix, 2021-12-22:
It has nothing to do with phoscon. Needs to be in this repo.