dresden-elektronik/deconz-rest-plugin #5590: Half brightness once Sengled LED is linked to deCONZ

steve30avs, 2021-12-20:
## Describe the bug

When a new Sengled E12-N1E color LED is added through Phoscon, the peak brightness drops to roughly half of what it was.

## Steps to reproduce the behavior

1. Plug in new Sengled E12-N1E or reset it to factory defaults (10 times on/off rapidly)
2. Search for new lights on the Phoscon website
3. Once the new bulb is added, watch its brightness
4. It should be roughly half of what it just was and cannot be raised higher through the brightness slider

## Expected behavior

Dragging the brightness slider to 100% should produce the whole 1300lm of brightness

## Screenshots

Photo of the connected bulb (left) vs an unconnected bulb (right):
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26660700)
 - deCONZ version: (2.13.03)
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? Yes, USB mouse

## deCONZ Logs

INFO_L2 log:
11:39:37:993 Websocket 192.168.1.230:42322 send message: {"e":"changed","id":"19","r":"lights","state":{"alert":null,"bri":254,"colormode":"xy","effect":"none","hue":0,"on":true,"reachable":true,"sat":0,"xy":[0.402,0.3768]},"t":"event","uniqueid":"b0:ce:18:14:03:53:83:94-01"} (ret = 219)

## Additional context

All other connected bulbs (Ikea, Philips, and other Sengleds, appear to reach full brightness fine)

steve30avs, 2021-12-24:
Update: It looks like the lights reach full brightness when on 'Color Temperature' mode through Phoscon, but cannot reach full brightness when selecting a color.

Smanar, 2021-12-24:
Ha yes, I already see this issue on extented light.
It s because the device can use different color mode, XY or Hue/Sat/Bri.
It will be better if the "colormode" was "hs".

Are you able to try to change the color yourself using the API, using H/S/B instead of XY ? https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

But compare with your workings bulbs (if they have too "XY" and "hue","sat", what is the colormode used by them ? If they use "HS", try to re-include the device, or force deconz to update the value with playing with the GUI on the color cluster.

steve30avs, 2021-12-26:
So using the API I discovered that my other Sengled bulbs also appear to dim a bit, vs adjusting the color temperature settings in the GUI.
I tried modifying both the XY and H/S/B values of another Sengled and Philips Hue bulb, and only the Philips Hue bulb was consistently bright.
All bulbs appear to be using both XY and H/S/B, as seen here for my Philps bulb:
`        "state": {
            "alert": "none",
            "bri": 254,
            "colormode": "hs",
            "ct": 219,
            "effect": "none",
            "hue": 29952,
            "on": true,
            "reachable": true,
            "sat": 254,
            "xy": [
                0.36,
                0.4213
            ]
        }`

I tried to re-include the device into the group which did not seem to work. It was interesting that it reverted back to full brightness when I removed it from the group, and retained the brightness up until I tried changing the Hue in the H/S/B sliders in the GUI, at which point it dimmed again.

So it sounds like this is either an issue with how the Sengled bulbs are mapped, or that we should just use the Color Temperature slider from now on for these bulbs.

Smanar, 2021-12-26:
Honnestly I don't know enought the bulb working mode in deconz but that I know :
- deconz make convertion you can send for exemple XY and deconz send HSB, can can have "unknown magic magic"
- some "unknown magic" can be solved with a different "colormode"
- bulbs don't react same at all color requests
- On an extented light I need to send in the json Hue, Sat, Bri AND Ct

>I tried to re-include the device into the group which did not seem to work

Not in the group, in the network, to try to have a different  "colormode" value. but idk what happen if all bulbs in a group don't use same color request.

steve30avs, 2021-12-30:
I tried your idea of keeping the light in the same network, but in a different group, and that actually worked! 
I can actually adjust the H/S/B when this light bulb is in a different group now.

So it appears that the original group itself is acting strange or just isn't sharing the same color request as you have mentioned.
I may try breaking up the groups further, as this original group had 8 bulbs in it, so it wouldn't hurt to do anyway.

Still a weird issue either way, but quite minor as well.

Mimiix, 2022-01-03:
@Smanar  Seems that we can have this fixed with DDF?

Smanar, 2022-01-03:
IDK, need to try with a DDF to force the "colormode" to "hs" instead of "xy" to see if it change something.