dresden-elektronik/deconz-rest-plugin #5591: Blizwolf shp15 not power cosumation on HA and state on/off not work

manustars, 2021-12-20:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi )
 - Running method: (Raspbian )
 - Firmware version: (26690700)
 - deCONZ version: (2.12.06)
 - Device: (RaspBee II)
 - 
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Mimiix, 2021-12-20:
Hi

Please update to the latest version as support was added  after 2.12.06

manustars, 2021-12-20:
now i see the power, the name changed after what I see it in ha? is it automatic or must it be updated manually on ha?

Mimiix, 2021-12-20:
I think automatically. Either way as this is not a bug on the device / implementation, i'm closing it.

closed 2021-12-20 by Mimiix

manustars, 2021-12-20:
yes, thanks