dresden-elektronik/deconz-rest-plugin #5593: Ikea TRADFRI bulb E27 WS globe opal 1055lm

maxnumberone, 2021-12-20:
Hello,
could you please add the bulb IKEA TRADFRI LED 1055lm ?

- Product name: IKEA TRADFRI LED 1055lm
- Manufacturer: IKEA of Sweden
- Model identifier: TRADFRIbulbE27WSglobeopal1055lm
- Device type : Please remove all unrelated device types. 
  - Light

## Screenshots

[image]

[image]

Thank you in advance,
Max

Mimiix, 2021-12-20:
Looks like it's already supported. What is not working?

maxnumberone, 2021-12-20:
it is, but the color temperature is not working fine from the phoscon app: I've assigned a switch command to increase and another one to decrease the color temperature but both commands increases it for this bulb.
in the same group there's a philips hue white ambiance and this one works fine.

Mimiix, 2021-12-20:
Does it work fine with the rest api? Or from something like home assistant?

maxnumberone, 2021-12-20:
yes it's ok from jeedom

Mimiix, 2021-12-20:
Then it's not a deconz bug. Please report it on the forums. There perhaps already are bugreports with scenes and ikea https://forum.phoscon.de

closed 2021-12-20 by Mimiix