dresden-elektronik/deconz-rest-plugin #5594: SNZB-03 fake motion reports

sk8er000, 2021-12-21:
## Describe the bug

 I have a couple of SNZB-03 motion sensors in a dark room with no windows or heating sources and the door is always closed, about every hour the sensors detect a motion (but there is no motion or repentine temparature change). Those sensors were previously associated to an Athom Homey Pro and both worked fine.

## Steps to reproduce the behavior

Pair a SNZB-03 with deConz 
Put a sensor in a place where there is no movement and wait for a motion report

## Expected behavior

Motion should be triggered only when there is a motion event

## Screenshots

## Environment

 - Host system: Raspberry Pi4 8gb
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26720700
 - deCONZ version: 2.13.04 / 18/12/2021
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? NO
 - 
## deCONZ Logs

## Additional context

Mimiix, 2021-12-21:
Hi,

This is something related to the device. It seems to be just a crappy device that does false positives. https://www.reddit.com/r/sonoff/comments/le6gj6/snzb03_dont_buy_it and https://tech.scargill.net/another-itead-sonoff-week/#comment-69873

Not a lot we can do if the device is not doing what it should.

sk8er000, 2021-12-21:
I understand what you mean, the strange part is that those sensors worked fine for months when paired with a Homey Pro which is not famous to have a reliable ZigBee environment

Mimiix, 2021-12-21:
There's not much i can do about it.  https://github.com/Koenkk/zigbee2mqtt/issues/4874 Seems that there is something going on, but it still is odd in the way this is working. 

The funny thing is, other Sonoff devices don't work different or strange.

sk8er000, 2021-12-21:
I have some SNZB-04 that seem to be reliable under deConz, those were one of the reason why I decided to migrate to deConz since under Homey sometimes get stuck on ON or OFF state. SNZB-03 is the opposite...
So if nothing can be done on deConz side the issue can be closed since it can't be solved.
Thank you for the explanation.

Mimiix, 2021-12-21:
@sk8er000 The SNZB-04 i have working at home and seem to be fine. 

Perhaps there's some other issues in your network. Feel free to open a post on the forum with some logs, so we can check it out.

http://forum.phoscon.de

I'm closing this issue on your request.

closed 2021-12-21 by Mimiix