dresden-elektronik/deconz-rest-plugin #5595: Tradfri GU10 bulbs restart/rejoin after firmware updates

justindekeizer, 2021-12-21:
## Describe the bug
Updated Tradfri firmware rejoins the Zigbee network after updating. I have updated 5 of my many bulbs, non updated do not show this behavior. All updated do show this behavior.

## Steps to reproduce the behavior
1. Update Tradfri GU10 bulb to latest firmware via OTAU
2. Switch on the lights
3. After 5-6min lights will start to go OFF and ON again every 5-6min

## Expected behavior
Normal behavior (no blinking)

## Screenshots
n/a

## Environment

 - Host system: PC
 - Running method: (Home Assistent deCONZ Add-on
 - Firmware version: (26xxyy00)
 - deCONZ version: 2.13.4.
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
`08:40:45:623 Turn on light 0xEC1BBDFFFE6CA093 on again with former brightness after powercycle 
08:40:45:623 DeviceAnnce of LightNode: 0xec1bbdfffe6ca093 Permit Join: 0 
08:40:45:723 ZCL got data for node=0x6A58, cl=0x0006, at=0x0000, status=0x00, type=0x10 
08:40:46:023 ZCL attribute report 0xEC1BBDFFFE6CA093 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 08:40:46:700 0xEC1BBDFFFE6CA093 error APSDE-DATA.confirm: 0xE9 on task 08:40:46:712 0xEC1BBDFFFE6CA093 force poll (2)`

## Additional context
Reported this first to HA addon (https://github.com/home-assistant/addons/issues/2315)

Mimiix, 2021-12-21:
It seems to have powercycled. That's why it blinks.

I forwarded this to the devs/manup anyway, as it seems the firmware changed some stuff.

Mimiix, 2021-12-21:
@ebaauw wrote: Make sure to re-read the node descriptors and Basic cluster after a firmware update, or deCONZ won't reflect the new data.

justindekeizer, 2021-12-21:
> It seems to have powercycled. That's why it blinks.
> 
> I forwarded this to the devs/manup anyway, as it seems the firmware changed some stuff.

Yes. Please note that the log entry with the powercycle is new since the addon was bumped to the 2.13.4 release last night. Before that (and after the firmware update) the behavior was identical but the logs did not mention the powercycle but did mention the other lines. So either is an added log feature/verbose setting that wasn't there in the previous version or the detection is different in this release. Behavior is identical.

Mimiix, 2021-12-21:
> 

Wait, you are confusing me now. What is the exact report? That they are in the logs? It is normal for a device to reboot after a firmware update.

justindekeizer, 2021-12-21:
> > 
> 
> Wait, you are confusing me now. What is the exact report? That they are in the logs? It is normal for a device to reboot after a firmware update.

The issue started after updating the firmware. It almost seems like the lights fall into a bootloop of some sort after updating or that Deconz keeps sending a reboot command to them (could not find any in the logs).

Situation:
All Lights are off
I switch the lights (or groups containing the lights on)
Updated lights will turn OFF and ON again after about 5-6 min of beeing ON. This repeats at the same interval.
All not updated lights remain ON the entire period.

The logs posted above are logged at the moment the lights turn OFF without any command issuing that behavior. So note, there are 0 actions on light state on that moment.

> @ebaauw wrote: Make sure to re-read the node descriptors and Basic cluster after a firmware update, or deCONZ won't reflect the new data.

Did all that. Rebooted the entire chain, powercycled all bulbs. Rejoined the bulbs manually to the network, issue persists.

ebaauw, 2021-12-21:
How much time between tuning off and on?  Do you see the _Device Announcement_ message in the deCONZ log before or after the light turns off?

justindekeizer, 2021-12-21:
> How much time between tuning off and on? Do you see the _Device Announcement_ message in the deCONZ log before or after the light turns off?
It switches OFF for about 2-3 seconds, haven't timed it exactly.

As soon as I see the light switch I refresh the log and then the log shows the entry but the light is also ON again. Can't answer this with certainty but it feels/looks like the light switches ON after the Announcement.

> Did you try and factory-reset and re-pair the light after the firmware update?
Re-pair yes. (by resetting the bulb)
Factory-reset, i'm not sure if the repair/reset does a full factory reset. I was actually looking for a method to restore earlier firmware to try that, haven't been able to.

ebaauw, 2021-12-21:
Indeed, this smells like the light firmware is rebooting, causing the spot to turn off.  Notice the new NWK address.  I think deCONZ turns it back on, due to some poweron setting (that I've never used myself).

Resetting the spot (using the power on/off sequence) should factory-reset the non-volatile memory, as far as I know.

Not sure why the firmware is rebooting.  Could be that the new firmware is a bit more sensitive to power fluctuations.  Normally I'd recommend checking the electric wiring.  Do all spots reboot at the same time?  Do they share a traditional wall switch?  Are they on the same breaker circuit?  Did you try swapping an upgraded spot with a not yet upgraded one?

But it could also very well be a firmware issue.  I'm not sure if IKEA supports downgrading firmware, but you might try that, if you still have the old firmware file.

Since this happens every 5-6 minutes, best bet is that the firmware chokes on a Zigbee message it receives.  You might be able to check that using the deCONZ log, with all sorts of debug options enabled, but I would use a Zigbee sniffer.  We would need to identify the message, before we can assess whether that's fixable/can be worked around on the deCONZ side.

justindekeizer, 2021-12-21:
@ebaauw Thanks for your reply and effort!

> Not sure why the firmware is rebooting. Could be that the new firmware is a bit more sensitive to power fluctuations. Normally I'd recommend checking the electric wiring. Do all spots reboot at the [...]

I'm pretty certain that its not powerfeed related, one of the spots is now in my 'deployment' cabling and still has the same issue. The remaining issue spots are on the same groups/cabling as non-issue spots. Most unfortunately I have updated single spots in different light zones so I now have blinking lights in most parts of the house 🗡️ should have done an entire single zone instead...

I'm also on the hypothesis that its related to some sort of choke/reset issued by a memory/command at interval issue. 

Can you suggest a sniffer or log setting to get more detail? Haven't been able to find downgrade instructions.

ebaauw, 2021-12-21:
I use [ZShark](https://www.dresden-elektronik.com/wireless/software/zshark.html) for capturing and [WireShark](https://www.wireshark.org) for display and analysis.  You need a second device to capture deCONZ traffic.  I think ZShark now also supports the ConBee II, next to the original ConBee and RaspBee (according to Help|Dependencies).

As for debug options, I'd go for INFO, INFO_L2, ERROR, ERROR_L2, APS, APS_L2, ZDP, ZCL.  Probably overkill, but we don't know what type of message it is.  If it's a unicast by another device, deCONZ might not even receive it.

To downgrade the device firmware, simply select the device, open the old firmware file and press _Update_ in the STD OTAU plugin panel.  Not all devices support downgrading, though.

justindekeizer, 2021-12-21:
Bummer don't have any spare Zigbee sticks yet.

Debug changed : event starts at 16:03:48:098
`
16:03:47:957 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 9, node: 0x33AC
16:03:47:957 MgmtBind_rsp id: 23 0xec1bbdfffe6ca093 seq: 47, status 0x00 
16:03:47:957 found binding 0x0008, 0x01 -> 0x00212EFFFF06BE1B : 0x01
16:03:47:957 add binding to check rule queue size: 0
16:03:47:957 found binding 0x0300, 0x01 -> 0x00212EFFFF06BE1B : 0x01
16:03:47:958 add binding to check rule queue size: 1
16:03:48:098 APS-DATA.indication srcAddr: 0xb323, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 164, rssi: -87
16:03:48:098 	asdu: 8123b3c5a859feffac33bc8e
16:03:48:098 ZDP device announce: 0xBC33ACFFFE59A8C5, 0xB323, 0x8E
16:03:48:098 ZDP add fast discover for 0xbc33acfffe59a8c5
16:03:48:098 Turn off light 0xBC33ACFFFE59A8C5 again after powercycle
16:03:48:099 DeviceAnnce of LightNode: 0xbc33acfffe59a8c5 Permit Join: 0
16:03:48:100 Websocket 172.30.32.1:60986 send message: {"attr":{"colorcapabilities":31,"ctmax":454,"ctmin":250,"id":"11","lastannounced":"2021-12-21T15:03:48Z","lastseen":"2021-12-21T15:03Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb GU10 CWS 345lm","name":"Aiden 02","swversion":"1.0.021","type":"Extended color light","uniqueid":"bc:33:ac:ff:fe:59:a8:c5-01"},"e":"changed","id":"11","r":"lights","t":"event","uniqueid":"bc:33:ac:ff:fe:59:a8:c5-01"} (ret = 412)
16:03:48:101 Websocket 172.30.32.1:60986 send message: {"e":"changed","id":"11","r":"lights","state":{"alert":null,"bri":230,"colormode":"ct","ct":454,"effect":"none","hue":0,"on":false,"reachable":true,"sat":228,"xy":[0.5017,0.4153]},"t":"event","uniqueid":"bc:33:ac:ff:fe:59:a8:c5-01"} (ret = 232)
16:03:48:129 APS-DATA.request id: 45, addrmode: 0x02, addr: 0xb323, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 3 tx.options 0x04
16:03:48:129 	asdu (length: 3): 11ce00
16:03:48:129 Delay APS request id: 45 delayed, 1000 ms till send
16:03:48:179 Delay APS request id: 45 delayed, 1000 ms till send
16:03:48:225 ZDP finished fast discover for 0xbc33acfffe59a8c5
16:03:48:228 Delay APS request id: 45 delayed, 1000 ms till send
16:03:48:267 poll node bc:33:ac:ff:fe:59:a8:c5-01
16:03:48:268 Poll light node <removed name> 02
16:03:48:279 Delay APS request id: 45 delayed, 1000 ms till send
16:03:48:320 read attributes of 0xBC33ACFFFE59A8C5 cluster: 0x0006: [ 16:03:48:320 0x0000 16:03:48:320 ]
16:03:48:321 add task 721 type 19 to 0xBC33ACFFFE59A8C5 cluster 0x0006 req.id 47
16:03:48:321 Poll APS request 47 to 0xBC33ACFFFE59A8C5 cluster: 0x0006
16:03:48:329 APS-DATA.request id: 47, addrmode: 0x03, addr: 0xbc33acfffe59a8c5, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
16:03:48:329 	asdu (length: 5): 10cf000000
16:03:48:329 Delay APS request id: 45 delayed, 800 ms till send
16:03:48:330 Delay APS request id: 45 delayed, 800 ms till send
16:03:48:345 Poll APS confirm 47 status: 0x00
16:03:48:346 Erase task req-id: 47, type: 19 zcl seqno: 207 send time 0, profileId: 0x0104, clusterId: 0x0006
16:03:48:346 APS-DATA.confirm id: 47, status: 0x00 SUCCESS
`

ebaauw, 2021-12-21:
That's not helping.  You need to analyse the messages to and from the spot, from before the Device Announcements, hoping to find a pattern.  Challenge is that the NWK address changes on Device Announcement, and some messages will be logged under the NWK address, others under the MAC address.  Also note that the message contents (`asdu`) are logged on a separate line, so simple filtering using `grep` is not gonna cut it.

Do the spots also blink when they're off?  It looks like deCONZ is actually turning them off.
```
16:03:48:098 Turn off light 0xBC33ACFFFE59A8C5 again after powercycle
...
16:03:48:129 APS-DATA.request id: 45, addrmode: 0x02, addr: 0xb323, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 3 tx.options 0x04
16:03:48:129 	asdu (length: 3): 11ce00
```
What setting did you use to enable this function?  It shouldn't even be needed, as IKEA support setting _PowerOn On/Off_ in recent firmware versions.

justindekeizer, 2021-12-21:
> That's not helping. You need to analyse the messages to and from the spot, from before the Device Announcements, hoping to find a pattern. Challenge is that the NWK address changes on Device [...]

Ah sorry, i'm getting the logs from the HA plugin page currently. I will open API access and reroute them to a permanent location to gather more details and history to analyse the pattern.

> Do the spots also blink when they're off? It looks like deCONZ is actually turning them off.

No they are simply off. Then (looking at the logs) they are reverted by Deconz to their last state prior to the rejoin uppon completion of that rejoin.

> ```
> 16:03:48:098 Turn off light 0xBC33ACFFFE59A8C5 again after powercycle
> ...
> 16:03:48:129 APS-DATA.request id: 45, addrmode: 0x02, addr: 0xb323, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 [...]

I didn't enable anything for that. More specific, I didn't change anything to the configuration of the bulbs, deconz, groups or anything related besides updating the firmware of the selected bulbs before the issue started. The issue also started right after updating.

I have updated to the latest HA deconz add-on version last night but had the issue before and after that.

The lights don't blink

ebaauw, 2021-12-21:
I really don't understand where this "Turn off light ... after power cycle" comes from.  Looking at the code, this feature should be for the FLS-NB only.

What does the _Basic_ cluster look like after the firmware update?

What does the REST resource look like?

justindekeizer, 2021-12-21:
Bulb with issues:
[image]
Bulb without issues:
[image]

NOTE! It reports the same software version in Deconz..

The OTAU entry for the issue bulb shows the following:
[image]
The OTAU entry for the non issue bulb show the following:
[image]

Note, I removed the update file to prevent accidental further deployment of the firmware.

How/what should I look at regarding the REST resource? Do you mean the following from Phoscon APP?
[image]

Regarding the software levels reported by the bulbs, its almost as if they havent run through the full update process. Could they be in some sort of pre/post update bootloop or something like that and are just rebooting every 5-6min pending completion?

Will do a 'factory-reset' of the issue bulb later this evening and report back.

justindekeizer, 2021-12-21:
OK so did some troubleshooting by completely resetting the bulbs (proved a challenge due to the location of some of them and the IKEA procedure) but so far, so good! Lights remain on, no entries in the log, functions as expected.

I think the bulbs have been stale in some sort of update check/post upgrade mode that made them behave the way they did. I did remove them from the powerfeed completely earlier but it seems they need the full reset cycle for them to come back to normal operation. 

Was not a deconz issue (besides perhaps the failed/stale update but I don't have any logs of that).

closed 2021-12-21 by justindekeizer

ebaauw, 2021-12-21:
> I think the bulbs have been stale in some sort of update check/post upgrade mode that made them behave the way they did.

As I said above, they might have changed the data structures in non-volatile memory, so the new firmware cannot handle the old structure.  A factory reset re-initialises the non-volatile memory.