dresden-elektronik/deconz-rest-plugin #5596: ubisys J1 - sensors not reachable after update to 2.13.04 / 18.12.2021 (26720700)

jan666, 2021-12-21:
## Describe the bug
I updated deCONZ to 2.13.04 / 18.12.2021 and my ConBee II to 26720700 yesterday. After that all /sensor/s (ZHAConsumption, ZHAPower, ZHASwitch) of my ubisys J1 are reachable: false. Only the /light/-Resource is still reachable (so its still possible to control the awning.

I never saw the sensors unreachable (over one year) but now they are - after the update. So this might be a problem with the new version?

## Steps to reproduce the behavior

## Expected behavior

The sensors should be reachable (again).

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26720700
 - deCONZ version: (2.13.04)
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: no

## deCONZ Logs

## Additional context

Mimiix, 2021-12-21:
Hi,

As this is not a bug with the software itself, but more of a user question, I am closing this issue (please read #5113 to check when something is classified as a bug).

Feel free to open an post on the forum: http://forum.phoscon.de

Mimiix, 2021-12-23:
After checking the setup, and confirming it with another user, it seems there is an bug in the codebase. @manup  has one for testing so he'll check on his side. 

We are proceeding to log the issue here.

martind1983, 2022-01-26:
HI, 

I also updated some days ago from version 2.12.06, I had no issues with the old firmware, but thought it make sense to update to 2.13.04.
Since my update my Zigbee LED Dimmer (so not only sensors as other people are reporing) is sometimes on a minute base not reachable. I did not change anything else, just updated the firmware and all plugins on my homebridge to the latest version.

these are the homebridge logs (or a few lines of it):
[Hue] Bad LED Leiste: attr changed event: 

{"id":"3","lastannounced":null,"lastseen":"2022-01-23T12:00Z","manufacturername":"Philips","modelid":"5112.80","name":"Bad LED Leiste","swversion":"2.5.3_r55","type":"Dimmable light","uniqueid":"80:4b:50:ff:fe:b7:30:3e-01"}

[Hue] Bad LED Leiste: attr changed event: {"id":"3","lastannounced":null,"lastseen":"2022-01-23T12:01Z","manufacturername":"Philips","modelid":"5112.80","name":"Bad LED Leiste","swversion":"2.5.3_r55","type":"Dimmable light","uniqueid":"80:4b:50:ff:fe:b7:30:3e-01"}

Examples:
[23/01/2022, 13:07:40] [Hue] Bad LED Leiste: set homekit status fault from 0 to 1
[23/01/2022, 13:07:40] [Hue] Bad LED Leiste: set homekit status fault from 1 to 0
[23/01/2022, 13:21:36] [Hue] Bad LED Leiste: set homekit status fault from 0 to 1
[23/01/2022, 13:26:16] [Hue] Bad LED Leiste: set homekit status fault from 1 to 0
[23/01/2022, 13:55:59] [Hue] Bad LED Leiste: set homekit status fault from 0 to 1
[23/01/2022, 13:55:59] [Hue] Bad LED Leiste: set homekit status fault from 1 to 0
[23/01/2022, 13:58:10] [Hue] Bad LED Leiste: set homekit status fault from 0 to 1
[23/01/2022, 13:58:11] [Hue] Bad LED Leiste: set homekit status fault from 1 to 0
[23/01/2022, 13:58:44] [Hue] Bad LED Leiste: set homekit status fault from 0 to 1
[23/01/2022, 14:01:08] [Hue] Bad LED Leiste: set homekit status fault from 1 to 0

martind1983, 2022-01-28:
@manup  does this also fix the issue with the Philips 5112.80 dimmer? or do we need another ticket there?

jan666, 2022-02-07:
Nothing changed in 2.14.01 / 6.2.2022

jan666, 2022-02-09:
@Mimiix can you please reopen the issue?

Mimiix, 2022-02-09:
Sure.

jan666, 2022-03-09:
Any news on this issue?

Mimiix, 2022-03-09:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5675 Was fixed in 2.14.1 afaik.

jan666, 2022-03-09:
I am on 2.14.01. Issue is still there

manup, 2022-03-09:
Hmm strange I did test this for v2.14.1 worked here :thinking: 
I've put it on the list for a further testing round.

jan666, 2022-05-16:
It's almost fixed in 2.15.03 / 21.4.2022 (Firmware 26720700). All 3 sensors are reachable; `ZHAConsumption` and `ZHAPower` are updating:

```
{
  "config": {
    "on": true,
    "reachable": true
  },
  "ep": 3,
  "etag": "447ca8e0fc881a049137c2e5b413cf04",
  "lastannounced": null,
  "lastseen": "2022-05-16T05:19Z",
  "manufacturername": "ubisys",
  "modelid": "J1 (5502)",
  "name": "Markise",
  "state": {
    "consumption": 172,
    "lastupdated": "2022-05-16T05:19:26.820",
    "power": 0
  },
  "swversion": "20200429-DE-FB0",
  "type": "ZHAConsumption",
  "uniqueid": "00:1f:ee:00:00:00:51:4a-03-0702"
}
```

```
{
  "config": {
    "on": true,
    "reachable": true
  },
  "ep": 3,
  "etag": "d0760fce8b91b292075aabb079823046",
  "lastannounced": null,
  "lastseen": "2022-05-16T05:20Z",
  "manufacturername": "ubisys",
  "modelid": "J1 (5502)",
  "name": "Markise",
  "state": {
    "current": 0,
    "lastupdated": "2022-05-16T05:20:31.656",
    "power": 0,
    "voltage": 234
  },
  "swversion": "20200429-DE-FB0",
  "type": "ZHAPower",
  "uniqueid": "00:1f:ee:00:00:00:51:4a-03-0b04"
}
```

...but the `ZHASwitch` is not updating. I used it several times during the weekend, but:

```
{
  "config": {
    "group": "20000",
    "mode": "rocker",
    "on": true,
    "reachable": true,
    "windowcoveringtype": 0
  },
  "ep": 2,
  "etag": "c6226d278d84b80d86f345c6a54b1a9a",
  "lastannounced": null,
  "lastseen": "2022-05-16T05:20Z",
  "manufacturername": "ubisys",
  "mode": 1,
  "modelid": "J1 (5502)",
  "name": "J1 (5502) 34",
  "state": {
    "buttonevent": null,
    "lastupdated": "none"
  },
  "swversion": "20200429-DE-FB0",
  "type": "ZHASwitch",
  "uniqueid": "00:1f:ee:00:00:00:51:4a-02-0102"
}
```

faxioman, 2022-07-12:
I don't know if is the same problem, but after upgrading to 2.17.01 / 26720700, I cannot reach lights and sensor.
I'm probably really unlucky... but about one upgrade every three, requires a full re pair for all my devices :(

I was upgrading from 2.12.06.

I'm on Raspberry PI Zero W, Debian buster, Conbee II.

faxioman, 2022-07-12:
Reverting to the previous version (2.12.06 / 26720700) doesn't solve the problem.
Four hours of works are waiting for me :D
I really need to find a more stable upgrade flow...

faxioman, 2022-07-12:
Ok, different problem this time... I cannot find any device anymore... I'll open a new ticket, tomorrow.

Mimiix, 2022-07-12:
@faxioman best to do on the forum. Otherwise I'll probably close it if it doesn't stick to #5113.

manup, 2022-10-07:
Closing this for now due further fixes and DDF improvements in PR's: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6219, https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6092 and https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5894

Current DDF docs can be found in
[J1 (5502)](https://dresden-elektronik.github.io/deconz-rest-doc/devices/ubisys/j1_5502) and [J1-R (5602)](https://dresden-elektronik.github.io/deconz-rest-doc/devices/ubisys/j1r_5602).

closed 2022-10-07 by manup