dresden-elektronik/deconz-rest-plugin #5599: Pairing Aqara Smart Wireless Switch WXKG11LM not working anymore with version 2.13.X

hille721, 2021-12-22:
## Describe the bug
The Aqara Smart Wireless Switch (WXKG11LM), which was easy to add in previous version, can not be added anymore via Phoscon (see also #5585).

## Steps to reproduce the behavior
  1. Go to 'Switches"
  2. Click on 'Add Switch'
  3. Click on 'Others'
  4. Press the reset button of the switch for ~ 5 seconds
  
In deconz a new node was created, but in Phoscon nothing happens.

## Expected behavior
Pairing should work in Phoscon as it worked in the previous 2.12.06 version

## Screenshots
in Phoscon
[image]

in deconz
[image]

[image]

## Environment
 - Host system: Fedora 35 (Raspberry Pi 4)
 - Running method: deconz-community/deconz-docker podman container
 - Firmware version: (26720700)
 - deCONZ version: (2.13.04)
 - Device: Conbee 2
 - Do you use an USB extension cable: yey
 - Is there any other USB or serial devices connected to the host system? If so: Which? - HardDisk

## deCONZ Logs

```
13:02:05:413 send permit join, duration: 65
13:02:19:315 0xFCFC nwk changed to 0xDEEA
13:02:19:316 device announce 0x00158D0006D4DEF5 (0xDEEA) mac capabilities 0x80
13:02:19:317 set fast probe address to 0x00158D0006D4DEF5 (0xDEEA)
13:02:19:317 FP indication 0x0000 / 0x0013 (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:317                       ...     (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:318 device announce 0x00158D0006D4DEF5 (0xDEEA) mac capabilities 0x80
13:02:19:336 DEV found DDF for 0x00158D0006D4DEF5, path: 
13:02:19:344 FP indication 0x0104 / 0x0000 (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:345                       ...     (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:345 ZCL attribute report 0x00158D0006D4DEF5 for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
13:02:19:534 FP indication 0x0104 / 0x0000 (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:535                       ...     (0x00158D0006D4DEF5 / 0xDEEA)
13:02:19:536 Remember Xiaomi special for 0x00158D0006D4DEF5
13:02:19:537 ZCL attribute report 0x00158D0006D4DEF5 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
13:02:19:537 0x00158D0006D4DEF5 extract Xiaomi special attribute 0xFF01
13:02:19:538 	01 battery 3145 (0x0C49)
13:02:19:539 	03 Device temperature 29 °C
13:02:19:539 	04 unknown 424 (0x01A8)
13:02:19:540 	05 RSSI dB (?) 108 (0x006C)
13:02:19:540 	06 LQI (?) 4294967296 (0x0100000000)
13:02:19:541 	0a Parent NWK 60247 (0xEB57)
13:02:20:213 [1] get node descriptor for 0x00158d0006d4def5
13:02:20:214 ZDP get node descriptor for 0xDEEA
13:02:24:414 saved node state in 0 ms
13:02:24:415 sync() in 0 ms
13:02:24:889 FP indication 0x0000 / 0x8002 (0x00158D0006D4DEF5 / 0xDEEA)
13:02:24:889                       ...     (0x00158D0006D4DEF5 / 0xDEEA)
13:02:24:890 ZDP indication search sensors 0x00158D0006D4DEF5 (0xDEEA) cluster 0x8002
13:02:24:890 ZDP indication search sensors 0x00158D0006D4DEF5 (0xDEEA) clear timeout on cluster 0x8002
13:02:24:905 [2] get active endpoints for 0x00158d0006d4def5
13:02:24:906 ZDP get active endpoints for 0xDEEA
13:02:25:654 ZCL attribute report 0x00158D000774335F for cluster: 0x0201, ep: 0x01, frame control: 0x1C, mfcode: 0x1037 
13:02:28:101 FP indication 0x0000 / 0x8005 (0x00158D0006D4DEF5 / 0xDEEA)
13:02:28:102                       ...     (0x00158D0006D4DEF5 / 0xDEEA)
13:02:28:102 ZDP indication search sensors 0x00158D0006D4DEF5 (0xDEEA) cluster 0x8005
13:02:28:103 ZDP indication search sensors 0x00158D0006D4DEF5 (0xDEEA) clear timeout on cluster 0x8005
13:02:28:115 [4] Skipping additional attribute read - Model starts with 'lumi.'
13:02:31:683 Bind response success for 0x00158d000774335f ep: 0x01 cluster: 0x0001
```

hille721, 2021-12-22:
my current workaround: using the Docker image with version 2.12.06, adding the sensor and upgrading to the new version again

Mimiix, 2021-12-22:
@manup was able to reproduce.

FuzzyMistborn, 2021-12-22:
Interesting, I'd tried reverting to 2.12.06 but it didn't work.  Traveling for the holidays, will try after I get back.

manup, 2021-12-22:
The problem appears to be a small regression from v2.13.3-beta https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5460#pullrequestreview-838547549

bumpyflea, 2021-12-29:
Can confirm I'm having the same problem since updating to HA core-2021.12.6 and Phoscon 2.13.04 / 12/18/2021

cmowgli, 2021-12-30:
Can confirm same problem. Firmware 26720700, Software 2.13.04 / 18.12.2021

Mimiix, 2021-12-30:
Can we please stop adding a can confirm?

TorbenEC, 2021-12-30:
[image]

Not being very adept with Github or development, does this mean this issue won't be fixed before beta v.2.14.0, which given next Beta is v2.13.3, seems quite far away?

Betoniara, 2021-12-30:
Can confirm I'm having the same problem since updating to HA core-2021.12.7 and deCONZ Current version: 6.11.1

oli737, 2021-12-30:
WXKG11LM Rev 2
Home Assistant 2021.12.5
deCONZ Current version: 6.11.1
FW 26720700
Same Problem, worked before. Cant add Sensor with Phoscon again.

Mimiix, 2021-12-30:
I'm locking this 

> 

The next beta is 2.14.0 as we already have a stable on 2.13.4.

closed 2022-01-08 by manup