dresden-elektronik/deconz-rest-plugin #5600: Device nameSupport for Salus SP600 Smart Plug

incarvr6, 2021-12-22:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

incarvr6, 2021-12-22:
As on this inclusion request
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/812

My last post is ignored.

Multiplier is wrong for current and total consumption.

incarvr6, 2021-12-22:
[image]

Mimiix, 2021-12-22:
Please fill the form instead of working around the template.

closed 2021-12-22 by close-issue-app[bot]

incarvr6, 2021-12-22:
Why do same thing twice?
It is nearly impossible to do it on phone!!!

Mimiix, 2021-12-22:
We need the screenshots regardless because those show how the device presents them in the clusters. That issue is ages old and could have a way different firmware then you do.

Working around the template is not appreciated and feels very disrespectful.

incarvr6, 2021-12-22:
So tell me how I'm supposed to do screenshots on phone??!

incarvr6, 2021-12-22:
[image]
Can you read something in here?? Well I can't...

Mimiix, 2021-12-22:
I can but it's not complete. But Frankly, it isn't our fault that you don't have access to a PC.

I'm locking this issue. Ether way, feel free to open another issue once you got access to a PC and the information.