dresden-elektronik/deconz-rest-plugin #5601: Since 2.13.04 No Motion detection

rnagler01, 2021-12-23:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Intel NUC)
 - Running method: (Home Assistent deCONZ Add-on)
 - Firmware version: (26660700)
 - deCONZ version: (2.13.04)
 - Device: (ConBee II )
 - Do you use an USB extension cable: (yes) 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

Mimiix, 2021-12-23:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

Can you check that?

rnagler01, 2021-12-23:
Yes i checked, didn´t helped.
[image]
[image]

rnagler01, 2021-12-23:
The Sonsors are the whole time in a connected state, its the motion detection which is not showing up.
I already repaired all devices.

rnagler01, 2021-12-23:
I see in Deconz both sensor types are reading the occupancy sensor
[image]
But in phoscon the occupancy is missing
[image]

Mimiix, 2021-12-23:
What does the redt api show?

rnagler01, 2021-12-23:
How can i check this?

Mimiix, 2021-12-23:
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

rnagler01, 2021-12-23:
Not getting too much.
[image]

I sent the get through my pc in the same local network.
Deconz is running in a container on a intel NUC.
I am using the supervisored hass.io Installation

rnagler01, 2021-12-23:
I get this if i open the url 
[image]

Mimiix, 2021-12-23:
You need to use the ip of the device running deconz.

rnagler01, 2021-12-23:
i got the IP, its the same you saw above. I ran the same command via cli and this is the right IP of my deconz device

rnagler01, 2021-12-23:
OK, so i have now the API Key

curl -X POST -H "Content-Type: application/json" -d '{ "devicetype": "my application" }' http://172.30.33.0:40850/api
[{"success":{"username":"XXXXXXXX"}}]

rnagler01, 2021-12-23:
This is the API respose of the sensors
Osram Motion sensor
"ep":1,"etag":"0129e78fccf8b6adb16b15bf3556b05f","lastannounced":null,"lastseen":"2021-12-23T18:38Z","manufacturername":"CentraLite","modelid":"Motion Sensor-A","name":"Motion Sensor","state":{"lastupdated":"2021-12-23T18:38:39.889","temperature":2372},"type":"ZHATemperature","uniqueid":"00:0d:6f:00:0f:7f:60:4f-01-0402"},"44":{"config":{"battery":80,"offset":0,"on":true,"reachable":true},
Sonoff Motion Sensor
"ep":1,"etag":"4f294468781f8196ecd97973103832e9","lastannounced":null,"lastseen":"2021-12-23T18:40Z","manufacturername":"CentraLite","modelid":"Motion Sensor-A","name":"motion.sonoff","state":{"lastupdated":"2021-12-23T18:37:26.870","temperature":2321},"type":"ZHATemperature","uniqueid":"00:0d:6f:00:0f:bf:9a:f9-01-0402"},"45":{"config":{"battery":100,"enrolled":3,"on":true,"pending":[],"reachable":true},

rnagler01, 2021-12-23:
So far i can´t find type ZHAPresence in the list of sensors, shouldn´t it be listed as its own sensor?

rnagler01, 2021-12-29:
Can you add sensors via API? Maybe it would be an idea to add the missing sensor manually?

Smanar, 2021-12-31:
No more users with this issues ?

You have changed a special setting on DDF setting ?

>Can you add sensors via API? Maybe it would be an idea to add the missing sensor manually?

It s possible using DDF file, but your device is a little strange, it use IAS cluster on an endpoint, and the presence cluster on a second endpoint ?
From the code It seem deconz ignore the second one

```
        else if (*i == OCCUPANCY_SENSING_CLUSTER_ID)
        {
            if (sensor->modelId() == QLatin1String("Motion Sensor-A"))
            {
                continue; // use ias zone cluster
            }
            val = sensor->getZclValue(*i, 0x0000); // occupied state
        }
```

You are sure don't have a ZHAalarm sensor somewhere ? (Not a ZHAPresence)

rnagler01, 2022-01-01:
Hi,

thank you for your reply. I got two out of the three sensors to work. I notice two sensors /Nodes in Deconz, i deleted these and was able to pair them successfully again.

  "50": {
        "config": {
            "battery": 90,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "842c11d07ea28331a9b7e4daf8d9f359",
        "lastannounced": null,
        "lastseen": "2022-01-01T17:45Z",
        "manufacturername": "CentraLite",
        "modelid": "Motion Sensor-A",
        "name": "motion.kitchen",
        "state": {
            "lastupdated": "2022-01-01T17:45:03.197",
            "temperature": 2277
        },
        "type": "ZHATemperature",
        "uniqueid": "00:0d:6f:00:0f:7f:60:4f-01-0402"
    },
    "51": {
        "config": {
            "battery": 90,
            "duration": 60,
            "enrolled": 1,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "e437240eb6f82fb8975fc79e948e0678",
        "lastannounced": null,
        "lastseen": "2022-01-01T17:45Z",
        "manufacturername": "CentraLite",
        "modelid": "Motion Sensor-A",
        "name": "motion.kitchen",
        "state": {
            "lastupdated": "2022-01-01T17:20:16.657",
            "presence": false
        },
        "type": "ZHAPresence",
        "uniqueid": "00:0d:6f:00:0f:7f:60:4f-01-0500"
    },

Take Motion.kitchen for example, same sensor and now i got two nodes with the same 
UniqueID but different ending
-01-0402 for ZHATemperature
-01-0500 for ZHAPresence

Always in the same order, first Temp then motion. 

But i have still the third sensor, same model osram + motion and i can´t figure out why i can´t get it running

"47": {
        "config": {
            "battery": 80,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "400cf0c33a2cf06c001d62baf9c3e6b2",
        "lastannounced": null,
        "lastseen": "2022-01-01T17:46Z",
        "manufacturername": "CentraLite",
        "modelid": "Motion Sensor-A",
        "name": "motion.desk",
        "state": {
            "lastupdated": "2022-01-01T17:42:52.343",
            "temperature": 2060
        },
        "type": "ZHATemperature",
        "uniqueid": "00:0d:6f:00:0f:bf:9a:f9-01-0402"
    },
    "5": {
        "config": {

PS All sensors worked before the metioned update.

Smanar, 2022-01-02:
So re-include the device make them works again ?
And I realy don't see a difference with the not working one ....
Can you try to keep it awake during the inclusion ? generaly with short press on the reset button ?

rnagler01, 2022-01-03:
No Luck. 
- delete Node and restart HA
- Add Sensor (again just the temp added)
- set phoscon in pairing mode read again basic attribute
- scan fail.

manup, 2022-01-03:
I have this sensor in my bathroom, here it works regardless of deCONZ version.
From the description above it looks like the ZHAPresence resource was lost somehow?

I'll create a DDF for the sensor, should be pretty simple in this case and solve above issue in future too.

rnagler01, 2022-01-03:
Hi Manup,
excuse my noobish question, but i check on the sensor by edit DDF and it seems it got already the ZHAPresence.
[image]

[image]

Seems like phoscon is somehow not getting the presence sensor from deconz. I can´t really wrap my head around, why on this sensor and not on the others.

Mimiix, 2022-01-03:
@rnagler01  you need to put the status on "bronze" otherwise it won't get loaded.

rnagler01, 2022-01-03:
Hi,
alright i changed and saved it
[image]
But still noh luck. I repaired the sensor unluckily no change.

Is there something else i can check?

Mimiix, 2022-01-03:
What does the api show?

rnagler01, 2022-01-03:
100 12898  100 12898    0     0  2519k      0 --:--:-- --:--:-- --:--:-- 2519k
        "config": {
            "battery": 80,
            "duration": 60,
            "enrolled": 1,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "7be4f7f88008253b6df1695c40bc4097",
        "lastannounced": null,
        "lastseen": "2022-01-03T19:54Z",
        "manufacturername": "CentraLite",
        "modelid": "Motion Sensor-A",
        "name": "motion.wc",
        "state": {
            "lastupdated": "2022-01-03T19:55:23.310",
            "presence": false
        },
        "type": "ZHAPresence",
        "uniqueid": "00:0d:6f:00:0f:bf:9a:f9-01-0500"
    },
    "47": {
        "config": {
            "battery": 80,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "12e0d9699e39d585fc16ae602cdc95de",
        "lastannounced": null,
        "lastseen": "2022-01-03T19:54Z",
        "manufacturername": "CentraLite",
        "modelid": "Motion Sensor-A",
        "name": "motion.wc",
        "state": {
            "lastupdated": "2022-01-03T19:54:35.412",
            "temperature": 2052
        },
        "type": "ZHATemperature",
        "uniqueid": "00:0d:6f:00:0f:bf:9a:f9-01-0402"
    },
    "5": {
--
    },
    "50": {
        "config": {
            "battery": 90,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "fa842ddaa4ef753fc5ee6080c3bf75f1",
        "lastannounced": null,
        "lastseen": "2022-01-03T19:55Z",
        "manufacturername": "CentraLite",
        "modelid": "Motion Sensor-A",
        "name": "motion.kitchen",
        "state": {
            "lastupdated": "2022-01-03T19:55:17.835",
            "temperature": 2297
        },
        "type": "ZHATemperature",
        "uniqueid": "00:0d:6f:00:0f:7f:60:4f-01-0402"
    },
    "51": {
        "config": {
            "battery": 90,
            "duration": 60,
            "enrolled": 1,
            "on": true,
            "pending": [],
            "reachable": true
        },
        "ep": 1,
        "etag": "fa842ddaa4ef753fc5ee6080c3bf75f1",
        "lastannounced": null,
        "lastseen": "2022-01-03T19:55Z",
        "manufacturername": "CentraLite",
        "modelid": "Motion Sensor-A",
        "name": "motion.kitchen",
        "state": {
            "lastupdated": "2022-01-03T19:51:14.553",
            "presence": true
        },
        "type": "ZHAPresence",
        "uniqueid": "00:0d:6f:00:0f:7f:60:4f-01-0500"
    },
    "58": {
--
    },
    "62": {
        "config": {
            "battery": 80,
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "67f1c113814257a1a9400f3abc8d6b35",
        "lastannounced": null,
        "lastseen": "2022-01-03T19:55Z",
        "manufacturername": "CentraLite",
        "modelid": "Motion Sensor-A",
        "name": "Motion Sensor",
        "state": {
            "lastupdated": "2022-01-03T19:52:42.378",
            "temperature": 2254
        },
        "type": "ZHATemperature",
        "uniqueid": "00:0d:6f:00:11:08:79:2b-01-0402"
    },
    "63": {
        "config": {

The Last Sensor ist the metioned one ("name": "Motion Sensor",), above are same model sensors which are working find, after the workaround i did.

Smanar, 2022-01-04:
It s this one ?

```
    "config": {
        "battery": 80,
        "duration": 60,
        "enrolled": 1,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 1,
    "etag": "7be4f7f88008253b6df1695c40bc4097",
    "lastannounced": null,
    "lastseen": "2022-01-03T19:54Z",
    "manufacturername": "CentraLite",
    "modelid": "Motion Sensor-A",
    "name": "motion.wc",
    "state": {
        "lastupdated": "2022-01-03T19:55:23.310",
        "presence": false
    },
    "type": "ZHAPresence",
    "uniqueid": "00:0d:6f:00:0f:bf:9a:f9-01-0500"
},
```

No ? so it is in the API.

rnagler01, 2022-01-04:
No, its this one
},
"62": {
    "config": {
        "battery": 80,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "67f1c113814257a1a9400f3abc8d6b35",
    "lastannounced": null,
    "lastseen": "2022-01-03T19:55Z",
    "manufacturername": "CentraLite",
    "modelid": "Motion Sensor-A",
    "name": "Motion Sensor",
    "state": {
        "lastupdated": "2022-01-03T19:52:42.378",
        "temperature": 2254
    },
    "type": "ZHATemperature",
    "uniqueid": "00:0d:6f:00:11:08:79:2b-01-0402"
},

motion.kitchen
motion.wc 
are the two working osrams

sn0wf4ke, 2022-01-15:
I have the same problems with this sensor. I edited the DDF manualy as shown:
[image]

but in the api these changes are not reflected:

[image]

Smanar, 2022-01-16:
The temperature sensor is reliable ?
And if you remove the ZHATemperature in the DDF to try ?
Same result if you are setting phoscon in permit jon and making "hot reload" ?
Nothing special in log with "info" and "ddf" ?

ronaldt80, 2022-01-16:
Same issue here, I have 3 of these sensors on my deconz network - worked ok for the past two years. Yesterday following a network collapse (ZigBee mesh was offline for 6hrs+) I had to repair 1 of the 3. Result is same as above, no motion, only temperature. Oddly though the two others work as intended.

ronaldt80, 2022-01-16:
Update - I rolled back to 2.12.06, repaired the sensor, motion recognized, upgraded back to 2.13.04 and sensor is still there working as intended.

sn0wf4ke, 2022-01-17:
for me it is the same ... took docker image for 2.12.06 and paired the sensor without any hassle. only the version string is the strange "????????????????????????" ... but the sensor works as expected

ahielg, 2022-01-17:
> Update - I rolled back to 2.12.06, repaired the sensor, motion recognized, upgraded back to 2.13.04 and sensor is still there working as intended.

Did you rolled back the conbee stick firmware too?

sn0wf4ke, 2022-01-17:
no... only the deconz software

ahielg, 2022-01-17:
what is your stick firmware version?

ronaldt80, 2022-01-17:
No, I installed the current stable fw already as part of 12.06; I did see however (ha community forum) that some people solved (other) pairing issues by reinstalling the fw

ronaldt80, 2022-01-17:
26720700

Smanar, 2022-01-17:
I don't think on my side the firmware have an impact on this issue

rnagler01, 2022-01-17:
Check my comments above, i already try to edit the DDF, without any success.
And yes Sensors ending with 01-500 Presence and the other Temp.

ronaldt80, 2022-01-17:
This seems to be a sonoff one, mine is an OSRAM /centralite sensor.

The Sonoff sensors never worked for me on deconz, I do have them working with ZHA.

Smanar, 2022-01-17:
Can you share your DDF to take a look ?
Someone is able to compile the code to test a change ? (need a linux OS with full/real OS)

rnagler01, 2022-01-17:
No, the sonoff are working, but i have the osram in use as well. 
Two are working again, after deleting some ghost in deconz. But one sensor this issue the whole time.
I will share quickly the DDF

rnagler01, 2022-01-17:
The DDF
[image]
[image]
[image]

[image]

So far i understand the system, if you look in deconz the sensor is complete, temp and presence.

But if you check in phoscon, then you have just the Temp sensor.

Smanar, 2022-01-17:
Just to be sure
All devices have 
```
"manufacturername": "CentraLite",
"modelid": "Motion Sensor-A",
```

There is a missing part on the DDF, but you have 
`"type": "$TYPE_PRESENCE_SENSOR",`
and
`"type": "$TYPE_TEMPERATURE_SENSOR",`

rnagler01, 2022-01-17:
> Just to be sure All devices have
> 
> ```
> "manufacturername": "CentraLite",
> "modelid": "Motion Sensor-A",
> ```
Yes thats True
> 
> There is a missing part on the DDF, but you have `"type": "$TYPE_PRESENCE_SENSOR",` and `"type": "$TYPE_TEMPERATURE_SENSOR",` ?

Yes in the DDF are both.

ronaldt80, 2022-12-12:
Still have the issue with deconz as well. Moved all my centralite sensors over to my other zigbee network on ZHA.. there they work.

TungstenE2, 2024-01-05:
> Hello, ran in the same problem with 2 CentraLite Motion sensors. Only temperature and battery is updated. Motion is detected (green flash on device), but not reflected in the Phoscon. Deleting and re-adding the sensor didn't help. Any advice? Thx a lot!

Same here, after repairing, no motion as data point any more, but green light on sensor.

Older motion sensor without repair still working.

deconz version 2.24.2

Smanar, 2024-01-05:
But state_motion_count not increase ? It's a counter from FHEM ?
Those device don't use occupancy cluster but the ias one, it mean they need enrollement (config/enrolled) but this value is hidden.

Can take time, if it have fail, or can try a new inclusion.
You can see the enrollement using logs or making this value visible

```
        {
          "name": "config/enrolled",
          "public": true
        },
```

TungstenE2, 2024-01-05:
> But state_motion_count not increase ? It's a counter from FHEM ? Those device don't use occupancy cluster but the ias one, it mean they need enrollement (config/enrolled) but this value is hidden.
> 

no, sorry. This is the sensore paired sometime ago and is still working.

Seems there used to be a motion count reading in the past, but was not updated by device for nearly 2 years.

TungstenE2, 2024-01-05:
by the way, this did not help
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

@cralar could you update the name of the ticket, so it better reflects the issue of the Osram smart+ motion sensor missing a value after update from 2.13.04

This ticket is open for 2 years now. Would be great if someone could fix, as it was working before.

Thx

SwoopX, 2024-01-06:
@TungstenE2 Please refrain from supposedly proving deconz errors with any 3rd party software, as this is not really helpful and often that software is the issue.

In your case, I believe I know what's going on, being some sort of chicken and egg situation between deconz and the perl module. Deconz exposed some more resource items than you currently see with the newly paired device. This can have 2 reasons: 1) those  items are not supported by the module (often the case for new items), 2) an item hasn't issues any websocket change notification.

If memory serves, the module adds the readings upon reception of a websocket notification, if it is not already present. That being said, it seems the PIR cannot generate such a change notification, as the DDF seems to be incomplete and missing a vital configuration section being the binding with the IAS Zone cluster. This is required so the sensor knows where to send its data to and currently, there's no recipient defined.

I'd suggest to replace the CentraLite DDF code with the one below and re-pair the device. You also might want to set the detection reset interval `config/duration`, but that resource item does not seem to be exposed to FHEM by the module.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "CentraLite",
  "modelid": "Motion Sensor-A",
  "product": "Motion Sensor-A",
  "vendor": "OSRAM",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0019", "at": "0x0002", "eval": "var v = Attr.val; Item.val = [(v >> 24) & 255, (v >> 16) & 255, (v >> 8) & 255, v & 255].join('.')"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0019", "at": "0x0002"},
          "refresh.interval": 84000
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
[... 128 lines cut ...]
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x0016"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

TungstenE2, 2024-01-06:
Thx @SwoopX for taking the time to look into it.

I did as you suggested.

Did a backup of file 'motion_sensor-a.jason' in /usr/share/deCONZ/devices/centralite.

Updated your code in 'motion_sensor-a.jason'.

Deleted the sensor in Phoscon.

Paired the sensor again in Phoscon.

Stil the same problem. Motion is not received.

The other sensor paired quite some time ago is still working fine.

So something must have changed in the code for the centralite sensor, that now with new paring motion is no longer transmitted,
but with existing old pairing it is still working. Same as other users reported before.

SwoopX, 2024-01-06:
You should restart deconz after having made the amendments. Was the screenshot taken after you made them? I'd say so and the missing third binding is a good indication for that.

There might be additional things to explore, as I checked the legacy code and there are quite some specialties, but one thing at a time.
A screenshot of the node info pane of the device from deconz GUI could also help spotting any differences.

TungstenE2, 2024-01-06:
I just checked and I did a copy&past error over VNC, so your chage was not copied over.

Fixed this now and after reboot 3rd binding IAS Zone is listed.

Deleted the sensor and did new pairing, but still no motion detected.

[image]

SwoopX, 2024-01-06:
Ok, 2 things to move ahead. Can you please check the "zone status" attribute in the IAS Zone cluster? When there's motion detected, the value should differ from 0x0000. But please note that you must manually read the attribute.

Presumably, the attribute reporting needs to be set up here as well, as it seems the device doesn't send any automatic status change notifications.

TungstenE2, 2024-01-06:
I see a difference between both sensors in cluster info for IAS Zone:

latest pairing:
[image]

old pairing with working motion detection:
[image]

TungstenE2, 2024-01-06:
btw, why are 2 senor IDs created for this device?

Other sensors only get 1 ID, although they also have the state, battery and teperature? eg Sonoff SNZB04 door sensor

[image]

SwoopX, 2024-01-06:
Because the device offers the functionality of 2 seperate devices. I personally dislike it, but that's the way it is.

So, you might want to try this modified DDF, now containing some reporting on the zone state

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "CentraLite",
  "modelid": "Motion Sensor-A",
  "product": "Motion Sensor-A",
  "vendor": "OSRAM",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0019", "at": "0x0002", "eval": "var v = Attr.val; Item.val = [(v >> 24) & 255, (v >> 16) & 255, (v >> 8) & 255, v & 255].join('.')"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0019", "at": "0x0002"},
          "refresh.interval": 84000
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
            "eval": "const vmin = 20; const vmax = 30; var bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;"
          },
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0001", "at": "0x0020"},
          "refresh.interval": 3700
        },
        {
          "name": "config/checkin",
          "public": false,
          "awake": true,
          "refresh.interval": 3600
        },
        {
          "name": "config/duration",
          "default": 60
        },
        {
          "name": "config/enrolled",
          "default": 1
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0402"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "read": {"fn": "none"},
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0019", "at": "0x0002", "eval": "var v = Attr.val; Item.val = [(v >> 24) & 255, (v >> 16) & 255, (v >> 8) & 255, v & 255].join('.')"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 255,
            "eval": "const vmin = 20; const vmax = 30; var bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 3600,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x0016"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500",
      "report": [
        {
          "at": "0x0002",
          "dt": "0x19",
          "min": 60,
          "max": 3600
        }
      ]
    }
  ]
}
```

TungstenE2, 2024-01-07:
Yes! This seems to be the solution. Thx a lot.

I now get motion again after re-pairing, although it took some moments.

[image]

But I also see a difference in the 2 sensors now.

Both sensors are on my desk just in front of me.

They both report motion quick, but the old sensors changes to no motion after some moment, while the new paired sensors keeps reporting motion, which is right. So the new paired sensors is more accurate. So I should also re-pair the old sensors, to match the behavior.

As this is a local fix now, will you deploy this to next build, in order to fix for all users?

And why was this not fix within 2 years, as several people reported? I assume the code was changed at one point in time, and should have just been rolled back to previous version from repository. Or am I worng in saying so?

But thx again.

SwoopX, 2024-01-08:
Finally, good to hear it works now 👍 

I don't know how the state of code was back then and prefer not dump hours into that, but I also don't want to pull out excuses here. What I can say is that the device works non-compliant to the standard here and requires a configuration being extremely unusual. In the end, that's the reason why it was difficult to spot and required some extensive code study. It's such cases which make the legacy code extremely error prone and maintenance intensive.

However, this is not yet the 100% solution to go with, as there are some further specialties which might be difficult to handle.

I've hardcoded 60 seconds "cool off" time through `config/duration`, which should take care of resetting the detection back to false. I've also set the minimum reporting interval to 60 seconds. What I don't know is how well the values play together and what potential impact there is on the reset. You can set `config/duration` anytime, but not the reporting invervals without restart. The legacy code also takes care that there's some alignment of those values, but I believe that is not necessarily important.

Long story short: I don't know if the device would send an attribute report on its own when motion goes back to false or what the minimum timeframe is in which the device can detect 2 motions (motion - nomotion - motion). As I don't have the device, it would be great if you could do some more tests to find out the right settings here.

[5 more replies]

closed 2024-02-03 by manup