dresden-elektronik/deconz-rest-plugin #5603: TS0601 / _TZE200_2atgpdho

wookash, 2021-12-23:
Please add this TRV device, I can pair it i deconz but it's not visible in Phoscon and in Home assitant it is an unknown device without entities

## Device

Product name: Tuya HY367 (it actually says Zigbee Radiator Actuator), looks like this https://pl.aliexpress.com/item/1005002848416473.html
Manufacturer: _TZE200_2atgpdho
Model identifier: TS0601
Device type : Thermostat (TRV)

## Screenshots

[image]
[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-12-23:
Missing all screens

wookash, 2021-12-23:
hi @Mimiix, apologies, I think I've added missing screens now, do let me know if anything else is needed

Mimiix, 2021-12-23:
Looks good :)

Mimiix, 2021-12-23:
@Smanar can this be done with ddf?

wookash, 2021-12-23:
btw. I'm using deconz in docker container on RPI 4

Smanar, 2021-12-23:
> @Smanar can this be done with ddf?

Hu :(, nope, DDF don't support yet the tuya cluster.

wookash, 2021-12-23:
Is there anything I can do now or should I return the device?

Mimiix, 2021-12-23:
@wookash what is your expectation?

wookash, 2021-12-23:
> @wookash what is your expectation?

Ideally, I'd like to be able to use the device in Home assistant. Is it possible to add this device to deconz so that it's recognised and can be used in HA? if yes, what is the ETA on this change please?

Mimiix, 2021-12-23:
> > @wookash what is your expectation?
> 
> Ideally, I'd like to be able to use the device in Home assistant. Is it possible to add this device to deconz so that it's recognised and can be used in HA? if yes, what is the ETA on this change please?

Not for another month, perhaps even longer.

wookash, 2021-12-23:
> > > @wookash what is your expectation?
> > 
> > 
> > Ideally, I'd like to be able to use the device in Home assistant. Is it possible to add this device to deconz so that it's recognised and can be used in HA? if yes, what is the ETA on this change please?
> 
> Not for another month, perhaps even longer.

ok, thanks. I've moved to zigbee2mqtt and was able to add the TRV there without a problem. Please let me know when this device is added to deconz, I'm not sure about stability of conbee II and zigbee2mqtt

edgardavim, 2022-01-02:
@wookash no error from your side?
I have also the Hy367, I managed to link to HA via zigbee2mqtt, but after connecting, just stops working. Showing a huge "1P" at the valve screen.

at Zigbee2mqtt I have this details:
Zigbee Model TS0601
Zigbee Manufacturer _TZE200_2atgpdho
Manufacturer TuYa
Model TS0601_thermostat

i already tried to remove and add, but back to same status error. 

Anyone had identical or know how to move from here?

The message at the TRV
https://live.staticflickr.com/65535/51722394114_d14a1f429a_c.jpg

wookash, 2022-01-02:
> @wookash no error from your side? I have also the Hy367, I managed to link to HA via zigbee2mqtt, but after connecting, just stops working. Showing a huge "1P" at the valve screen.
>
> at Zigbee2mqtt I [...]

hi, I was able to pair it with Z2M no problem, When I get this message that you see on your screenshot, I just press 'Boost OK' (to the bottom of the display) - maybe it's stuck, give it a few tries, you should be able to configure the time and the day of the week, then it should pair no problem.

edgardavim, 2022-01-02:
@wookash dzieki Lukasz! I will give it a try tomorrow. Winter is coming in Krakow and I want to have the trv working in the flat.

edgardavim, 2022-01-03:
@wookash what do you have in Availability? In mine is showing "Disabled".

It looks to be working, but the value of temperature showing in HA, only updates if I touch the screen of the valve. Then feels like it wakes up and sends data back to HA.

[image]

wookash, 2022-01-03:
hi @edgardavim, I'm new ti Z2M and my all devices are  showing unavailable (havent configured it), but I think it's normal. This TRV does not update too often, so I created an automaiton in HA ass per below, I'm calibrating temp using external sensor, but you can just set payload to 0 and run this automation e.g. every 5 min or so - this will update the temp reading form the TRV

`service: mqtt.publish
data:
  topic: zigbee2mqtt/{firendly name}/set/local_temperature_calibration
  payload_template: >-
    {{(state_attr('termostat entity name','local_temperature_calibration')|float
    -
         state_attr('termostat entity name', 'current_temperature')|float +
        states('external sensor entity name')|float)|round(1,'half')}}`

Mimiix, 2022-01-03:
@wookash @edgardavim this isn't the z2m git repo.

wookash, 2022-01-05:
> > > @wookash what is your expectation?
> > 
> > 
> > Ideally, I'd like to be able to use the device in Home assistant. Is it possible to add this device to deconz so that it's recognised and can be used in HA? if yes, what is the ETA on this change please?
> 
> Not for another month, perhaps even longer.

Thanks, where can I check for this information so that I'm not bothering you? Or shall I get back here to check?

Mimiix, 2022-01-05:
> Thanks, where can I check for this information so that I'm not bothering you? Or shall I get back here to check?

Just keep track of this issue :) @Smanar probably replies when he did a PR or something else.
@Smanar  is a PR easy for this one?

Smanar, 2022-01-05:
It seem it s a clone of the device "_TZE200_ckud7u2l"
So there is only 1/2 line to edit to add this device, but all PR are locked, not sure if it something usefull ?

Mimiix, 2022-01-05:
Switches take a while. I'll check with manup.

Smanar, 2022-01-06:
Sure if needed I can make a PR, will be fast, and lot of chance it will work direclty without tests.

Adzter, 2022-01-28:
@Smanar is it possible to make a PR for this? It looks like this will work

wookash, 2022-01-28:
Hello. Is there any update on this please?

Smanar, 2022-01-28:
I think too it will work, but even 1 code line edition are blocked https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5404

So I can explian you how to do if you can compile the code, but I make the less PR possible.

Adzter, 2022-01-28:
If you can explain the change so that it can be compiled manually that will be fantastic.

Is there a ticket for tracking Tuya Cluster support for DDF?

Smanar, 2022-01-28:
>Is there a ticket for tracking Tuya Cluster support for DDF?

Nope, for the moment need to wait, the next step is the battery switch support, and I hope after the tuya cluster.

So first change todo.
- look for "_TZE200_ckud7u2l"
- add {"_TZE200_2atgpdho", "TS0601", "Tuya", "Tuya_THD HY369 TRV"},
- look for "kud7u2l"
- add 2 time `if (manufacturer.endsWith(QLatin1String("atgpdho"))`

For the compilation you need a linux machine with Full/real OS.

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz (The code is based on realy last one, so need a recent version):

```
sudo apt install deconz-dev
git clone https://github.com/dresden-elektronik/deconz-rest-plugin.git
cd deconz-rest-plugin
```
make the changes.
```
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

It don't impact your zigbee database, but can use a lot the CPU (at least the first time, compilation will be faster after).
To rollback fastly you can make a backup of the file /usr/share/deCONZ/plugins/libde_rest_plugin.so in another folder.

Exellent1988, 2022-11-25:
So it's been over a year now but it's still not included, corret? Ny progress so far?

bchhabra, 2022-12-18:
Any update on this? Would be really nice if this is included.

Smanar, 2022-12-19:
Not about the c++ code, but we will have "soon" the tuya support by DDF, but there is already 100 PR in waiting list, so can take some time.