dresden-elektronik/deconz-rest-plugin #5606: 13.4: 0x[MAC] error APSDE-DATA.confirm: 0xE9 on task, 0xE1 on task, 0xAE on task, etc.

SanderTRBRT, 2021-12-24:
## Describe the bug

I've already had problems with the batteryless switches that gave this error (#5318), but now my whole network is starting to give errors after going to the latest version (13.4). Rebooting does not help and also half or more of the neighbour lines are not showing up.

## Steps to reproduce the behavior

Look at the network logs

## Expected behavior

Work like before
## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: 26720700
 - deCONZ version: 2.13.4
 - Device: ConBee II
 - Do you use an USB extension cable: yes-- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? Zwave stick

## deCONZ Logs

See screenshot

## Additional context

Adding a device does work but becomes unworkable after adding. Sometimes turning on, but most of the times not. It used to work stable, until the previous release seem to do something with FoH switches which started to give this error, but now most of my other devices also give these errors.

Smanar, 2021-12-24:
Hello, you have modified something on your hardware ? New SSD, New USB device, another USB port ?

Mimiix, 2021-12-24:
Hi,

As this is not a bug as per #5113, I am closing this issue.

Error codes :https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

Please post on the forum.

closed 2021-12-24 by Mimiix

Goodbonehun, 2022-12-11:
Hello Mimiix and Support Team,

I read lot of interesting articles from You. Could you please so kind to provide some assistance?
My biggest problem is that my devices and entities in deconz become unavailable after a while and I don't know what causes this and how to fix this error. Mostly lights are affected, but I have some automation definied, e.g. Turn on lights when motion sensor activated. So this kind of automation doesn't work as entities become unavailable. There is a similar issue for Philips Hue bulbs.

My system parameters:

I have Raspberry Pi 4.
Home Assistant 2022.12.1
Supervisor 2022.11.2
Operating System 9.3
Frontend 20221208.0 - latest
------------
deCONZ
Current version: 6.17.1 (Changelog)
Ports:  40850, 5900, 8081
-----------
Conbe II firmware:
2.19.03 / 2022. 09. 19.
26780700

Really appreciate all comments, fix and support.

Mimiix, 2022-12-11:
> Hello Mimiix and Support Team,
>
> I read lot of interesting articles from You. Could you please so kind to provide some assistance? My biggest problem is that my devices and entities in deconz become [...]

Hi, its best to ask on the forums :) https://forum.phoscon.de