dresden-elektronik/deconz-rest-plugin #5608: Can't detect "IKEA Leptiter" lights

bpatath, 2021-12-24:
## Describe the bug
I am not able to add IKEA Leptiter lights using the Phoscon app.
It is probably linked to #2238, which did not provide a clear solution...

I can factory reset the lights just fine using the on/off 6 times procedure. It is confirmed by the lights flashing for few seconds and the IKEA switches not controlling them anymore.

I am trying to pair them using my laptop and an USB extension cable to avoid interference.
It is worth noting that the IKEA switches are able to pair with the lights really easily, even when they are at ~20cm of the lights.
This is all while the laptop and Conbee II stay in the same position, and with the Conbee being basically touching the light (I tried anything from touching to ~10cm off of the lights).
So it is not likely for it to be an interference issue.

On the deCONZ gui, nothing was showing up for most of the time until at one point, two new nodes appeared but without any information so I can't say if the are the Leptiten lights. (see screenshot below)

Also note that I can successfully pair IKEA Tradfi bulbs, as well as other IKEA sensors and switches.
My only issue is with the Leptiter ones.

As I am new to Zigbee, I can't really do more debugging on my own.
Any help would be appreciated and I can provide more information if required for debugging purpose.

## Steps to reproduce the behavior

- Open deCONZ
- Open phoscon
- Go to Lights and Add new lights
- Factory reset the lights
- Place the Conbee II really close to the lights
- Wait

## Expected behavior
The lights should eventually pair with the Conbee.

## Screenshots

[image]

## Environment
 - Host system: PC
 - Running method: Ubuntu
 - Firmware version: 26720700
 - deCONZ version: 2.12.6
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
Logs taken with the suggested options. I started the pairing process in Phoscon before resetting the lights.
After few seconds, the first light looks like it paired to something (it stopped blinked) so I moved the Conbee to a second light, which kept blinking all along.
On the deCONZ gui app, it looked like the screenshot above before and after. No changes that I could see.
On the phoscon app, no lights were detected until the process enventually timed out.

```
17:03:45:296 Mgmt_Lqi_req zdpSeq: 92 to 0x00212EFFFF0708FB start index 0
17:03:45:297 APS-DATA.request id: 70, addrmode: 0x03, addr: 0x00212effff0708fb, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
17:03:45:312 APS-DATA.confirm id: 70, status: 0x00 SUCCESS
17:03:45:313 APS-DATA.confirm request id: 70 -> confirmed, timeout 8184478
17:03:45:320 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 221, rssi: 30
17:03:45:321 APS-DATA.indication request id: 70 -> finished
17:03:45:322 APS-DATA.request id: 70 erase from queue
17:03:45:322 ZDP status = 0x00 -> SUCCESS
17:03:45:323 ZDP Mgmt_Lqi_rsp zdpSeq: 92 from 0x00212EFFFF0708FB total: 1, startIndex: 0, listCount: 1
17:03:45:324     * neighbor: 0x5C0272FFFE9BE427 (0x2647), LQI: 255, relation: 0x01 rxOnWHenIdle: 0
17:03:53:820 APS-DATA.request id: 106, addrmode: 0x02, addr: 0xfffc, profile: 0x0000, cluster: 0x0036, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x00
17:03:53:821 send permit join, duration: 59
17:03:53:822 APS-DATA.request id: 107, addrmode: 0x02, addr: 0xfffc, profile: 0xA1E0, cluster: 0x0021, ep: 0xF2 -> 0xF2 queue: 1 len: 6 tx.options 0x00
17:03:54:495 APS-DATA.confirm id: 106, status: 0x00 SUCCESS
17:03:54:496 aps request id: 106 finished, erase from queue
17:03:54:501 APS-DATA.confirm id: 107, status: 0x00 SUCCESS
17:03:54:576 aps request id: 107 finished, erase from queue
17:04:01:381 Search sensors done
17:04:01:819 APS-DATA.request id: 140, addrmode: 0x02, addr: 0xfffc, profile: 0x0000, cluster: 0x0036, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x00
[... 686 lines cut ...]
17:07:55:968 APS-DATA.request id: 187 erase from queue
17:07:55:968 ZDP status = 0x00 -> SUCCESS
17:07:55:969 ZDP Mgmt_Lqi_rsp zdpSeq: 107 from 0x00212EFFFF0708FB total: 2, startIndex: 1, listCount: 1
17:07:55:970     * neighbor: 0x5C0272FFFEF269E1 (0x9B73), LQI: 252, relation: 0x01 rxOnWHenIdle: 0
17:07:56:069 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:56:372 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:56:676 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:56:978 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:57:282 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:57:584 MAC Poll 0x02 0x9B73, life time: 59998 sec, dev timeout: 61440 sec
17:07:57:887 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:58:205 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:58:494 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:58:797 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:59:101 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:59:403 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:07:59:706 MAC Poll 0x02 0x9B73, life time: 59998 sec, dev timeout: 61440 sec
17:08:00:009 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:08:00:312 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:08:00:615 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
17:08:00:917 MAC Poll 0x02 0x9B73, life time: 60000 sec, dev timeout: 61440 sec
```

## Additional context
Nothing

Qonstrukt, 2022-01-01:
Funny you mention my original ticket, I eventually got it to work by keeping my RPi with Raspbee super close to the Leptiter lamps. But I'm facing this problem again with LED2005R5 lamps. I cannot for the life of me get them paired.
I'm always baffled how horribly bad pairing still works for Zigbee 3.0 lamps.

WhimsySpoon, 2022-01-08:
I too have this issue with the LED2005R5. I can reset the bulb into pairing mode without issue, but it's just not detected by Phoscon or Deconz.

closed 2022-02-24 by github-actions[bot]