dresden-elektronik/deconz-rest-plugin #5613: Tuya Temp/Hum sensor TS0201 ( no screen & LR03 version )

KipK, 2021-12-27:
## Device

- Product name: Tuya Smart Zigbee 
- Manufacturer: _TZ3000_bguser20
- Model identifier: TS0201
- Device type : 
  - Sensor

## Screenshots

[image]

[image]

[image]

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature
[image]
[image]

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]

Smanar, 2021-12-29:
Hello, this device realy look classic, I think it can work with a DDF file, like here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5483#issuecomment-981131783

KipK, 2021-12-31:
Do you have some documentation somewhere about how using this?
I've tried a bit, could have an initial update of values through api then nothing. Seems I miss some details here

Smanar, 2021-12-31:
If you have initial value but nothing after it s probably because of that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5483#issuecomment-981757918

Bind is used to tell the device to send value, and the reporting need to be configured to set mworking mode.

simontegelid, 2022-01-01:
@KipK try this: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5299#issuecomment-1003585131 I have a bunch of these sensors working with that DDF file.

KipK, 2022-01-03:
@simontegelid 
Thanks for pointing it. I'm not home for a week, will try that when I'm back

Cheers

KipK, 2022-01-08:
The provided DDF file works well by renaming the manufacturer to "_TZ3000_bguser20"

Thanks for the fast prompt guys.

simontegelid, 2022-01-08:
Glad it worked for you! I didn't have to rename the manufacturer, even though bguser20 is in my manufacturer strings on the sensors. Why is it important that they match?

KipK, 2022-01-08:
You've posted while I was editing my previous one.

As soon as I changed the manufacturer, on Deconz gui the node got a name instead of hex strings.

simontegelid, 2022-01-08:
Ah. Ok. I got various strange names, so maybe that explains it. However, I managed to rename the sensors in the phoscon app using a workaround for a bug in there. I posted the workaround in the corresponding issue on the matter, but cant find it atm. Will post a link here if I can find it.

KipK, 2022-01-08:
After some testing, seems It doesn't survive to Phoscon reboot.
Some new FF endpoints mysteriously appears thereafter.

simontegelid, 2022-01-09:
Hm, ok. I have my status set to Bronze and the "wrong" manufacturer. Survives reboots and reconnects as expected. Simply works like a charm. Would be great to hear from someone more introduced to the internals of Deconz explaining the meaning of a correct manufacturer string and the meaning of various status levels.

Smanar, 2022-01-09:
On my side, I don't know what happen, but I know 2/3 things
- the issue appear on launch and can be solved using the "gold" status.
- on a previous comment manup had a rough idea where the issue is from
- with the bad "manufacturename" I think the DDF is just not used, easy to check, right clic on the node, "edit DDF", and check the path in the window title.
- Status will be for future, ATM the only impact they have is in the "basic" mode on DDF control.

KipK, 2022-01-10:
Found what was causing my reboot problem. I was storing the DDF file in /usr/share/dresden.... , so my edited ddf through the GUI ( when I've changed the manufacturer name ) was not saved because of write permission ( It didn't complained while saving it ).
So putting it in ~/config/dresden... , things survive reboot now. My mistake sorry for that.

I'm really happy with those module. Seems they take in count the binding rates parameters as I receive an update each 5mn, seems in sync with the 300s I have setup.

Smanar, 2022-01-10:
Ha nice, realy usefull to known, perhaps it's the solution.

closed 2022-02-09 by github-actions[bot]