dresden-elektronik/deconz-rest-plugin #5614: Cant control Philips LLC010 as single device

yesnomaybewtf, 2021-12-28:
## Describe the bug

I added a Philips LLC010 (Iris) to my setup. 
Im not able to switch it on, change color etc pp, except when im switching on/off the whole group where the lamp is added.
Via Hue Essentials im able to change color etc, but also only when im doing this on the group.
## Steps to reproduce the behavior

Add LLC010 to deconz, try to change color on the single lamp.
## Expected behavior

Well, ich expect to change color etc on this single lamp, not on the group
## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS): Rpi 4
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine): Docker Container
 - Firmware version: (26xxyy00): 26720700
 - deCONZ version: (2.xx.yy): 2.13.01 / 6.10.2021
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II): Conbee 2
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which?: no

## deCONZ Logs

Trying to switch the light on via Home (iOS). 
`{
    "08:16:13:209": {
        "e": "changed",
        "id": "29",
        "r": "lights",
        "state": {
            "alert": "none",
            "bri": 254,
            "colormode": "xy",
            "effect": "none",
            "hue": 52457,
            "on": true,
            "reachable": true,
            "sat": 254,
            "xy": [
                0.1438,
                0.7997
            ]
        },
        "t": "event",
        "uniqueid": "00:17:88:01:00:0d:8e:70-0b"
    }
}`
Trying to switch the light on via Phoscon App:
`{
    "08:17:55:1": {
        "e": "changed",
        "id": "29",
        "r": "lights",
        "state": {
            "alert": "none",
            "bri": 254,
            "colormode": "xy",
            "effect": "none",
            "hue": 52457,
            "on": true,
            "reachable": true,
            "sat": 254,
            "xy": [
                0.1438,
                0.7997
            ]
        },
        "t": "event",
        "uniqueid": "00:17:88:01:00:0d:8e:70-0b"
    }
}`
## Additional context

Mimiix, 2021-12-28:
Hi,

Can you provide screenshots of all clusters?

yesnomaybewtf, 2021-12-28:
Hey Dennis,
is a screenshot from the vnc ok?

Mimiix, 2021-12-28:
Sure, but make sure to have all clusters

yesnomaybewtf, 2021-12-28:
Here we go.
The node Kugel ist the LLC010, strangly i dont see a link to another node, but im able to turn it on with the Phoscon App
[image]
.

Mimiix, 2021-12-28:
We need all clusters read https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

yesnomaybewtf, 2021-12-28:
Sorry, the screenshots are attached
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
.

Mimiix, 2021-12-28:
@ebaauw is this one you can integrate?

ebaauw, 2021-12-28:
OP didn't read the cluster values, or the _Read_ failed as well.

The LLC010 is a Hue Living Color Iris, a first-generation, gamut-A _Color Light_.  I don't have that model myself, but afaik it uses the same firmware as the Hue Bloom (LLC011) and original Hue light stripe (LST001).  These have been supported by deCONZ and by the REST API plugin for at least six years, even before my time.

If a light can be controlled through a group, but not directly, that usually indicates routing issues on the Zigbee network.  Do you see any Zigbee errors in the deCONZ log?  Did you try and power cycle the light?  Did you try to enable or disable source routing?

> The node Kugel ist the LLC010, strangly i dont see a link to another node, but im able to turn it on with the Phoscon App

That means the light hasn't been reported by any router in their neighbour table.  Could just be a timing issue; the routes are refreshed at low priority in background, but could also mean that indeed none of the other routers know of this light.  This would cause unicast commands to fail, as the gateway wouldn't know where to send them.  Again, power cycling the light (causing a _Device Announcement_) should probably fix this situation (at least temporarily).

yesnomaybewtf, 2021-12-28:
Its very strange.
The lightstrip next to the iris works well, also the bloom, one floor above.
I placed the iris now in the room where my deconz stick is located.
The iris now has connections to other nodes, and changing color etc works well.
Its not the optimal solution, but the best for now.

Thanks for your help!

closed 2021-12-28 by yesnomaybewtf

Pindar, 2022-02-16:
@yesnomaybewtf I'm experiencing the exact same problem. Did you make any progress?

yesnomaybewtf, 2022-02-17:
> @yesnomaybewtf I'm experiencing the exact same problem. Did you make any progress?

The only thing that helped was placing the bloom next to my deconz stick. So the bloom has a direct connection to deconz without any other device between.

yesnomaybewtf, 2022-02-17:
Convinced my wife to buy another lightstrip, and now got the iris in my room :)