dresden-elektronik/deconz-rest-plugin #5620: Thermostat offset unnecessarily clamped to -25/25 for Eurotronics SPZB001

LordOfTheRats, 2021-12-29:
## Describe the bug

Using the REST API to set 

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Mimiix, 2021-12-29:
There is no information in the template. I'm coding the report.

LordOfTheRats, 2021-12-29:
Sorry, i fat-fingered the submit button early. I have edited the report. Can you please reopen it?

Thank you!

Mimiix, 2021-12-29:
Sure :)

LordOfTheRats, 2021-12-29:
Thank you :)

In current master the clamping happens here https://github.com/dresden-elektronik/deconz-rest-plugin/blame/master/rest_sensors.cpp#L979

LordOfTheRats, 2022-01-21:
Still relevant

LordOfTheRats, 2022-02-14:
Still relevant.

I realized I didn't include a screenshot of the device details.
[image]

Anything else I can do to get this looked at?

LordOfTheRats, 2022-03-14:
Still relevant.

@Mimiix Any suggestion who the best person to tag would be? Maybe @SwoopX, since they seem to be the last person to work on this?

LordOfTheRats, 2022-04-05:
Anything I can do to get this looked at?

LordOfTheRats, 2022-06-22:
bump

@Mimiix @SwoopX what can I do to get this looked at?

manup, 2022-06-22:
Just did a little test with my SPZB001 when I write a value below 500 (5°C) to the *Current Temperature Setpoint* 0x4003 attribute, like 0, the device clamps it automatically to min. 500.

From the manual control buttons on the device the minimum settable value is  5°C, however max. value seems to be 30°C.

LordOfTheRats, 2022-06-23:
Hi @manup,

thank you for having a look :)

My issue is with the _Local Temperature Calibration_ 0x0010. The SPZB001 accepts a (raw) range of -128 to 127 (= -1280 to 1270 using the REST API). This, however, gets clamped down to -25 to 25 at https://github.com/dresden-elektronik/deconz-rest-plugin/blame/master/rest_sensors.cpp#L1115. (I am currently running deCONZ 2.16.1)

Clamping happens for all devices that are not handled otherwise. Does this make sense? Is a range of -25/25 the default for most devices, except the SPZB001?

In any case, an exception for the SPZB001 would be nice, clamping to -128/127, or not clamping at all.

Thank you again for your help :)

LordOfTheRats, 2022-07-15:
I propose something like this:
```diff
diff --git a/rest_sensors.cpp b/rest_sensors.cpp
index d9fa48eb..8711fa4b 100644
--- a/rest_sensors.cpp
+++ b/rest_sensors.cpp
@@ -1108,6 +1108,16 @@ int DeRestPluginPrivate::changeSensorConfig(const ApiRequest &req, ApiResponse &
                             updated = true;
                         }
                     }
+                    else if (sensor->modelId() == QLatin1String("SPZB0001"))
+                    {
+                        if (data.integer < -128) { data.integer = -128; }
+                        if (data.integer > 127)  { data.integer = 127; }
+
+                        if (addTaskThermostatReadWriteAttribute(task, deCONZ::ZclWriteAttributesId, 0x0000, THERM_ATTRID_LOCAL_TEMPERATURE_CALIBRATION, deCONZ::Zcl8BitInt, data.integer))
+                        {
+                            updated = true;
+                        }
+                    }
                     else if (sensor->type() == "ZHAThermostat")
                     {
                         if (!devManaged)
```

SwoopX, 2022-08-07:
I'm currently working on the DDF version. Then you should be able to set whatever you want or amend any boundaries at will.

closed 2022-09-06 by github-actions[bot]