dresden-elektronik/deconz-rest-plugin #5622: deconz-rest-plugin did not work anymore with Version > 2.11.02

dghh, 2021-12-29:
## Describe the bug
I always compile the plugin by myself. Due to some other circumstances i did not compile a new version since deconz-2.09.02.
To update my system i compiled the actual version deconz-2.13.04. But the service won't start as usual.
There is an error when starting the service:

"deconz.service - deCONZ: ZigBee gateway -- REST API
Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
Active: activating (auto-restart) (Result: exit-code) since Wed 2021-12-29 15:14:24 CET; 6s ago
Process: 12550 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=8080 (code=exited, status=127)
Main PID: 12550 (code=exited, status=127)"

And in the raspberry logfile i find something like this " /usr/bin/deCONZ: symbol lookup error: /usr/share/deCONZ/plugins/libde_rest_plugin.so: undefined symbol: _ZNK18QHttpRequestHeader5valueERK13QLatin1String"

## Steps to reproduce the behavior

I've compiled several different versions and everything bigger then the 2.11.02 version crashes.
Compilation with:

sudo apt install deconz-dev
git clone https://github.com/dresden-elektronik/deconz-rest-plugin.git
cd deconz-rest-plugin
git checkout -b mybranch HEAD
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins

I've also updated to the actual RaspBee firmware 26400500
## Expected behavior
It should not crash with the latest version after compilation

## Screenshots

## Environment
    Host system: Raspberry Pi 4 on Raspbian GNU/Linux 10 (buster)
    Running method: Buster
    Firmware version: 26400500
    deCONZ version: 2.09.03 / 9/19/2021
    Device: RaspBee I
    Do you use an USB extension cable:
    Is there any other USB or serial devices connected to the host system? NO

## deCONZ Logs
## Additional context
I'm running RaspBee since years and checked nearly everything, but no success. I saw that at least something changed between 2.11.02 and 2.11.03
--> "Important: If you use a custom compiled plugin, it needs to be recompiled cleanly due ABI changes in the deCONZ library."
But i did not understand what it means and what to do ?? But maybe it is something completely different because didn't follow the last year the discussions here as usual in the years before

regards Georg

Smanar, 2021-12-30:
Hello, there was so much change on the dev package since this version.
You can install the last deconz version or try again

`sudo apt install deconz-dev`

deconz-dev est déjà la version la plus récente (2.13.01-raspbian-buster-beta).

dghh, 2021-12-31:
Thanks. I already tried "sudo apt install deconz-dev" two times. 

What do you mean with "last deconz version" ? Not the plugin ?

But there is this error in the log. It seems that something is missing, but i cannot find out what. Scanned already many articles about this in the net. What is this symbol and where is it used in the code ?
" /usr/bin/deCONZ: symbol lookup error: /usr/share/deCONZ/plugins/libde_rest_plugin.so: undefined symbol: _ZNK18QHttpRequestHeader5valueERK13QLatin1String"

Smanar, 2021-12-31:
You can't use a lib compiled for deconz 2.13, but using a deconz 2.09, the dev package is updated for the actual deconz version, so the 2.13 ATM.

The compilation just update the lib, not the deconz version.

dghh, 2022-01-01:
Ok, thanks. You mean i have to update the .deb and then i can compile and the 2.13.x lib ?

So something like this first:
wget -O deconz-latest-beta.deb https://deconz.dresden-elektronik.de/raspbian/beta/deconz-latest-beta.deb 
sudo dpkg -i deconz-latest-beta.deb

But i wonder why the compilation of the lib worked for so long time (many releases before 2.09.x) without updating my initial deconz version, because i normally only compile the lib and nothing else.

Anyway, i will try

Thanks again for the fast help !

Smanar, 2022-01-01:
Yep, I think it will be fine.

The code have moved a lot this year, and if the lib is able for exemple to use news feature, the deconz app need to be updated to use them too.

dghh, 2022-01-04:
Ok, it was the missing update to the new 2.13.xxx.deb

Thanks !

closed 2022-01-04 by dghh