dresden-elektronik/deconz-rest-plugin #5633: TZ3000_gjnozsaz appears but only power and link quality - No power meter

waazdakka, 2022-01-03:
## Describe the bug
The plug appears in deCONZ with all its capabilities, but values stay by default in the electrical mesurement and the simple metering panels

## Steps to reproduce the behavior
- Pair Tuya TZ3000_gjnozsaz TS011F plug to deCONZ
- See them appear into deCONZ and succeed to use them (on / off)
- Go to Status panel
- See nothing changing in the electrical measurement and simple metering panels

## Expected behavior
- Obtain the consumption and other electrical values on deCONZ

## Screenshots
[image]
[image]
[image]

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Home Assistant deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.13.04 / 18/12/2021
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

Mimiix, 2022-01-03:
What if you click read?

If the values don't change, the device probsbly doesn't support it.

waazdakka, 2022-01-03:
Hmm it's sold as power metering plug :'( 
https://fr.aliexpress.com/item/1005003523027932.html

Mimiix, 2022-01-03:
If you click read on the clusters, does it show power consumption? Make sure something is plugged in.

waazdakka, 2022-01-04:
Nothing, it stays at zero and I have a device plugged in. In found some similar issue on Z2MQTT, don't know if it could be a related-issue? https://github.com/Koenkk/zigbee2mqtt/issues/10487

Mimiix, 2022-01-04:
That doesn't say anything about the device power metering

akinek, 2022-01-08:
I have the same plug. 

[image]
[image]
[image]
[image]

Smanar, 2022-01-09:
Ok, so for you it seem working, can try with a DDF https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
you have a sample here, used for most of plugs https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5577

raekmacka, 2022-01-10:
> Ok, so for you it seem working, can try with a DDF https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet you have a sample here, used for most of plugs #5577
> 
> With text editor, just need to edit thoses lines, the modelid is probably the same (visible on basic clsuter).
> 
> ```
>     "manufacturername": "_TZ3000_w0qqde0g",
>     "modelid": "TS011F",
>     "product": "NAS-WR01B",
>     "path": "/devices/neo/NAS-WR01B.json",
> ```

I am not that familiar to DDF (and kind of new to deconz). l don't get this to work. Ill changed the manufacturnam and got new entities in HA but with value unknown. Trying to understand what I am doing wrong but have no clue.

[image]

[image]

[image]

[image]

Smanar, 2022-01-10:
I don't understand whay you have 3 entities ?

All 3 are for the same device ? you can find the device JSON in phoscon / help / API Information.
The DDF don't look so bad
The status is other than "draft" ?
You have enabled DDF in the "control" tab ?
Can take "hot reload" in the menu of the DDF editor.
When you select "edit DDF" on the node, what is the path in the windows title ?
where have you saved the DDF ?

raekmacka, 2022-01-11:
> I don't understand why you have 3 entities ?
> 
> All 3 are for the same device ? you can find the device JSON in phoscon / help / API Information. The DDF don't look so bad The status is other than "draft" ? You have enabled DDF in the "control" tab ? Can take "hot reload" in the menu of the DDF editor. When you select "edit DDF" on the node, what is the path in the windows title ? where have you saved the DDF ?

No it aint the same. Ill got five plugs so there is nothing wrong there. But somehow I manage to get double up of all of them like power_28_2 and so on when i changed in the DDF and hot reloaded.
 
Oh thanks, I was trying to find where I could find the API JSON on a easy way.

    "lastannounced": null,
        "lastseen": "2022-01-11T07:38Z",
        "manufacturername": "_TZ3000_gjnozsaz",
        "modelid": "TS011F",
        "name": "Consumption 27",
        "state": {
            "consumption": null,
            "lastupdated": "none"
     
I have been tested status draft and status silver.
DDF is enabled in control tab
Have tested hot reload

now its devices/smartplug.json
I am on a Windows machine and have tested to save the DDF in AppData/Local/Dresden..../deconz/devices~ and C:/Deconz/devices/...

Now I found something! In the API information I found that I have sensors named power 36,38,40,42 also and they got the power information! But I can't find the sensors in HA. Somehow they are doubled up and just the first one with no information Ill find in HA.

Mimiix, 2022-01-11:
In the Control tab, is the Status you use for the DDF enabled? So if you have it set on "silver" , is silver also enabled in the control tab?

raekmacka, 2022-01-11:
Hmm no clue on that one. In control I have set DDF to normal. DDF status is now draft.
How do I see if the status I have been chosen have DDF enabled in control tab?

Mimiix, 2022-01-11:
[image]

If you have the DDF for the device set to "silver" make sure that "Silver" is also checked in the above screen.

raekmacka, 2022-01-11:
Hmmm.. For me settings look like. Old version? Settings somewhere else?

[image]

Mimiix, 2022-01-11:
Are you on the latest 2.13.4?

raekmacka, 2022-01-11:
Nope. Im on 2.13.2. Now I updated and got the same setting as you. Still same problem.

I deleted the devices from deconz and connected them again. Now I only got one power and one consumption. Thanks for all the help.

Mimiix, 2022-01-11:
So it's solved?

raekmacka, 2022-01-11:
> So it's solved?

Yep, its solved. Got the right information in HA now.

Mimiix, 2022-01-11:
Can you make a PR With the DDF in bronze? Just so others can benefit this?

raekmacka, 2022-01-11:
> Can you make a PR With the DDF in bronze? Just so others can benefit this?

Ill can make a try later. Maybe I should rename it, Now its called smartplugtest.json :)
Is there any instruction how it should be set up to PR with DDf in bronze?

Mimiix, 2022-01-11:
Just make sure the DDF is set to bronze. Then probably you should be able to go here:
https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/tuya and click "Add file" and upload there ( i think).

Then it should make a PR.

karsie1994, 2022-01-27:
> I am not that familiar to DDF (and kind of new to deconz). l don't get this to work. Ill changed the manufacturnam and got new entities in HA but with value unknown. Trying to understand what I am [...]

How did you fill the binding table? I have no idea how to fill the binding table or upload the DDF file to Deconz when running from within Hassio on a Rasp Pi 4.

Smanar, 2022-01-28:
> How did you fill the binding table?
> I have no idea how to fill the binding table or upload the DDF file to Deconz when running from within Hassio on a Rasp Pi 4.

You can have an issue on HA for bind https://forum.phoscon.de/t/ddf-editor-how-to-drag-bindings/1351
But we can edit the file manualy for  that, will see later, without the bind, values will be updated in HA when you read them in deconz.

To manage file on HA with docker https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5459#issuecomment-1019496135

Else can use the editor and "pre-used" path.

waazdakka, 2022-02-21:
Hello!
Coming back to the subject, I didn't understand all the edit suggestions in deConz, not sure of what to edit 🙄 But is there any chance to be corrected in the future to allow users to get power information without editing DDF-thing? 😅

Smanar, 2022-02-21:
>But is there any chance to be corrected in the future to allow users to get power information without editing DDF-thing?

DDF will be added 1 by 1 on the deconz package. But for the moment only DDF submitted in PR have a chance to be in the deconz package.

avrel-kb, 2022-03-19:
Hi I just installed a plug that is showing as “_TZ3000_gjnozsaz” and does not show the power consumption.
Reading the messages above it seems the solution was found but never finalised. 
what should be done to have the issue resolved?

Smanar, 2022-03-19:
The DDF was not submitted ^^
@raekmacka can you make the PR or paste it here for someone make it ?

avrel-kb, 2022-03-20:
Hi, so i did a few tests and tried to understand this DDF concept that was new to me.
I could get the power information in Home Assistant.
The power data is refreshed every 6 minutes, this is not ideal but fine for what I intend to do with it.

I created a DDF but it is in Deconz add-on in HomeAssistant and no idea How to export it... anyone knows ?

Smanar, 2022-03-20:
I think it s better first to try to change the reporting setting, on the third tab (the min max / chnage).
The option in red is a poll, for device that don't support reporting

To export a DDF, you can copy it from the "preview" tab, but there is probably a magic trick to do for copy/paste, you can't use a simple text editor ?

avrel-kb, 2022-03-20:
So copy / paste from the preview tab does not work when done from Home Assistant directly.
It did work when opening deconz directly from VNC (it must be activated in the options of the addon).
Please Note that the DDF was created on the basis of the existing device  neo/NAS-WR01B.json
I only changed the device manufacturer name, product ID, Vendor and Product + the read interval for the power (as screenshot above) to 30s. It seems to work.

[_TZ3000_gjnozsaz.json.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8311586/_TZ3000_gjnozsaz.json.zip)

Regarding the reporting setting, on the third tab (the min max), I tried and it had no effect. Setting to 30 s instead of 300 still give a refresh of the data every 6 minutes. 

[image]

Smanar, 2022-03-20:
Yes can be normal, there is a new tuya plug generation that don't support reporting.

De-k, 2022-03-30:
I am using 2.14.01.
Path was /usr/share/deCONZ/devices/tuya/_TZ3000_gjnozsaz.json

Will try gold setting

Dek

duepuntodue, 2022-04-02:
Hi everyone, I've done everything listed above because only on/off are displayed on Home Assistant.
Now, I can see the values...but they are not updated in real time.
Am I missing something?

Smanar, 2022-04-02:
Yep, probably the missing binding, can you show or tell me wich one DDF you are using ?

duepuntodue, 2022-04-02:
Thanks for the reply.
I'm using the one linked above by **avrel-kb** with the settings suggested by **De-k**

You can see my settings below, please let me know if you need more info, thanks a lot

[image]
[image]
[image]

duepuntodue, 2022-04-02:
[image]

Smanar, 2022-04-02:
Can you check the application version in cluster 0x0000, old version don't support reporting (69 from my memory) so deconz use polling for them.

duepuntodue, 2022-04-02:
[image]

duepuntodue, 2022-04-02:
am 68, is it a problematic one? in your experience is it a solvable problem?

Smanar, 2022-04-02:
If you have time, take a look here https://forum.phoscon.de/t/new-tuya-smart-plug-nas-wr01b-or-tz3000-w0qqde0g/819

From my memory, some users have updated their plugs.

duepuntodue, 2022-04-02:
tnx i'll let you know!

duepuntodue, 2022-04-02:
So to my understanding I only need a  Tuya Hub to upgrade the plug version, am I correct? sorry but my english is not that good to fully understand the whole thread :-(

Smanar, 2022-04-02:
Same for me ^^, and too much thing to read.

You can increase the refresh interval, to speed up the polling, but yes it seem lot of users have updated their plugs

Here it s the same than you, the TZ3000_gjnozsaz  https://forum.phoscon.de/t/new-tuya-smart-plug-nas-wr01b-or-tz3000-w0qqde0g/819/79

duepuntodue, 2022-04-02:
The important thing is that I can make it work somehow
Thank you VERY much for your kind support.

De-k, 2022-04-04:
Hi everybody,

i got it working. As mentioned before, I am using fhem as a smart home SW. One thing that bothers me is the frequency of the updates:
[power value]:
2022-04-04_16:57:15 6
2022-04-04_16:57:16 40
(...)
2022-04-04_16:57:36 6
2022-04-04_16:57:37 104
2022-04-04_16:57:38 41

I get a new value every second. I am struggling to find a specification of the json-content of the file. Is there something where this can be limited?

Dek

Smanar, 2022-04-04:
Lol, one user have too much notification and the other too less, swap your devices ^^.

Seriously, are you using a DDF ?
If yes just go on the binding tab (the third) and at reporting you can change the setting (min / max / change), if it don't work your device use probably polling, try to increase the "interval value"

duepuntodue, 2022-04-04:
> Hi everybody,
>
> i got it working. As mentioned before, I am using fhem as a smart home SW. One thing that bothers me is the frequency of the updates: [power value]: 2022-04-04_16:57:15 6 [...]

Can you please check the application version in cluster 0x0000? Are you using DDF?

De-k, 2022-04-05:
Hi,

currently I am using
[_TZ3000_gjnozsaz.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/8415841/_TZ3000_gjnozsaz.zip)

I copied the file to 
`~/.local/share/dresden-elektronik/deCONZ/devices/_TZ3000_gjnozsaz.json`
Of the user running deconz (default user pi on Raspberry)

I cannot use the gui, since my raspi is headless: only ssh terminal available. So i will try to adjust the json file according to your hints

Dek

cobirnm, 2022-04-13:
Hello,

Have you been able to pair this plug to the Zigbee network? I have received 3 of these plugs and I'm unable to pair any of them. Well I'm able to pair but they get unreachable after the pairing procedure is over. 
When I press the plug for 5 seconds to initiate the pairing procedure, it stays in pairing mode for over 7 minutes. During that time it gets in to deconz and I can control on and off. When the pairing finishes the plug gets unreachable and can't control it.

Has anyone experienced this?

lordCONAN, 2022-04-28:
I edited the json @De-k linked to match my plug (_TZ3000_kx0pris5), SSH'd into my HAOS machine, and placed the file in /mnt/data/supervisor/addons/data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices
This got power monitoring working for both the 16A and 20A version of the same plug I have.

Smanar, 2022-04-29:
But I don't understand why some users have the good folder and other no ...
And I have perhaps something new too for HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-ha-using-text-editor/1839/2

nukleuz80, 2022-05-12:
Hi there,
i tried the DDF file from @De-k and now i get the measurement sensors.
It seems to work fine but the power in Watt is 0.

Do you have any suggestions for it?

Here a screenshot from ioBroker objects and API - they are identical to each other...

[image]

`                  
{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "0bbbdee622e75c565a02da42c91c5a43",
    "lastannounced": "2022-05-12T07:47:01Z",
    "lastseen": "2022-05-12T08:21Z",
    "manufacturername": "_TZ3000_gjnozsaz",
    "modelid": "TS011F",
    "name": "Power 125",
    "state": {
        "current": 17,
        "lastupdated": "2022-05-12T08:21:28.521",
        "power": 0,
        "voltage": 233
    },
    "type": "ZHAPower",
    "uniqueid": "a4:c1:38:81:56:ec:63:9a-01-0b04"
}
`

Smanar, 2022-05-12:
So it work fine but every 90s you have a 0 zero value that overwrite the good one ?

I can't make test on my side but some device use a "filter" feature

```
        {
          "name": "state/power",
          "refresh.interval": 360,
          "read": {
            "at": "0x0400", 
            "cl": "0x0702", 
            "ep": 2,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0400",
            "cl": "0x0702",
            "ep": 1,
            "eval": "if (Attr.val != -16777215 && Attr.val != 16777215) { Item.val = Attr.val; }",
            "fn": "zcl"
          },
          "default": 0
        }
```

nukleuz80, 2022-05-12:
Hi @Smanar,

there is no other value than 0 (zero). 
I have a light with 15Watt in the plug but the value is 0 when the light/plug is on.

So there is every 90 seconds a 0/zero value.

Smanar, 2022-05-12:
Ha nope, I have bad understand.
You don't something with more consumption to test ? Because the tension is working, and they are working same in the DDF.

phwagner82, 2022-05-19:
Hi everyone!
what is the current status on this topic? I have the same problems with this type of plugs...
Hope you can fix this problem!
Kind regards

nukleuz80, 2022-05-19:
> > Hi @Smanar,
> 
> > 
> 
> > there is no other value than 0 (zero). I have a light with 15Watt in the plug but the value is 0 when the light/plug is on.
> 
> > 
> 
> > So there is every 90 seconds a 0/zero value.
> 
> > 
> 
> > EDIT: So i will check the logs for other values...
> 
> 
> 
> 
> 
> 
> 
> What does the deCONZ Gui show in the respective clusters and attributes?
> 
> 

Here is the cluster...

[image]

Smanar, 2022-05-20:
So the value is 0 too on your capture, you haven't a more powerfull device to test with your plug to have a bigger vlaue ?

phwagner82, 2022-06-15:
@Smanar  thank you very much for the file and description. The power metering is working, but only once. 
the data will not be refreshed. 
can i support you with some logs to fix the issue? 
br
Philipp

Smanar, 2022-06-15:
You have the issue on all return ? Power/ voltage / current/ consumption ?
Because the binding don't look so bad, and if the reporting don't working, the poll is enabled too in the DDF ...

[9 more replies]

closed 2022-09-07 by github-actions[bot]