dresden-elektronik/deconz-rest-plugin #5635: Switching from CT to HS isn't working

nicedevil007, 2022-01-04:
## Describe the bug
I'm not able to set the color with hue and saturation on a bulb that it's previous colormode is "ct".
But I'm able to change the color with hue and saturation if I change the mode to HS with the help of the phoscon webui.

## Steps to reproduce the behavior
setting the color of a Innr or Philips Hue bulb with Saturation and Hue values is only possible if the bulb is in XY or HS colormode
switching from CT colormode to HS isn't possible!
switching from CT colormode to XY is possible => but creates weird mixed colors (it is more white than anything else)

## Expected behavior
Switching from colormode CT to HS with just sending a hue and saturation to a bulb should be doable.

## Screenshots

## Environment

 - Host system: LXC on Proxmox
 - Running method: Ubuntu 20.04
 - Firmware version: 26720700
 - deCONZ version: 2.13.04 / 18.12.2021
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system?: no

## deCONZ Logs

## Additional context
As a workaround or hotfix I have this solution right now:
I'm creating all this stuff with node-red

Writing the last colormode to a flow/context variable.
A funciton node then checks for the old colormode, If it was CT then it will publish 2 messages instead of 1:
```
msg1 = { query, payload: { xy } delay: 0}
msg2 = { query, payload: { hue, sat }, delay: 500 }
```
msg1 to switch the bulb to anything else than CT
msg2 to adjust the color with hue/saturation

on msg2 a delay is setup to get a clean result.

nicedevil007, 2022-01-26:
Nobody can fix this?

nicedevil007, 2022-02-06:
still nobody here that knows anything about this issue?

ebaauw, 2022-02-06:
I've never seen this behaviour on any of my Hue lights.  I have no experience with innr colour lights, but this would be a very weird firmware bug.

Can you reproduce your issue using the API directly?  And in the GUI (sending a _Move to Hue and Saturation_ while the light is in _Color Temperature_ mode)?

```
$ ph get /lights/221
{
  "colorcapabilities": 31,
  "ctmax": 500,
  "ctmin": 153,
  "etag": "9f5c27b294b9ff37885553f1a66e1757",
  "hascolor": true,
  "lastannounced": "2022-01-15T22:12:17Z",
  "lastseen": "2022-02-06T19:30Z",
  "manufacturername": "Philips",
  "modelid": "LCT003",
  "name": "Living Room 1",
  "state": {
    "alert": "none",
    "bri": 254,
    "colormode": "ct",
    "ct": 369,
    "effect": "none",
    "hue": 0,
    "on": true,
    "reachable": true,
    "sat": 254,
    "xy": [
      0.675,
      0.322
    ]
  },
  "swversion": "5.130.1.30000",
  "type": "Extended color light",
  "uniqueid": "00:17:88:01:00:f0:71:18-0b"
}
$ ph put /lights/221/state '{"hue": 0, "sat": 254}'
{
  "hue": 0,
  "sat": 254
}
$ ph get /lights/221/state
{
  "alert": "none",
  "bri": 254,
  "colormode": "hs",
  "ct": 369,
  "effect": "none",
  "hue": 0,
  "on": true,
  "reachable": true,
  "sat": 254,
  "xy": [
    0.675,
    0.322
  ]
}
```

closed 2022-03-07 by github-actions[bot]