dresden-elektronik/deconz-rest-plugin #5636: vibration detection part of the Samsung Multipurpose Sensor (Samjin) either doesnt register or is very slow/stuck

wifiuk, 2022-01-05:
## Describe the bug

## Expected behavior

## Screenshots

## Environment

 - Host system: Synology NAS 
 - Running method: Docker
 - Firmware version: 26660700
 - deCONZ version: 2.13.01 / 06/10/2021
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: Which? Z-Wave Dongle

## deCONZ Logs

## Additional context

Mimiix, 2022-01-05:
Can you first make sure to run the latest version and try again?.

wifiuk, 2022-01-05:
my docker is set to pull the latest automatically, guessing this isn't the latest version then.

wifiuk, 2022-01-05:
seems something got stuck, any way, now on version:

2.13.04 / 18/12/2021

still got similar issue

Mimiix, 2022-01-05:
How long does it stay that way?

wifiuk, 2022-01-05:
i deleted the device since the upgrade, and removed integrations with home assistant it now seems to be working as expected.

Mimiix, 2022-01-05:
Could be sometimes.

Case closed?

Mimiix, 2022-01-06:
I suppose case closed

closed 2022-01-06 by Mimiix