dresden-elektronik/deconz-rest-plugin #5639: Device Request (Tuya Smart Plug 16A EU)  ( _TZ3000_typdpbpg)

m4d-maNu, 2022-01-06:
## Device

- Product name: Tuya Smart Plug 16A EU  ( _TZ3000_typdpbpg)
- Manufacturer: _TZ3000_typdpbpg
- Model identifier: TS011F
- Device type : 
  - Light

## Screenshots

[image]
[image]

Mimiix, 2022-01-06:
Perhaps you can make yourself a DDF: 

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

m4d-maNu, 2022-01-06:
I can try, but I don't think this will work. Then I use deCONZ in a Docker Container.

Flight777, 2022-01-26:
Having the same device with the same issue

Flight777, 2022-02-05:
I'm willing to give DDF a shot, but to be honest I've no clue what I should do based on the cheat sheet. Is there any tutorial/guide for beginners to follow?

Flight777, 2022-02-06:
Ok found it, if you update the manfacturer of the DDF that is now auto assigned (which is based on this one: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5577 as per https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5334), to  _TZ3000_typdpbpg it works.

Somehow this manufacturer is on another _TZ3000 models in the auto-assigned DDF.

No idea how I can make this a permanent fix part of Deconz? Anyone that could advise would be greatly appreciated 👍

ha-michael, 2022-02-12:
Hi @Flight777 , would you care to share the DDF you created?
I've got the same device and I'm also new to DDF.

AzraelDarksights, 2022-02-15:
Same Device _TZ3000_typdpbpg here

MrTopKat, 2022-02-24:
Hi,

I just received two smart plugs (Device _TZ3000_typdpbpg, model TS011F). Adding them as Sensor using Phoscon (from Home Assistant) works, but they show up under Lights and not Sensors.

When I use deCONZ I can select the On/Off plugin unit and read the Electrical Measurement (RMS Voltage and RMS Current).

In Home Assistant a Device 'On/Off plug-in unit' is created. And I'm able to control the On/Off switch.
Also two entities (Consumption and Power) are created, but they don't get updated.

I'm using HA version 2022.2.9, deCONZ (Add-on version 6.12.0 (with auto-update on), deCONZ version  2.14.1), with a Conbee II stick.

How can I get the Energy metrics working in Home Assistant?

Regards,
Guido

Smanar, 2022-03-16:
Are you using a DDF too ?
Can you give the manufacture name of your device ?

MrTopKat, 2022-04-01:
> @MrTopKat Did you figure this out? I'm in the same position. Thanks!

@bluesheep Sorry not yet

RouNNdeL, 2022-04-04:
Why does deconz not report the reading to HA when it clearly is visible in the electrical measurement cluster?

m4d-maNu, 2022-04-20:
> Perhaps you can make yourself a DDF:
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

So I tried, but I have no idea how I create an DDF when the deCONZ run in an Docker Container.

Smanar, 2022-04-20:
Some OS on docker can have access to VNC stuff like on HA (but I can't say how they do that)

closed 2022-05-19 by github-actions[bot]

TeHashX, 2022-08-02:
I have the same device, scaned for sensors with phoscon (hassos), nothing found but showed as light.
In home assistant it's just one entity, no measurement sensors :)
How to add ddf file in home assistant os?

TeHashX, 2022-08-02:
Could be an old firmware?
I don't have a tuya zigbee hub, only Conbee II, can I update firmware?

Smanar, 2022-08-02:
Hello, pls try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_typdpbpg",
  "modelid": "TS011F",
  "vendor": "Tuya",
  "product": "Tuya random plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "eval": "Item.val = Attr.val * 10;"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 360
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 360
        },
        {
          "name": "state/voltage",
          "refresh.interval": 360
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

TeHashX, 2022-08-02:
Thanks, I will, but can you please explain where should I put this code?

Smanar, 2022-08-02:
Depend of your OS, if you have HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
For other it s easy you just need to create a text file called "wahtyouwant.json" with this contain in the "devices" folder.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

TeHashX, 2022-08-03:
Thank you, I make some progress, I loaded your json but will not survive deconz restart, anyway it's an old Firmware: 0x0000004A, I ordered a hub to make firmware update and will try again.

Smanar, 2022-08-03:
>not survive deconz restart

You are using HA or a docker OS ?

TeHashX, 2022-08-04:
Home Assistant OS (Alpine linux), I entered deconz container with
`docker container ls | grep deconz
docker exec -it 5cd419070189 /bin/bash`
I'm abble to save some.json in /data/.local/share/dresden-elektronik/deCONZ/devices/ and survive reboot but if I go with vnc and load some json in ddf editor, will not survive deconz reboot.
Also, I think is a different model: TS011F by _TZ3000_gjnozsaz

TeHashX, 2022-08-04:
I found the right NAS-WR01B_2.json ddf file here https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/neo and now I have power monitoring :))
Thank you very much for the help!!!

Yuke-hd, 2022-08-08:
> Hello, pls try this DDF
>
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "_TZ3000_typdpbpg",
>   "modelid": "TS011F",
>   "vendor": "Tuya",
>   "product": "Tuya random plug",
>   "sleeper": [...]

For `Consumption` it should be `"Item.val = Attr.val * 100;"` instead of 10

I have a 1500W heater, it ran for an hour, but `Consumption` shows it only consumes 0.15kWh so it should be `* 100`
[image]
[image]

phil-schneider, 2022-10-04:
This solution with the custom DDF is also working for me.

How can we make this a permanent & user friendly (without ssh and ddf file) solution?
What needs to be done?

VivantSenior, 2022-10-19:
Hi,

Tuya TS011F _TZ3000_typdpbpg supports current and voltage reporting (see bug #6437 for more details). Would it be possible to forward this electrical measurements to Home Assistant?

Best,
Viv