dresden-elektronik/deconz-rest-plugin #5641: Matcall ZG 401224 dimmable light doesn't work correctly

moostmartijn, 2022-01-06:
## Describe the bug

I added a Zigbee dimmable light in the Deconz add-on in Home Assistant a few weeks ago. A Matcall ZG 401224. The light worked correctly, but after updating Home Assistant to a different version the light stopped working correctly.

When I turn on the switch of the light within Home Assistant, the light turns on and the switch turns off after 1 or 2 seconds. To turn off the light I have to turn the switch on and off within 1 second.

I think the light is trying to report a color status, but it is a dimmable only light.?

## Steps to reproduce the behavior

The issue happens always after adding the dimmable light. Can't get it to work correctly. 

## Expected behavior

Within Home Assistant I should be able to turn on the light with a switch without any errors. 

## Screenshots

## Environment

Host system: Raspberry Pi 4 -> Home Assistant
Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
Firmware version: 26660700
deCONZ version: 2.13.04
Device: ConBee II
Do you use an USB extension cable: yes
Is there any other USB or serial devices connected to the host system? If so: Which? Arduino Mega

## deCONZ Logs

```
TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'
2022-01-05 19:19:00 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/pydeconz/gateway.py", line 132, in session_handler
    self.event_handler(self.websocket.data)  # type: ignore
  File "/usr/local/lib/python3.9/site-packages/pydeconz/gateway.py", line 173, in event_handler
    device_class.process_raw({device_id: event})
  File "/usr/local/lib/python3.9/site-packages/pydeconz/api.py", line 40, in process_raw
    obj.update(raw_item)
  File "/usr/local/lib/python3.9/site-packages/pydeconz/api.py", line 126, in update
    async_signal_update()
  File "/usr/src/homeassistant/homeassistant/components/deconz/deconz_device.py", line 92, in async_update_callback
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 464, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 500, in _async_write_ha_state
    attr.update(self.state_attributes or {})
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 878, in state_attributes
    data.update(self._light_internal_convert_color(color_mode))
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 805, in _light_internal_convert_color
    if color_mode == COLOR_MODE_HS and self.hs_color:
  File "/usr/src/homeassistant/homeassistant/components/deconz/light.py", line 170, in hs_color
    return (self._device.hue / 65535 * 360, self._device.sat / 255 * 100)
TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'
```

## Additional context

I tried to downgrade deConz and Home Assistant without any luck. Somethimes the dimmable light is added but is unavailable within Home Assistant. Setting the brightness would still work.

Mimiix, 2022-01-06:
Hi,

This is not an error in Deconz.

@kane610 Can you point us a bit?

moostmartijn, 2022-01-06:
It is a log from Home Assistant. I also made a bug report for the Home Assistant add-on:
https://github.com/home-assistant/core/issues/63512

Mimiix, 2022-01-06:
@Kane610 @manup is there already a git issue for it?

Kane610, 2022-01-06:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5536 one example

Mimiix, 2022-01-06:
@Kane610 @manup if there's a bug that's more widely spread, can either of you please create it on behalf of the user? Now we seem to have multiple issues for the same thing. @manup can you create it?

Kane610, 2022-01-19:
This is the earliest reported incident that I know of . https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5261

moostmartijn, 2022-02-10:
Keeping it alive; this is still an issue.

moostmartijn, 2022-03-05:
Unfortunately this issue still exists!

manup, 2022-03-27:
Can anybody provide the Draft DDF as shown in the DDF editor? 

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

moostmartijn, 2022-04-13:
@manup 
> Can anybody provide the Draft DDF as shown in the DDF editor?
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

As a Home Assistant user I do not have much knowledge about Deconz itself, so I'm not sure if this is excactly what you're looking for:
[image]

If you need more information besides this screenshot, please let me know.

moostmartijn, 2022-05-05:
This issue is solved with Home Assistant update 2022.5

closed 2022-05-05 by moostmartijn