dresden-elektronik/deconz-rest-plugin #5642: Update Rasbee II  certificate error

tomjuuu, 2022-01-06:
## Describe the bug

I wanted to perform an update from, the following error message came up. What can I do here?

pi@phoscon:/tmp $ wget https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26720700.bin.GCF
--2022-01-06 15:09:33--  https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26720700.bin.GCF
Resolving deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)... 144.76.96.194
Connecting to deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)|144.76.96.194|:443... connected.
ERROR: The certificate of ‘deconz.dresden-elektronik.de’ is not trusted.
ERROR: The certificate of ‘deconz.dresden-elektronik.de’ has expired.

thanks a lot

Tom

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian 
 - Firmware version: (26690700)
 - deCONZ version: (2.08.00)
 - Device: RaspBee II

## deCONZ Logs

## Additional context

Mimiix, 2022-01-07:
Forwarded to Manup.

Mimiix, 2022-01-07:
I just checked the certificate, but it seems to be fine here:
[image]

tomjuuu, 2022-01-07:
ok thanks for your answer but then what can it be? Is it possible to turn off the certificate check?

Mimiix, 2022-01-07:
Is the time and date correct on the machine?

tomjuuu, 2022-01-07:
Yes, time and date are correct on the machine?

Mimiix, 2022-01-07:
No clue really, as the certificate seems normal. Perhaps a reboot helps?

tomjuuu, 2022-01-07:
I have already restarted several times, maybe I will reinstall.

thanks a lot.

manup, 2022-01-08:
When I run in to this problem usually the system packages for certificates were out of date (ca-certificates etc.).
Is your OS up to date via `sudo apt update && sudo apt upgrade`?

closed 2022-02-06 by github-actions[bot]

Tinge76, 2022-02-22:
Hi, I get the same error:
pi@phoscon:~ $ wget https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_                                       Rpi_0x26400500.bin.GCF
--2022-02-22 09:31:56--  https://deconz.dresden-elektronik.de/deconz-firmware/de                                       CONZ_Rpi_0x26400500.bin.GCF
Resolving deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)... 144.76.                                       96.194
Connecting to deconz.dresden-elektronik.de (deconz.dresden-elektronik.de)|144.76                                       .96.194|:443... connected.
ERROR: The certificate of ‘deconz.dresden-elektronik.de’ is not trusted.
ERROR: The certificate of ‘deconz.dresden-elektronik.de’ has expired.

What can this be?

Mimiix, 2022-02-22:
Do a wget on HTTP endpoint

Tinge76, 2022-02-22:
I'm slow on linux...
Full syntax is then: wget deconz.dresden-elektronik.de

Tinge76, 2022-02-22:
Now i am not able to start Deconz anymore....

@phoscon:~ $ systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
pi@phoscon:~ $ systemctl status deconz-gui
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2022-02-22 12:16:30 CET; 6s ago
  Process: 2641 ExecStart=/usr/bin/deCONZ --http-port=80 (code=exited, status=127)
 Main PID: 2641 (code=exited, status=127)

Feb 22 12:16:30 phoscon systemd[1]: deconz-gui.service: Failed with result 'exit-code'.
pi@phoscon:~ $ ^C
pi@phoscon:~ $ systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
pi@phoscon:~ $

Mimiix, 2022-02-22:
Not sure what you did but it Is off topic. I recommend opening an topic on the forum.