dresden-elektronik/deconz-rest-plugin #5644: eMylo Zigbee Smart Temp/Hum Sensor

Dominik-1980, 2022-01-07:
## Device

- Product name: eMylo Zigbee Smart Temp/Hum Sensor
- Manufacturer: _TZ2000_a476raq2
- Model identifier: TS0201
- Device type : Sensor

## Screenshots

[image]

[image]

[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Dominik-1980, 2022-01-07:
Okay. Sorry. It works now. Just had to wait a moment...
Only in Phoscon the sensor doesn't show up...

Mimiix, 2022-01-07:
Does it show In ha?

Dominik-1980, 2022-01-08:
> Does it show In ha? 

Yes.

Mimiix, 2022-01-08:
In that case you need to open a bug report on the phoscon forum. They add support different there

closed 2022-01-19 by Dominik-1980