dresden-elektronik/deconz-rest-plugin #5646: Firmware website down http://deconz.dresden-elektronik.de/deconz-firmware?C=M;O=D

hangfly, 2022-01-07:
The website to get firmware is down, any alternate site?

closed 2022-01-07 by close-issue-app[bot]

Mimiix, 2022-01-07:
Seems to be fine here

pdfowler, 2022-03-09:
For anyone else who finds this from a google search like I did, I'll add a little information. I was seeing the same problems yesterday and I did a little digging and found that the site wasn't down, it was just taking a really long time to resolve.
* `traceroute deconz.dresden-elektronik.de` took several minutes to complete
* `wget https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26720700.bin.GCF` took over a minute to start the download. I thought it had failed and only today did I notice it had completed.
* The `https://deconz.dresden-elektronik.de/deconz-firmware/` page never loaded in my browser

Today, it is different:
* The wget command completes in 0.4s
* The firmware page loads instantly
* `traceroute` (trace included below) still takes about a minute

So, for those who find themselves here, wait a few hours 🤷 and see if it resolves.

<details>
<summary>Traceroute Conducted 2022-03-09T11:34:41-08:00</summary>
```
% date; traceroute deconz.dresden-elektronik.de; date
Wed Mar  9 11:34:41 PST 2022
traceroute to deconz.dresden-elektronik.de (144.76.96.194), 64 hops max, 52 byte packets
 1  192.168.13.1 (192.168.x.x)  3.358 ms  94.623 ms  2.055 ms
 2  lo0.bras1.snfcca05.sonic.net (50.0.x.x)  3.553 ms  2.816 ms  3.685 ms
 3  300.irb.cr1.snfcca05.sonic.net (198.27.186.138)  274.484 ms  364.187 ms
    301.irb.cr2.snfcca05.sonic.net (198.27.186.210)  100.213 ms
 4  0.ae14.cr2.colaca01.sonic.net (198.27.244.57)  10.210 ms
    0.ae14.cr2.snrfca01.sonic.net (198.27.244.49)  18.859 ms  13.797 ms
 5  0.ae1.cr1.colaca01.sonic.net (157.131.209.65)  654.138 ms  863.540 ms  1540.885 ms
 6  0.ae0.cr1.lsatca11.sonic.net (157.131.209.86)  187.675 ms * *
 7  * 0.ae1.cr1.snjsca11.sonic.net (157.131.209.149)  9.467 ms *
 8  * * *
 9  100.ae1.nrd1.equinix-sj.sonic.net (75.101.33.185)  5.737 ms *  6.318 ms
10  100.ae1.nrd1.equinix-sj.sonic.net (75.101.33.185)  5.089 ms
    ae17.cr6-sjc1.ip4.gtt.net (69.22.130.109)  5.821 ms
    100.ae1.nrd1.equinix-sj.sonic.net (75.101.33.185)  6.375 ms
11  ae17.cr6-sjc1.ip4.gtt.net (69.22.130.109)  7.242 ms  5.777 ms
    ae9.cr6-fra2.ip4.gtt.net (141.136.110.41)  164.922 ms
12  * ae9.cr6-fra2.ip4.gtt.net (141.136.110.41)  153.164 ms  158.529 ms
13  * ip4.gtt.net (154.14.78.154)  157.160 ms
    core24.fsn1.hetzner.com (213.239.229.78)  159.549 ms
14  core24.fsn1.hetzner.com (213.239.229.78)  246.493 ms *
    ex9k2.dc10.fsn1.hetzner.com (213.239.229.62)  244.628 ms
15  static.194.96.76.144.clients.your-server.de (144.76.96.194)  217.605 ms *  376.617 ms
Wed Mar  9 11:35:47 PST 2022
```
</details>