dresden-elektronik/deconz-rest-plugin #5651: Illuminize 511.012 w/ colormode attribute though its just a dimmable light

embeddeddev, 2022-01-09:
## Describe the bug
Illumize 511.012 is just a _dimmable light_ controller **w/o color support**.

Since several months I see plenty of bugs in my Home Assistant relating to HS errors originating from the deconz-Plugin. At first I assumed that's a plugin bug for a long time. After even HomeKit was completley refusing to work with the light I digged deeper.

Now I realized by examining the device in deCONZ GUI and zll.db that the device is errorneously _recognized_ as a dimmable light **w/ color support**.

Here the related JSON attributes from node's attributes in zll.db::nodes table:
```
{"attr/id":"25","attr/lastannounced":"2021-11-22T21:53:23Z","attr/lastseen":"2022-01-09T00:34Z","attr/manufacturername":"iluminize","attr/modelid":"511.012","attr/name":"some_light","attr/swversion":"2.5.3_r49","attr/type":"Dimmable light","attr/uniqueid":"cc:cc:cc:ff:fe:xx:xx:xx-01","state/alert":null,"state/bri":54,"state/colormode":"hs","state/on":true,"state/reachable":true}
```
The light is working well w/ home assistant and connected HomeKit as long the device is "off". When it turns on it propagates the colormode attribute and deconz Hass Plugin tries to read HS values which will fail and the light isn't controllable any more in Hass or HomeKit.

I could image that this problem is related to possible DDF changes several months ago, when the problems started.
Unfortunately I don't know the correct time of first occurrence any more.

Please let me know if there are some specific more informations required.

## Expected behavior
Device recognized correctly as dimmable light **w/o** any color support

## Environment
 - Host system: Docker
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26400500
 - deCONZ version: 2.13.04 / 18.12.2021
 - Device: ConBee I
 - Do you use an USB extension cable:  yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

closed 2022-01-09 by close-issue-app[bot]

Mimiix, 2022-01-09:
You can fix this with the ddf

clic on the node in deconz then "edit ddf" 
look at the type, can replace the "Color Dimmable light" by "Color Temperature light" (use the "delete" key to remove the old one)
Save the DDF
select "hot reload"
For more detail https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
(need to enable DDF, change the satus to bronze, ect ...)

embeddeddev, 2022-01-09:
Hi @Mimiix, dank je wel.

I tried up w/ the DDF editor now up for a long time, following the cheat sheet, setting nodes DDF to gold status, many retries, manual removals of ``"state/colormode":"hs"`` from nodes attributes in zll.db, but deCONZ still announces hascolor + colormode for the node in the REST API.

DDF JSON: [preview.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/7835809/preview.json.txt)

[image]

Do you still have an idea whats going wrong?
In node's (sub-)devices view the device shows even up w/ 0300 Color Control cluster, that shouldn't be there. Reading all its attributes results in no supported attributes at all (everything is greyed out).

Attached to this post is the DDF JSON, screenshot of its gold setting, the REST API response and the node's subdevices/cluster view in GUI.

Thanks a lot so far.

[image]
[image]

Mimiix, 2022-01-09:
Asked Manuel to reply :)

manup, 2022-01-10:
The DDF looks alright :+1: 
One thing to mention here is that currently the DDF works side by side with the old code, which somehow wrongfully adds the `state/colormode`. The DDF code is a bit shy so that it only adds or updates `state` and `config` attributes but it won't remove any attributes to not cause damage :) For this we need to add a mechanism to allow this, like a strict mode for specific DDFs.

embeddeddev, 2022-01-10:
Hey Manuel and Dennis,

thanks for your quick reply though close-issue-bot directly closed my issue because I removed the "superfluous" template comments before I submitted the issue.

After a bit digging into the code I found the place to add an exception for the iluminize device, compiled and tested it successfully. DDF was turned off again so it's working w/ legacy code as expected.

I'll create a pull request afterwards.