dresden-elektronik/deconz-rest-plugin #5653: Tint remote, ZBT-Remote-ALL-RGBW, incorrect button mapping?

mwhrtin, 2022-01-10:
## Describe the bug

I just added Müller-Licht's Tint remote to my setup and I'm experience strange button behavior. Since the remote have built-in logic for long and short press for the color temperature buttons I'm unsure if this is a bug or if it's by design by Müller-Licht.

From what I understand, warmer temperature is Button 4 (value 4002) and colder temperature is Button 5 (value 5002). When you short press these two buttons the remote will cycle between three pre-defined temperature values (warm, neutral, cold), if you long press the remote allows you to set a user defined temperature within the lights full temperature range.

If I long press button 4 to set a warmer tone than the remote's preset "warm", the next time I short press button 4, button event 5002 is received instead of 4002 from Button 4 when the remote reset temperature to preset "warm". If I set a warmer tone than the remote's preset using Phoscon and press button 4, button event 4002 is received as expected. So "Button 4 triggering button event 5002" only occur when a warmer tone that the preset is set using the remote.

If the controlled light does not support color temperature or if the group is empty, a long press on Button 4 will always result in button event 5002 the next time Button 4 is pressed.

## Steps to reproduce the behavior

1. Add lights to the light group, or leave the group empty.
2. Select the correct group using the remote's group button.
3. Hold Button 4 (warmer temperature) until all lights in the group have changed temperature to a color warmer than the remote's preset "warm". If no lights exist in the group, or if the lights does not support temperature, this happens immediately.
4. Once the warmer color is set, release Button 4 and press Button 4 again.
5. Now value 5002 is received instead of 4002, and the lights will set the temperature to the preset "warm"

## Expected behavior

When pressing Button 4, button value 4002 is always sent regardless of the light's temperature or how the temperature was set.

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26400500
 - deCONZ version: 2.13.04 / 2021-12-18
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? yes, RFXtrx433XL

## deCONZ Logs

## Additional context

Mimiix, 2022-01-10:
Is this related>?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1209

mwhrtin, 2022-01-10:
Yes, I think what I reported is what @ebaauw mention in this post as the _"unsolvable bug"_  in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1209#issuecomment-636373482.

Perhaps this is a dead end then. If both Button 4 and Button 5 will result in the `ct` going down to 370 and the only way to determine what button was pushed is by evaluating how it affected the `ct` value.

closed 2022-02-09 by github-actions[bot]