dresden-elektronik/deconz-rest-plugin #5657: Offset not shown in temperature value

boehser-enkel, 2022-01-12:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian 
 - Firmware version: 26400500
 - deCONZ version: (2.12.06
 - Device: RaspBee I 
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II - 
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

Smanar, 2022-01-12:
There is a comment about offset and your device _TZE200_yw7cahqs

I don't remember all the story, but on some device it impact the termostat, not the measured value.
No impact on the device itself ?

Can be the deconz version too ? (the last PR about offset was validated in december)

boehser-enkel, 2022-01-13:
Using Phoscon on an RPI this is the latest stable version

boehser-enkel, 2022-01-13:
Same with 2.13.04 / 18.12.2021 and repairing

boehser-enkel, 2022-01-13:
Now temp got updated and the offset is shown correctly

closed 2022-01-13 by boehser-enkel