dresden-elektronik/deconz-rest-plugin #5660: Kaipule Smoke Detector ES63 (_TZE200_oapmkvnc / TS0601)

GoaMind, 2022-01-12:
## Device

- Product name: [Kaipule Smoke Detector ES63](http://www.kaipule.net/product/detail/l/en/id/53.html)  -  model no: ES63-D5Z
- Manufacturer: _TZE200_oapmkvnc
- Model identifier: TS0601
- Device type :
  - Sensor
  - Siren (not sure if it is applicable here, but device has integrated buzzer)

## Screenshots

  - Endpoints and clusters of the node
  
[image]

  - Node Info panel
  
[image]

Gateway version
[image]

### Basic

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]

### Other clusters that are not mentioned above

#### Tuya specific Cluster
[image]

#### OTAU
[image]

#### Time

***

Please let me know if any further information is needed.

GoaMind, 2022-02-08:
Do I need to provide some more info or how can I help with this?

Mimiix, 2022-02-08:
As it's a tuya device and ddf prolly don't work (@Smanar?) and #5733 being in place. You can't.

Smanar, 2022-02-09:
Yes as it use the tuya cluster and not the IAS one, will be not possible for the moment, sorry.

closed 2022-03-10 by github-actions[bot]