dresden-elektronik/deconz-rest-plugin #5663: Device name

benpalm, 2022-01-13:
## Device

- Product name: BSEED Smart Zigbee Wall-Switch.
- Manufacturer: _TZ3000_hafsqare
- Model identifier: TS0011
- Device type:
  - Other: Wall-Switch

## Screenshots

[image]

### Basic
[image]
### Identify
[image]
### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2022-01-13:
Missing screenshots. We need screens of all clusters read.

benpalm, 2022-01-13:
> Missing screenshots. We need screens of all clusters read.

Edited and attached new screenshots