dresden-elektronik/deconz-rest-plugin #5665: Tuya door magnetic sensor (iH-F001)

fmacmahon, 2022-01-13:
## Device

- Product name: Door magnetic sensor
- Manufacturer: Tuya (E-CrossStu GmbH)
- Model identifier: iH-F001
- Device type : 
  - Sensor

## Screenshots

[image]

[image]

[image]

Mimiix, 2022-01-13:
Missing screenshots

fmacmahon, 2022-01-13:
Which screenshots are missing?

Mimiix, 2022-01-13:
All clusters.

fmacmahon, 2022-01-14:
[image]

fmacmahon, 2022-01-14:
Can you please check if there is anything missing now?

Mimiix, 2022-01-14:
These aren't the clusters of the device. Please read:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

fmacmahon, 2022-01-14:
Updated! Thanks!

closed 2022-02-20 by github-actions[bot]

powermarcel10, 2022-06-18:
Can this one be added?? I have 3 of them.. I thought that they where allready compatible with deconz, but there are not..

Anything that I can do to help?

Smanar, 2022-06-18:
Sure can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_oxslv1c9",
  "modelid": "TS0203",
  "vendor": "Tuya",
  "product": "iH-F001 Door Window Contact",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/open"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

powermarcel10, 2022-06-18:
Allright thanks, but what do I have to do with this code?

Smanar, 2022-06-18:
It's a DDF https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
Just create a text file with this contain in the "devices" folder and rename it thatyouwant.json

powermarcel10, 2022-06-18:
Allright thanks! I'm trying it now but when creating a folder and/or try to save the file I'm getting an error; access denied. Any idea what to do?

Smanar, 2022-06-18:
Yep depend of OS, but on linux 

`sudo chmod -R 777 folder_name`

powermarcel10, 2022-06-18:
Wow, you're so helpful. Thanks! I'm using a raspberry pi so I think Linux indeed. I wil try it out later!

Smanar, 2022-06-18:
If you can test it and confirm it work, yes, I can purpose it.

powermarcel10, 2022-06-19:
I can confirm that they are working great! :D Thanks a lot for your help.

Smanar, 2022-06-19:
Thss, device added https://github.com/dresden-elektronik/deconz-rest-plugin/pull/6125