dresden-elektronik/deconz-rest-plugin #5666: Avatto Light Switch

Larry661, 2022-01-14:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: Avatto / Tuya Zigbee
- Model identifier: TS0011
- Device type : 
    - Switch

## Screenshots

## Screenshots

### Basic
[image]
### Identify
[image]
[image]

Mimiix, 2022-01-14:
Missing screenshots.

Larry661, 2022-01-14:
hi @Mimiix , 
do you need more screenshots ?

Mimiix, 2022-01-14:
As written in the template: We need all cluster screenshots.

Larry661, 2022-01-14:
updated @Mimiix, it is ok for you ?

Mimiix, 2022-01-14:
This aren't all clusters but just the on/off.

We need all of them.

Larry661, 2022-01-14:
ok sorry for that, i didn't unterstand.
I hope now it is good.

Mimiix, 2022-01-14:
Still missing basic

Larry661, 2022-01-17:
hi @Mimiix, there is basic info just above OTAU info, tell me if it is right ? and i saw #5663 it is look like mine, but not the same manufacturer.

Mimiix, 2022-01-17:
Looks good now :) 

@Smanar Are #5663 and this one the same? Or different?

Smanar, 2022-04-04:
Sorry for late answer, no, sot the same device, but will use same hardware I think.
And have same problem, they use a new cluster insteaod of the classic tuya one.

Smanar, 2022-04-04:
From that I m reading it use the Cluster 0xE001 instead of the Cluster 0xEF00, but works same.
So can try with a recent deconz version > = 2.15.0 and mimic this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5869

Edit:
Ok so not possible for the moment, the tuya write fonction use only one cluster. So need to edit the code first.
Are you able to compile the code to make a test ? You need a linux machine with full OS.

Mimiix, 2022-04-04:
Reopened on request.

Smanar, 2022-04-04:
The code can be more critic that I thought, will be nice if someone can test it.

pgoujon, 2022-04-05:
Hello I have this device in my home, but it doesn't work in Home Assistant (driver)
I can test something if you want but I am newbee so you should explain what to do.
I am under Windows 10 and Home Assistant is on a Raspberry Pi3 in docker.

Smanar, 2022-04-05:
Thx, but I need someone with a real/full OS, it's possible too on docker, but I can't help you for that.
I think if it don't break something we will try in blind mode, because will be hard to find someone with a full OS AND the device.

closed 2022-05-05 by github-actions[bot]