dresden-elektronik/deconz-rest-plugin #5669: Ikea Floalt light panels not changing color temperature

miguelatv, 2022-01-14:
## Describe the bug

My Ikea Floalt panels do not change color temperature when using the Phoscon app or via apple HomeKit via the Philipps Hue homebridge integration.

## Steps to reproduce the behavior

## Expected behavior
Adjusting the color temperature slider on the Phoscon app should change the color temperature of panel light.

## Screenshots

## Environment

-Host system: Linux computer
-Running method: Marthoc Docker container
-Firmware version: 26660700
-deCONZ version: 2.13.01
-Device: ConBee II
-Do you use an USB extension cable: no
-Is there any other USB or serial devices connected to the host system? no
## deCONZ Logs

## Additional context

Mimiix, 2022-01-14:
Does it work when using the rest api directly?

miguelatv, 2022-01-14:
It does not. When I change "ct" to different numbers it does not change the panels color temperature, other types of lights do work.

senilio, 2022-01-14:
I have a FLOALT panel WS 30x90, Version 1.2.217 running on deCONZ 2.13.03, and both brightness and temp works just fine via Phoscon 🤷

Mimiix, 2022-01-14:
@miguelatv it seems that it probably has to do with your personal setup. Please open an topic on the forums.

I'm closing this issue.

closed 2022-01-14 by Mimiix